VoodooI2C/VoodooI2C #281: Cursor often stops/gets stuck

Benjoyo, 2020-04-11:
On my XPS 15 9570 (i7-8750h) under Catalina 10.15.4, multi-touch and everything is working fine, but often when I simply move the cursor with one finger on the trackpad it will just stop halfway and stay there until I lift the finger and try again. This happens in intervals of just a few seconds.
Strangely there a also times where everything works fine and smooth for some minutes and the it will get stuck again. This doesn't have anything to do with CPU load (I thought that at first).

ben9923, 2020-04-13:
You'll need to attach a troubleshooting archive.

Benjoyo, 2020-04-13:
@ben9923 You are right, I'll do that if I go back to 2.4 - currently I am on 2.3 and I can say with 99% certainty that the issue is gone there (didn't occurr yet). So that is a strong hint that 2.4 broke something.
On the other hand multitouch (3 fingers) seemed to work better on 2.4, now it sometimes takes 2 tries for them to be registered. Feels like you have to put down your fingers in a very specific way/order or something.

Goshin, 2020-04-13:
try polling mode

Snoozhead, 2020-04-13:
@Benjoyo I had the exact same issue with my XPS 7590 (i9-9980HK) in Catalina 10.15.4 on release 2.4. 
Like you, I downgraded back to 2.3 where the issue hasn't reoccured - did you end up re-upgrading and submitting a troubleshooting archive? If not I'll give it a go

Benjoyo, 2020-04-14:
@Snoozhead @ben9923  I just re-upgraded to 2.4 and made an archive:

[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4476549/Archive.zip)

The issue did appear again, but not as frequent until this point. Will have to try for longer.
What I can tell for sure now is that the gestures work _way_ better in 2.4 compared to 2.3, but I guess this isn't by accident. I am just saying that because it would suck to stay on 2.3 for that reason.

@Goshin some lines of the log read:
```
2CControllerDriver::VoodooI2CDeviceNub Warning: Incompatible APIC interrupt pin (0x3b > 0x2f) and no GPIO interrupts found; if your chosen satellite implements polling then VoodooI2CDeviceNub will run in polling mode.
2020-04-14 18:21:36.211800+0200 0x105      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a369 Found I2C device: SYNA2393
2020-04-14 18:21:36.214236+0200 0x115      Default     0x0                  0      0    kernel: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM488F Warning: Could not get interrupt event source, using polling instead
```
I suppose that means I already am in polling mode?

kecinzer, 2020-04-15:
> try polling mode

Exact issue here. Switching to polling mode is much better, but in polling mode cursor often jumps when I release finger or tap and with multiple fingers its slower and laggy.

Benjoyo, 2020-04-15:
@kecinzer glad to hear others have this issue as well, so chances are good this will be fixed soon.
I guess this is a really minor bug because it is quite intermittent. 

Apart from this the magic trackpad emulation is so amazing, all gestures work flawlessly!

Goshin, 2020-04-15:
@Benjoyo 
mind giving a try to this? [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483782/Archive.zip)

msbence, 2020-04-15:
I can confirm this as well.
Polling mode, whiskey lake era dell touchpad

Benjoyo, 2020-04-16:
> @Benjoyo
> mind giving a try to this? [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483782/Archive.zip)

I will only be able to say with certainty after some more testing, but right now it seems like that fixed it! Didn't occur in the 15 minutes I am testing this. Will report back later today.
Thank you!

msbence, 2020-04-16:
> @Benjoyo
> mind giving a try to this? [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483782/Archive.zip)

I tested it as well, and the problem seems resolved, BUT: my touchpad is super "laggy", seems like the cursor is slow when my finger is on the trackpad, but the moment I lift my finger up, the cursor speeds up.

Goshin, 2020-04-16:
> I tested it as well, and the problem seems resolved, BUT: my touchpad is super "laggy", seems like the cursor is slow when my finger is on the trackpad, but the moment I lift my finger up, the [...]

I had adjusted the polling frequency in that build but I am not sure if the freq causes the stuck problem or not. Try this and see if it improves.
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4486350/Archive.zip)

msbence, 2020-04-16:
> I had adjusted the polling frequency in that build but I am not sure if the freq causes the stuck problem or not. Try this and see if it improves. [...]

This one is perfect, thank you! \o/

Benjoyo, 2020-04-16:
@Goshin I didn't have this problem @msbence had but the second build works fine for me as well 👍

kecinzer, 2020-04-17:
> > > @Benjoyo
> > > mind giving a try to this? [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483782/Archive.zip)
> > 
> > 
> > I tested it as well, and the problem seems resolved, BUT: my touchpad is super "laggy", seems like the cursor is slow when my finger is on the trackpad, but the moment I lift my finger up, the cursor speeds up.
> 
> I had adjusted the polling frequency in that build but I am not sure if the freq causes the stuck problem or not. Try this and see if it improves.
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4486350/Archive.zip)
> 
> Related commits:
> [Goshin@69254b0](https://github.com/Goshin/VoodooI2C/commit/69254b05fc3b193cc4d6b6db436f0dfc4862b080)
> [Goshin/VoodooI2CHID@2b1e7c6](https://github.com/Goshin/VoodooI2CHID/commit/2b1e7c63b2f51078eb882e29cc6c2ade987635f5)

Just tested it and in POLLING mode there is no change - cursor often jumps different way etc. like before.
But I also tested it in INTERRUPT mode and that's completely different story. OMG! Trackpad works like a charm without any jumps, breakups freezes etc. I's fluid as on my MacBook Pro 2014. Thank you a lot!

msbence, 2020-04-17:
Yeah, polling mode jumps are there for me as well, but that's not getting stuck/hanging that's why I not mentioned it. I'll try to fix my interrupt mode and see if it works.

kecinzer, 2020-04-17:
@Goshin I still experiencing weird drops when I selecting text od rearanging icons in dock. I have set 3 finger selecting and draging, so I can drag and drop items with 3 finger, but it happens also with physical button.

Benjoyo, 2020-04-18:
@Goshin I wouldn't describe it like @kecinzer does, but I also have issues with dragging with the second build you posted. The physical click behaves differently than a tap. I can tap an icon in the dock to launch the app, but when I physically click on an icon this will immediately trigger App Exposé, as if I had done a 3 finger swipe down. This makes it impossible to drag and drop dock icons.
Also, probably directly related to this, when I tap a file in Finder it will just be selected. If I physically click it however, it will directly go into the renaming mode.

I just tried, these issues aren't present in the 2.4 release build.

Goshin, 2020-04-18:
Disable **System -> Preferences -> Trackpad -> Force Click and haptic feedback**

Benjoyo, 2020-04-18:
> Disable **System -> Preferences -> Trackpad -> Force Click and haptic feedback**

Aah, how did I miss this? Thanks, awesome work you are doing here!

nguyentdat27, 2020-04-22:
> I had adjusted the polling frequency in that build but I am not sure if the freq causes the stuck problem or not. Try this and see if it improves. [...]
I had trouble with drag item within one finger double tap and drag since offical released 2.4. Your built one fixed this issue, thanks so much.

kecinzer, 2020-04-26:
Are the changes in latest Archiv.zip now included in version 2.4.1?

kprinssu, 2020-04-28:
@kecinzer the above changes are now in 2.4.2. Closing as this should be fixed. Please reopen if it's not.

closed 2020-04-28 by kprinssu

LeeBinder, 2020-04-28:
profoundly awesome - you folks ROCK !!!