VoodooI2C/VoodooI2C #282: Touchscreen doesn't work after wake from sleep

WildBeastRouen, 2020-04-12:
My device is Dell XPS 15 7590 which has I2C touchscreen(WCOM490A) and trackpad(SYNA2393).
The trackpad's GPIO mode works perfectly with VoodooI2C 2.3, but the touchscreen doesn't work after wake from sleep (it works well before sleep), and there's an error in Voodoo log "pin 36 cannot be used as IRQ". 
I've checked that 36 is equal to the IRQ value which showed in win10 and mac's IOreg.

BTW, after updated I2c to 2.4, the touchscreen become weird even at polling mode. When I touch screen with 2-4 fingers, it performs like only one fingers for trackpad, when I touch screen with 5 fingers, it performs like 3 fingers for trackpad.
The trackpad sometimes doesn't work at all after wake from sleep.

Could you help me to solve these problems? Thanks!

ben9923, 2020-04-13:
Please attach a troubleshooting archive, preferably with both polling and GPIO.

WildBeastRouen, 2020-04-14:
I've tried 4 options and saved each Voodoo Logs: 
1. GPIO trackpad + polling touchscreen with VoodooI2c 2.3
Both devices's multitouch gesture works well on this option, no matter at boot or wake from sleep.

2. GPIO trackpad + GPIO touchscreen with VoodooI2c 2.3
Both devices's multitouch gesture works well at boot, but touchscreen doesn't work after wake from sleep.
 
3. GPIO trackpad + polling touchscreen with VoodooI2c 2.4
Only trackpad's multitouch gesture works well at boot, touchscreen can only recognize 1 finger if I put less than 4 fingers on screen, after system wake from sleep, these two devices work as same as at boot.Sometimes trackpad will not work at all after wake.

4. GPIO trackpad + GPIO touchscreen with VoodooI2c 2.4
Only trackpad's multitouch gesture works well at boot, touchscreen can only recognize 1 finger if I put less than 4 fingers on screen, after system wake from sleep, only trackpad works as same as at boot, the touchsereen doesn't work.

Log files :
[gpio+poll_2.3.docx](https://github.com/alexandred/VoodooI2C/files/4472852/gpio%2Bpoll_2.3.docx)
[dual gpio_2.3.docx](https://github.com/alexandred/VoodooI2C/files/4472853/dual.gpio_2.3.docx)
[gpio+poll_2.4.docx](https://github.com/alexandred/VoodooI2C/files/4472854/gpio%2Bpoll_2.4.docx)
[dual gpio_2.4.docx](https://github.com/alexandred/VoodooI2C/files/4472856/dual.gpio_2.4.docx)

ben9923, 2020-04-14:
> I've tried 4 options and saved each Voodoo Logs:
>
> 1. GPIO trackpad + polling touchscreen with VoodooI2c 2.3
>    Both devices's multitouch gesture works well on this option, no matter at boot or wake [...]

Please attach copies of IOReg for each, and original + patched ACPI :)

WildBeastRouen, 2020-04-14:
> Please attach copies of IOReg for each, and original + patched ACPI :)
> 
> Check for a BIOS update and update if possible. This requires re-patching ACPI if you're statically patching.

I've updated the BIOS to latest version, here are the IORegs and original+patched ACPI :
(Because IORegs zip file exceeds 10MB, I uploaded it to Google Grive) 

[https://drive.google.com/file/d/19uAAf6arXR43VY6PF76UJeoZLBkJRCyo/view?usp=sharing](url)
[origin.zip](https://github.com/alexandred/VoodooI2C/files/4476955/origin.zip)
[patched.zip](https://github.com/alexandred/VoodooI2C/files/4476956/patched.zip)

kprinssu, 2020-04-28:
Please try using version 2.4.2 of VoodooI2C.

WildBeastRouen, 2020-04-29:
> Please try using version 2.4.2 of VoodooI2C.

After updated to 2.4.2, the touchscreen still not able to use and showed the same error in Voodoo log as before.BTW I solved this problem by changing the GPIO pin to 0x1B manually (using SSDT patch).

But there's something wrong with touchpad, it showed hundreds of VoodooGPIO warning "Failed getting hardware pin for GPIO pin 28". (This GPIO pin has no problem in old version of VoodooI2C.)

IOReg file:
[IOReg.zip](https://github.com/alexandred/VoodooI2C/files/4553406/IOReg.zip)

packyan, 2020-04-30:
> > 请尝试使用2.4.2的VoodooI2C版本。
> 
> 更新到2.4.2之后，触摸屏从睡眠状态唤醒后仍然无法使用，并且在Voodoo日志中显示了与以前相同的错误。顺便说一句，我通过手动将GPIO引脚更改为0x1B（使用SSDT补丁）解决了此问题。
> 
> 但是触摸板出了点问题，它显示了数百个VoodooGPIO警告“无法获取GPIO引脚28的硬件引脚”。（此GPIO引脚在旧版本的VoodooI2C中没有问题。）
> 
> IOReg文件：
> [IOReg.zip](https://github.com/alexandred/VoodooI2C/files/4553406/IOReg.zip)

Yeah, I have the same issue, tons of GPIO pin 28 Error

kprinssu, 2020-05-15:
@WildBeastRouen I assume that your issue is resolved?

WildBeastRouen, 2020-05-15:
> @WildBeastRouen I assume that your issue is resolved?

The touchscreen's problem is solved, but the problem mentioned 16 days ago still not solved yet.

Tons of GPIO pin 28 error in VoodooGPIO log.

Goshin, 2020-05-15:
@WildBeastRouen 28(0x1C) is not a valid GPIO pin. The ranges of valid GPIO pins in the first community are 0-24, 32-57. Did you set pin 28 in DSDT?

WildBeastRouen, 2020-05-15:
> @WildBeastRouen 28(0x1C) is not a valid GPIO pin. The ranges of valid GPIO pins in the first community are 0-24, 32-57. Did you set pin 28 in DSDT?

Nope. The original APIC pin is auto assigned by ACPI, which is 0x33. 

Because cpu is Core i7-9750H, so I followed the table below and get the Group B3 IRQ:
[https://github.com/coreboot/coreboot/blob/master/src/soc/intel/cannonlake/include/soc/gpio_defs_cnp_h.h#L42](url)

And then follow the table below to convert the IRQ to GPIO pin, that is 28:
[https://github.com/coreboot/coreboot/blob/master/src/soc/intel/cannonlake/include/soc/gpio_soc_defs_cnp_h.h#L40](url)

tiger511, 2020-05-15:
@Goshin  GPIO pin 28 is valid and correct.

Goshin, 2020-05-15:
> @Goshin GPIO pin 28 is valid and correct.The bug related to not valid pin is introduced in the latest commits probably related to your pin caching.Old VoodooGPIO works fine.

At least on CannonLake, it is not.

https://github.com/VoodooI2C/VoodooGPIO/blob/master/VoodooGPIO/CannonLake-H/VoodooGPIOCannonLakeH.hpp#L414
```cpp
static struct intel_padgroup cnlh_community0_gpps[] = {
    CNL_GPP(0, 0, 24, 0),               /* GPP_A */   // 0-24
    CNL_GPP(1, 25, 50, 32),             /* GPP_B */.  // 32-57
};
```

I tend to believe there is a bug in `VoodooI2CDeviceNub` or maybe otherwhere when two devices attached.

tiger511, 2020-05-15:
The trackpad works fine though.

Goshin, 2020-05-15:
No, the error shows up because the GPIO interrupt is getting toggled by the `IOInterruptEventSource` of HIDDevice via its provider `VoodooI2CDeviceNub`, so it started to happen since v2.4.2 ([05f94a](https://github.com/VoodooI2C/VoodooI2CHID/commit/05f94a87658815e4241ba2e359c726094223e4bb)), nothing to do with the later changes of VoodooGPIO.

I am curious why it doesn't fall back to polling mode. Can you guys provide the log of loading the kexts with `kextload`?

WildBeastRouen, 2020-05-16:
> No, the error shows up because the GPIO interrupt is getting toggled by the `IOInterruptEventSource` of HIDDevice via its provider `VoodooI2CDeviceNub`, so it started to happen since v2.4.2 [...]

I've tried to put VoodooI2C and VoodooI2CHID to L/E and changed two kext's permission to root:wheel, but it's not able to load these two kexts with kextload command.

sh-3.2# sudo kextload /Library/Extensions/VoodooI2C.kext 
/Library/Extensions/VoodooI2C.kext failed to load - (libkern/kext) authentication failure (file ownership/permissions); check the system/kernel logs for errors or try kextutil(8).

Goshin, 2020-05-16:
@WildBeastRouen don't place them in /L/E.

4.4. Manually loading VoodooI2C
https://voodooi2c.github.io/#Build%20Environment/Build%20Environment

WildBeastRouen, 2020-05-16:
> @WildBeastRouen don't place them in /L/E.
> 
> 4.4. Manually loading VoodooI2C
> https://voodooi2c.github.io/#Build%20Environment/Build%20Environment

I tried to load VoodooI2C and VoodooI2CHID manually with kextload command.

After loaded these kexts, the touchscreen can only recognize one finger, and the trackpad doesn't work at all.

Here's the Voodoo log file:
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/4638863/log.txt)

Goshin, 2020-05-16:
@WildBeastRouen thanks. For WCOM490A, pin 27 (0x1B) is not correct. So it worked in polling mode.

```
kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::Failed getting hardware pin for GPIO pin 27
kernel: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM490A Warning: Could not get interrupt event source, using polling instead
```

For SYNA2393, pin 35 was somehow changed to its corresponding hardware pin 28. It is pretty weird.

```
kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::Registering hardware pin 0x1C for GPIO IRQ pin 0x23
kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::Failed getting hardware pin for GPIO pin 28
```

Can you test this one with `kextload`? It can print some additional info log.
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4638995/Archive.zip)

WildBeastRouen, 2020-05-16:
> @WildBeastRouen thanks. For WCOM490A, pin 27 (0x1B) is not correct. So it worked in polling mode.
>
> ```
> kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::Failed getting hardware pin for GPIO pin 27
> kernel: [...]

Hello.

Because the pin 27(0x1B) is not correct GPIO pin for touchscreen, I removed custom GPIO pin from SSDT and load the kexts above with kextload command. The touchscreen still doesn't work after wake from sleep.

Here're the Voodoo logs at boot and after wake from sleep:
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/4639042/log.txt)
[log_afterwake.txt](https://github.com/VoodooI2C/VoodooI2C/files/4639043/log_afterwake.txt)

Goshin, 2020-05-16:
@WildBeastRouen so the problem of pin 28 error gone?

WildBeastRouen, 2020-05-16:
> @WildBeastRouen so the problem of pin 28 error gone?

Yes, there's no pin 28 error anymore. However, once using the touchscreen, trackpad will stop working until sleep and wake from sleep.

And it seems back to the original issue, error log "pin 36 cannot be used as IRQ" popped up after wake from sleep.

Goshin, 2020-05-16:
> And it seems back to the original issue, error log "pin 36 cannot be used as IRQ" popped up after wake from sleep.

Since v2.3? The problem is that the pin turns to be in ACPI mode after waking up. I don't know much about it, probably related to ACPI or BIOS.

```cpp
    /*
     * If the pin is in ACPI mode it is still usable as a GPIO but it
     * cannot be used as IRQ because GPI_IS status bit will not be
     * updated by the host controller hardware.
     */
    if (intel_pad_acpi_mode(pin)) {
        TryLog("%s:: pin %u cannot be used as IRQ\n", getName(), pin);
        return false;
    }
```

WildBeastRouen, 2020-05-16:
> > And it seems back to the original issue, error log "pin 36 cannot be used as IRQ" popped up after wake from sleep.
> 
> Since v2.3? The problem is that the pin turns to be in ACPI mode after waking up. I don't know much about it, probably related to ACPI or BIOS.
> 
> ```c++
>     /*
>      * If the pin is in ACPI mode it is still usable as a GPIO but it
>      * cannot be used as IRQ because GPI_IS status bit will not be
>      * updated by the host controller hardware.
>      */
>     if (intel_pad_acpi_mode(pin)) {
>         TryLog("%s:: pin %u cannot be used as IRQ\n", getName(), pin);
>         return false;
>     }
> ```

2.2 or earlier versions of VoodooI2C also have this problem. I've tried to change touchscreen's interrupt mode from APIC to GPIO with modified GRUB shell but doesn't help.

Goshin, 2020-05-16:
@WildBeastRouen Have you tested the v2.4.2 release with the current config? Did you get the pin 28 error?

WildBeastRouen, 2020-05-17:
> @WildBeastRouen Have you tested the v2.4.2 release with the current config? Did you get the pin 28 error?

I tested the latest release version with current config, pin 36 error popped up after wake from sleep, and the pin 28 error still exist.

tiger511, 2020-05-17:
@WildBeastRouen which EFI are you using? 

Disable your touchscreen from BIOS and Use the last zip file from here : https://github.com/xxxzc/xps15-9570-macos/issues/29#issuecomment-629720735_

WildBeastRouen, 2020-05-17:
> @WildBeastRouen which EFI are you using?
> 
> Disable your touchscreen from BIOS and Use the last zip file from here : [xxxzc/xps15-9570-macos#29 (comment)](https://github.com/xxxzc/xps15-9570-macos/issues/29#issuecomment-629720735)_

I built the EFI with xxxzc's repository and added/modified some SSDT for XPS 15 7590, trackpad works fine with your kexts. Hope touchscreen's issue can be solved in the future.

Goshin, 2020-05-17:
> I tested the latest release version with current config, pin 36 error popped up after wake from sleep, and the pin 28 error still exist.

OK, thanks. I fix some buggy workloop issue in the debug kext (https://github.com/VoodooI2C/VoodooI2C/issues/282#issuecomment-629686426), so it seems work.

Could you plz test this to help me make sure?
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4639727/Archive.zip)

WildBeastRouen, 2020-05-17:
> > I tested the latest release version with current config, pin 36 error popped up after wake from sleep, and the pin 28 error still exist.
> 
> OK, thanks. I fix some buggy workloop issue in the debug kext ([#282 (comment)](https://github.com/VoodooI2C/VoodooI2C/issues/282#issuecomment-629686426)), so it seems work.
> 
> Could you plz test this to help me make sure?
> [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4639727/Archive.zip)

There's also no pin 28 error with these two kexts.

Because the gpio pin 36 still not able to use after wake, I changed touchscreen to polling mode. However it'll popped tons of "WCOM490A Device sent a 0-length report" error, and some "pci8086,a368 Timeout waiting for bus to accept transfer request" warning.

By the way, it becomes a little laggy while scrolling with these two modified version of VoodooI2C.

Here's the Voodoo log:
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/4639757/log.txt)

Goshin, 2020-05-17:
@WildBeastRouen yes, it works as is intended because I enabled some additional debugging log.

Plz try this one [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4639761/Archive.zip)

WildBeastRouen, 2020-05-17:
> @WildBeastRouen yes, it works as is intended because I enabled some additional debugging log.
> 
> Plz try this one [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4639761/Archive.zip)

The "WCOM490A Device sent a 0-length report" error gone, but it still pops "pci8086,a368 Timeout waiting for bus to accept transfer request" warning sometimes.

Goshin, 2020-05-17:
> The "WCOM490A Device sent a 0-length report" error gone, but it still pops "pci8086,a368 Timeout waiting for bus to accept transfer request" warning sometimes.

It is because the current polling freq is too high (500Hz). It is a little out of the scope of this issue. If it is annoying to you, maybe you can raise a new issue and mention **lvs1974**.

WildBeastRouen, 2020-05-18:
@Goshin 

After using the kexts in [#320](https://github.com/VoodooI2C/VoodooI2C/pull/320), it'll boot up slowly, both touchscreen and trackpad are not working.

Here're full boot log and Voodoo log:
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/4641837/log.txt)
[log-Voodoo.txt](https://github.com/VoodooI2C/VoodooI2C/files/4641841/log-Voodoo.txt)

Goshin, 2020-05-18:
> @Goshin
>
> After using the kexts in [#320](https://github.com/VoodooI2C/VoodooI2C/pull/320), it'll boot up slowly, both touchscreen and trackpad are not working.
>
> Here're full boot log and Voodoo log: [...]

@WildBeastRouen 
Thanks. It seems a deadlock happened when loading. Can you load this debug kext? It will collect some additional logs and help us identify the problem. [Debug.zip](https://github.com/VoodooI2C/VoodooI2C/files/4642411/Archive.zip)

After that, can you test this one? [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4642281/Archive.zip) It might fix the problem.

WildBeastRouen, 2020-05-18:
> @WildBeastRouen
> Thanks. It seems a deadlock happened when loading. Can you load this debug kext? It will collect some additional logs and help us identify the problem. [...]

The latest Archive.zip works fine with trackpad in GPIO mode and touchscreen in polling mode, do I need to provide the log while using the kexts in Debug.zip?

Goshin, 2020-05-18:
@WildBeastRouen yes if you can, it will be better for identifying the root cause.

WildBeastRouen, 2020-05-18:
> @WildBeastRouen yes if you can, it will be better for identifying the root cause.

Here're the full log and Voodoo log:
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/4645661/log.txt)
[log_Voodoo.txt](https://github.com/VoodooI2C/VoodooI2C/files/4645663/log_Voodoo.txt)

gimdh, 2020-07-11:
I have encountered same issue with 2.4.3, just for the record. My device has GPIO touchpad, GPIO stylus, and USB touchscreen /w SunrisePointLP. Sadly, I can't reproduce it consistently and it seems to happen time to time.

ben9923, 2020-07-11:
> I have encountered same issue with 2.4.3, just for the record. My device has GPIO touchpad, GPIO stylus, and USB touchscreen /w SunrisePointLP. Sadly, I can't reproduce it consistently and it seems to happen time to time.

Please attach two troubleshooting archives - one made when it's working, one when it's not.

gimdh, 2020-07-12:
It's not reproducible for now, though. I'll post report when it happens.

ben9923, 2021-01-09:
@WildBeastRouen It seems to be the same as #388 that's resolved in VoodooI2C v2.6.2+
Please give latest release a try!

Closing the issue for now, feel free to re-open as necessary.

closed 2021-01-09 by ben9923

qibinghua, 2023-02-16:
XPS 9300 has same issue.