VoodooI2C/VoodooI2C #283: pci8086,9de9 (Whiskey Lake) touchpad not working with VoodooI2C 2.4

msbence, 2020-04-14:
I have a Dell Latitude 5400 laptop with Catalina 10.15.4 installed, running on a Whiskey Lake CPU.
It has an I2C hard-surface touchpad which is getting recognized as: `DELL08B8`
The PCIID is `8086,9de9`which is supported by VoodooI2CHID.
I did the GPIO pinning via SSDT, and my touchpad is working properly with VoodooI2C 2.3. (except the buttons which is not the best with a hard-surface trackpad because I'm unable to click).

Anyways: **It's working with 2.3, but not with 2.4.** (IORegExporer shows VoodooI2C, but SysPrefs states there is no trackpad, and of course I'm unable to move the cursor.)
I'm using OpenCore 0.5.6 with Vanilla macOS (SIP enabled), VoodooI2C and VoodooI2CHID injected, VoodooInput bundled in VoodooPS2 disabled.

I'm not an expert on this, so I've attached my logs (extracted with RunMe).
[raptor-5400-v2_4.zip (2.4 - not working)](https://github.com/alexandred/VoodooI2C/files/4474976/raptor-5400-v2_4.zip)
[raptor-5400-v2_3.zip (2.3 - working)](https://github.com/alexandred/VoodooI2C/files/4474975/raptor-5400-v2_3.zip)

ben9923, 2020-04-14:
Doesn't seem like VoodooInput is loaded.
Make sure you have it injected in your OpenCore config (VoodooI2C.kext/Contents/PlugIns/VoodooInput.kext).

msbence, 2020-04-14:
Yes, you were right!
I just did a copy&replace instead of an OC snapshot...

Thank you and have a nice day! :)

closed 2020-04-14 by msbence