VoodooI2C/VoodooI2C #284: [BUG] System reboots while on sleep

dreamwhite, 2020-04-14:
As per title, when I put my hackintosh in sleep mode, by closing the lid, after few hours it reboots. The issue was the same even with VoodooI2C 2.3. I'm actually using VoodooI2C 2.4 + VoodooI2CHID. 

Below a screenshot of the panic report (which is pretty the same of VoodooI2C 2.4):

[image]

My EFI folder is available [here](https://github.com/dreamwhite/dell-inspiron-5370-hackintosh)
Actually everything is working except this problem.

`hibernatemode` and `proximitywake` are set to 0:

[image]

Many thanks in advance

ben9923, 2020-04-14:
Please provide hardware details, kernel log & IOReg :)

dreamwhite, 2020-04-14:
Oh sorry haha, here you are my specs:

**Laptop** Dell Inspiron 5370
**CPU** i5-8250U
**GPU** Intel Graphics UHD 620
**RAM** 8GB DDR4 @ 2400 MHz

**GenI2C**

- Information
[image]

- Diagnosis
[image]

**IORegistryExplorer 3.0.2 dump**
[dreamBookPro.ioreg.zip](https://github.com/alexandred/VoodooI2C/files/4475050/dreamBookPro.ioreg.zip)

**Kernel log (last 30 hours)**

[kernel_log.txt.zip](https://github.com/alexandred/VoodooI2C/files/4475060/kernel_log.txt.zip)

dreamwhite, 2020-04-14:
As suggested by @marianopela, below the dump of `sudo dmesg | grep -i voodoo > log.txt` with and without `SSCN` and `FMCN` bus speed config:

[log_with_sscn_fmcn.txt](https://github.com/alexandred/VoodooI2C/files/4475178/log_with_sscn_fmnc.txt)
[log_without_sscn_fmcn.txt](https://github.com/alexandred/VoodooI2C/files/4475179/log_without_sscn_fmcn.txt)
 
My DSDT.aml doesn't have SSCN and FMCN bus speed config and that seems a little bit strange. Also I'm running in polling mode, may I have any chance to run it with Interrupt mode?
Many thanks

marianopela, 2020-04-14:
> Please provide hardware details, kernel log & IOReg :)

Same happening on MSFT0001 with i5 10210u. I'll attach logs in a few hours

marianopela, 2020-04-14:
Here is my last panic 
[Kernel_2020-04-14-174349_MacBook-Pro.txt](https://github.com/alexandred/VoodooI2C/files/4476518/Kernel_2020-04-14-174349_MacBook-Pro.txt)

Goshin, 2020-04-15:
@marianopela @dreamwhite 
It seems to only happen in polling mode. I was wondering if it can be due to the interrupt timer not being disabled before going to sleep.

mind testing this?
https://github.com/Goshin/VoodooI2CHID/commit/8e6533c0102e6a2aad9617ab9837dc6b593f1e9a
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483279/Archive.zip) Built on 10.15.4, might not compatible with 10.14.

dreamwhite, 2020-04-15:
> @marianopela @dreamwhite
> It seems to only happen in polling mode. I was wondering if it can be due to the interrupt timer not being disabled before going to sleep.
> 
> mind testing this?
> [Goshin/VoodooI2CHID@8e6533c](https://github.com/Goshin/VoodooI2CHID/commit/8e6533c0102e6a2aad9617ab9837dc6b593f1e9a)
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483279/Archive.zip) Built on 10.15.4, might not compatible with 10.14.

I'll surely test it out on 10.15.4. Hope they fix it asap

marianopela, 2020-04-16:
> It seems to only happen in polling mode. I was wondering if it can be due to the interrupt timer not being disabled before going to sleep.

Thanks, it seems to be fixing my sleep/wake problem. Unfortunately I still experienced a KP and reboot while in sleep, as you can see here [tonight_cut.txt](https://github.com/alexandred/VoodooI2C/files/4485488/tonight_cut.txt) , with the usual `VoodooI2CControllerDriver::pci8086,2e8 Timeout waiting for bus to accept transfer request` etc

I think https://github.com/alexandred/VoodooI2CHID/pull/31 could have something to do with my KPs, as with those changes I didn't experience any.

I'll be testing more in the next hours.

Goshin, 2020-04-16:
@marianopela mind giving a try to this? have removed the `read_in_progress`.

[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483782/Archive.zip)

marianopela, 2020-04-16:
> @marianopela mind giving a try to this? have removed the `read_in_progress`.

Sure, right away. It might take some time to test eventual problems.

edd1024, 2020-04-16:
Thanks @Goshin! Works great here on a Lenovo Yoga 920. Excellent job!

marianopela, 2020-04-16:
> ~[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483782/Archive.zip)~
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4486350/Archive.zip)

What changed in the second archive?
I'm still running on the first one. Just checked the laptop, no Panics for now.

Also, could you please explain how removing `read_in_progress` doesn't compromise trackpad functioning?

Goshin, 2020-04-16:
@marianopela 
> What changed in the second archive?
> I'm still running on the first one. Just checked the laptop, no Panics for now.
> 
> Also, could you please explain how removing `read_in_progress` doesn't compromise trackpad functioning?

just removed the polling frequency adjustment, it should make no difference to your test.
from here https://github.com/alexandred/VoodooI2C/issues/281#issuecomment-614557087

> > > > mind giving a try to this? 
> > > 
> > > 
> > > I tested it as well, and the problem seems resolved, BUT: my touchpad is super "laggy", seems like the cursor is slow when my finger is on the trackpad, but the moment I lift my finger up, the cursor speeds up.
> > 
> > 
> > I had adjusted the polling frequency in that build but I am not sure if the freq causes the stuck problem or not. Try this and see if it improves.
> > [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4486350/Archive.zip)
> 
> This one is perfect, thank you! \o/

---
You can check out the explanation for the related commits here. 
https://github.com/alexandred/VoodooI2CHID/pull/31#issuecomment-614625475

marianopela, 2020-04-17:
@Goshin No panics nor reboots, trackpad always worked after every sleep/wake. Same for @dreamwhite . It's a keeper!

edd1024, 2020-04-17:
Same here, this is good! Thumbs up!

gujiangjiang, 2020-04-19:
Same with me.
Always reboot while sleep.
[image]

gujiangjiang, 2020-04-20:
> @marianopela mind giving a try to this? have removed the `read_in_progress`.
> 
> ~[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4483782/Archive.zip)~
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4486350/Archive.zip)

I tried this but still have problem with wake.
panic(cpu 0 caller 0xffffff800ce41420): "Spinlock acquisition timed out: lock=0xffffff800d6a12e0, lock owner thread=0xffffff8046de2520, current_thread: 0xffffff8046de2520, lock owner active on CPU 0x0, current owner: 0xffffff8046de2520, time: 11686257223331"@/AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-6153.101.6/osfmk/i386/locks_i386.c:501
Backtrace (CPU 0), Frame : Return Address
0xffffff81e0b6c5a0 : 0xffffff800cd215cd 
0xffffff81e0b6c5f0 : 0xffffff800ce5a3c5 
0xffffff81e0b6c630 : 0xffffff800ce4bf7e 
0xffffff81e0b6c680 : 0xffffff800ccc7a40 
0xffffff81e0b6c6a0 : 0xffffff800cd20c97 
0xffffff81e0b6c7a0 : 0xffffff800cd21087 
0xffffff81e0b6c7f0 : 0xffffff800d4c2c7c 
0xffffff81e0b6c860 : 0xffffff800ce41420 
0xffffff81e0b6c8a0 : 0xffffff800d3c5a73 
0xffffff81e0b6ccb0 : 0xffffff800d3c77dd 
0xffffff81e0b6ccd0 : 0xffffff800d3f489e 
0xffffff81e0b6ce80 : 0xffffff7f90c18d68 
0xffffff81e0b6ceb0 : 0xffffff7f90c1c223 
0xffffff81e0b6ced0 : 0xffffff800d451d93 
0xffffff81e0b6cf10 : 0xffffff7f906b9a9d 
0xffffff81e0b6cf50 : 0xffffff7f8f9d888c 
0xffffff81e0b6cf60 : 0xffffff7f8f9e1547 
0xffffff81e0b6cf80 : 0xffffff800ce4b7cd 
0xffffff81e0b6cfd0 : 0xffffff800ccc7bed 
0xffffff8200f4bad0 : 0xffffff800d3c77dd 
0xffffff8200f4baf0 : 0xffffff800d3f489e 
0xffffff8200f4bca0 : 0xffffff7f90bfa2af 
0xffffff8200f4bd00 : 0xffffff7f90bf9e35 
0xffffff8200f4bd40 : 0xffffff800d430048 
0xffffff8200f4bda0 : 0xffffff7f90c5469a 
0xffffff8200f4be00 : 0xffffff7f90c54bca 
0xffffff8200f4be20 : 0xffffff7f90bfa3de 
0xffffff8200f4be50 : 0xffffff7f90bfb93b 
0xffffff8200f4be60 : 0xffffff800d42d2fe 
0xffffff8200f4bea0 : 0xffffff7f90bfb82d 
0xffffff8200f4bec0 : 0xffffff800cd63545 
0xffffff8200f4bf40 : 0xffffff800cd63071 
0xffffff8200f4bfa0 : 0xffffff800ccc713e 
      Kernel Extensions in backtrace:
         com.apple.driver.AppleACPIPlatform(6.1)[06B0C2E0-18E0-3E3B-8A46-17BF2A713C34]@0xffffff7f8f9d1000->0xffffff7f8fa6bfff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9D1FF279-C4A2-3344-902F-E0B22B508689]@0xffffff7f8de45000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f8d731000
            dependency: com.apple.driver.AppleSMC(3.1.9)[601E041E-0A5C-3B6A-A4E6-45ECC7F48A2B]@0xffffff7f8de57000
         com.apple.driver.AppleAPIC(1.7)[B20F3195-7BF4-3B68-831E-2C180BE8CA78]@0xffffff7f906b8000->0xffffff7f906bafff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f8d731000
         as.acidanthera.voodoo.driver.PS2Controller(2.1.3)[FC4904F4-E6BE-345D-98C5-EA01141C24C1]@0xffffff7f90bf8000->0xffffff7f90c08fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9D1FF279-C4A2-3344-902F-E0B22B508689]@0xffffff7f8de45000
         com.alexandred.VoodooI2C(2.4)[A57A158F-92C3-3254-B47C-D81BB0EAD0E0]@0xffffff7f90c15000->0xffffff7f90c3dfff
            dependency: com.alexandred.VoodooI2CServices(1)[E1E325A5-365B-3FF0-96A8-CF330377D2BE]@0xffffff7f90771000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9D1FF279-C4A2-3344-902F-E0B22B508689]@0xffffff7f8de45000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[C0E7E4BE-56E8-38D5-93D5-1474FEC39BD6]@0xffffff7f8dad3000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f8d731000
            dependency: org.coolstar.VoodooGPIO(1.1)[5A04B728-4FE7-36C0-B12B-A6D9BECDF1D2]@0xffffff7f90bdf000
         as.acidanthera.voodoo.driver.PS2Trackpad(2.1.3)[256E243E-2561-3784-B9C5-7A529E25E158]@0xffffff7f90c51000->0xffffff7f90c5bfff
            dependency: as.acidanthera.voodoo.driver.PS2Controller(2.1.3)[FC4904F4-E6BE-345D-98C5-EA01141C24C1]@0xffffff7f90bf8000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[C0E7E4BE-56E8-38D5-93D5-1474FEC39BD6]@0xffffff7f8dad3000

BSD process name corresponding to current thread: kernel_task
Boot args: agdpmod=vit9696 darkwake=1 

Mac OS version:
19E287

Kernel version:
Darwin Kernel Version 19.4.0: Wed Mar  4 22:28:40 PST 2020; root:xnu-6153.101.6~15/RELEASE_X86_64
Kernel UUID: AB0AA7EE-3D03-3C21-91AD-5719D79D7AF6
Kernel slide:     0x000000000ca00000
[... 136 lines cut ...]
>!ACredentialManager	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.101.3
|IOSCSIArchitectureModel!F	422.101.1
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	840.3
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

dreamwhite, 2020-04-20:
Yesterday, with great amazement, my system rebooted while in sleep. I don't know if the cause of the panic-reboot is due to VoodooI2C but I've saved the log of "Report Panic".

[image]
[image]
[image]
[report_panic.txt](https://github.com/alexandred/VoodooI2C/files/4501553/report_panic.txt)

Thanks in advance. Hope this bug can be fixed asap :D

Goshin, 2020-04-20:
@gujiangjiang 
<details>
  <summary>comment</summary>

> I tried this but still have problem with wake.
> panic(cpu 0 caller 0xffffff800ce41420): "Spinlock acquisition timed out: lock=0xffffff800d6a12e0, lock owner thread=0xffffff8046de2520, current_thread: [...]

</details>

From your log, deadlock happened in the interrupt handler when trying to read register, maybe related to https://github.com/alexandred/VoodooI2C/issues/279.

---
@dreamwhite 
<details>
  <summary>comment</summary>

> Yesterday, with great amazement, my system rebooted while in sleep. I don't know if the cause of the panic-reboot is due to VoodooI2C but I've saved the log of "Report Panic".
>
> [image]
> [image] [...]
</details>

Can't tell the cause from the given log.

gujiangjiang, 2020-04-20:
@Goshin 
Use released version it shows this log after panic of wake.
[image]
`panic(cpu 1 caller 0xffffff800d04c32a): Kernel trap at 0xffffff7f90e08e19, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff820178e000, CR3: 0x00000000257d1000, CR4: 0x00000000003626e0
RAX: 0xffffff8201135000, RBX: 0xffffff80442072a0, RCX: 0x000000000000006c, RDX: 0x0000000000000001
RSP: 0xffffff820f61be90, RBP: 0xffffff820f61beb0, RSI: 0x000000000000006c, RDI: 0xffffff8043ae9a80
R8:  0x0000000000000000, R9:  0x0000000000989680, R10: 0xffffff800d9f5c40, R11: 0xffffff800d9f5280
R12: 0x00000099ccab88e5, R13: 0xffffffca5b78dcac, R14: 0x000000000000006c, R15: 0xffffff8044183b80
RFL: 0x0000000000010286, RIP: 0xffffff7f90e08e19, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff820113506c, Error code: 0x0000000000000000, Fault CPU: 0x3, PL: 0, VF: 1

Backtrace (CPU 1), Frame : Return Address
0xffffff820f61b8f0 : 0xffffff800cf215cd 
0xffffff820f61b940 : 0xffffff800d05a3c5 
0xffffff820f61b980 : 0xffffff800d04bf7e 
0xffffff820f61b9d0 : 0xffffff800cec7a40 
0xffffff820f61b9f0 : 0xffffff800cf20c97 
0xffffff820f61baf0 : 0xffffff800cf21087 
0xffffff820f61bb40 : 0xffffff800d6c2c7c 
0xffffff820f61bbb0 : 0xffffff800d04c32a 
0xffffff820f61bd30 : 0xffffff800d04c028 
0xffffff820f61bd80 : 0xffffff800cec7a40 
0xffffff820f61bda0 : 0xffffff7f90e08e19 
0xffffff820f61beb0 : 0xffffff7f90e0c314 
0xffffff820f61bee0 : 0xffffff800d62eb6d 
0xffffff820f61bf30 : 0xffffff800d62d43e 
0xffffff820f61bf70 : 0xffffff800d62ca36 
0xffffff820f61bfa0 : 0xffffff800cec713e 
      Kernel Extensions in backtrace:
         com.alexandred.VoodooI2C(2.4)[029EE5E5-D135-31E8-9AC1-6F05AFDF165C]@0xffffff7f90e05000->0xffffff7f90e2cfff
            dependency: com.alexandred.VoodooI2CServices(1)[66104E8E-A22B-3AC4-922E-CA959EFFABB5]@0xffffff7f90971000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9D1FF279-C4A2-3344-902F-E0B22B508689]@0xffffff7f8e045000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[C0E7E4BE-56E8-38D5-93D5-1474FEC39BD6]@0xffffff7f8dcd3000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f8d931000
            dependency: org.coolstar.VoodooGPIO(1.1)[731F2935-AAFA-3470-9CEA-762F7CE72E1D]@0xffffff7f90def000

BSD process name corresponding to current thread: kernel_task
Boot args: agdpmod=vit9696 darkwake=1 

Mac OS version:
19E287

Kernel version:
Darwin Kernel Version 19.4.0: Wed Mar  4 22:28:40 PST 2020; root:xnu-6153.101.6~15/RELEASE_X86_64
Kernel UUID: AB0AA7EE-3D03-3C21-91AD-5719D79D7AF6
Kernel slide:     0x000000000cc00000
Kernel text base: 0xffffff800ce00000
__HIB  text base: 0xffffff800cd00000
System model name: MacBookPro13,3 (Mac-A5C67F76ED83108C)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 660564039537
last loaded kext at 139034861783: @filesystems.smbfs	3.4.2 (addr 0xffffff7f8fce3000, size 454656)
last unloaded kext at 448625221601: ru.joedm.SMCSuperIO	1.1.2 (addr 0xffffff7f90e58000, size 49152)
loaded kexts:
me.kishorprins.VoodooInput	1.0.4
as.acidanthera.BrcmPatchRAM3	2.5.2
as.acidanthera.voodoo.driver.PS2Trackpad	2.1.3
as.acidanthera.voodoo.driver.PS2Keyboard	2.1.3
ru.joedm.SMCSuperIO	1.1.2
org.rehabman.driver.FakePCIID	1.3.15
com.alexandred.VoodooI2CHID	1
com.alexandred.VoodooI2C	2.4
org.coolstar.VoodooGPIO	1.1
as.acidanthera.voodoo.driver.PS2!C	2.1.3
as.lvs1974.RTCMemoryFixup	1.0.4
as.vit9696.SMCProcessor	1.1.2
as.acidanthera.BrcmFirmwareStore	2.5.2
[... 116 lines cut ...]
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.101.3
|IOSCSIArchitectureModel!F	422.101.1
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	840.3
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

`

Goshin, 2020-04-20:
@gujiangjiang Same cause. Are you using in GPIO mode or polling mode?

gujiangjiang, 2020-04-20:
@Goshin 
GPIO 
[image]
[image]

Goshin, 2020-04-20:
@dreamwhite @gujiangjiang Try this one
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4504259/Archive.zip)

gujiangjiang, 2020-04-20:
@Goshin 
Thanks. I will test tonight and report the result in tomorrow.

gujiangjiang, 2020-04-20:
@Goshin
Still kernel panic and reboot after wake.

panic(cpu 0 caller 0xffffff800be4c32a): Kernel trap at 0xffffff7f8fc1898b, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff820122606c, CR3: 0x00000000245e8000, CR4: 0x00000000003626e0
RAX: 0xffffff8201226000, RBX: 0xffffff8043023d40, RCX: 0x000000000000006c, RDX: 0x000000004d3f6400
RSP: 0xffffff81dfcece90, RBP: 0xffffff81dfceceb0, RSI: 0x000000000000006c, RDI: 0xffffff8042900580
R8:  0x0000000000000003, R9:  0x0000000000000000, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0xffffff804325e000, R13: 0x0000000000000000, R14: 0x000000000000006c, R15: 0xffffff80430b0ce0
RFL: 0x0000000000010086, RIP: 0xffffff7f8fc1898b, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff820122606c, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 2, VF: 5

Backtrace (CPU 0), Frame : Return Address
0xffffff81dfcec8f0 : 0xffffff800bd215cd 
0xffffff81dfcec940 : 0xffffff800be5a3c5 
0xffffff81dfcec980 : 0xffffff800be4bf7e 
0xffffff81dfcec9d0 : 0xffffff800bcc7a40 
0xffffff81dfcec9f0 : 0xffffff800bd20c97 
0xffffff81dfcecaf0 : 0xffffff800bd21087 
0xffffff81dfcecb40 : 0xffffff800c4c2c7c 
0xffffff81dfcecbb0 : 0xffffff800be4c32a 
0xffffff81dfcecd30 : 0xffffff800be4c028 
0xffffff81dfcecd80 : 0xffffff800bcc7a40 
0xffffff81dfcecda0 : 0xffffff7f8fc1898b 
0xffffff81dfceceb0 : 0xffffff7f8fc1be99 
0xffffff81dfceced0 : 0xffffff800c451d93 
0xffffff81dfcecf10 : 0xffffff7f8f6b9a9d 
0xffffff81dfcecf50 : 0xffffff7f8e9d888c 
0xffffff81dfcecf60 : 0xffffff7f8e9e1547 
0xffffff81dfcecf80 : 0xffffff800be4b7cd 
0xffffff81dfcecfd0 : 0xffffff800bcc7bed 
0xffffff8035393f40 : 0xffffff800bd466c7 
0xffffff8035393f80 : 0xffffff800bd46888 
0xffffff8035393fa0 : 0xffffff800bcc713e 
      Kernel Extensions in backtrace:
         com.apple.driver.AppleACPIPlatform(6.1)[06B0C2E0-18E0-3E3B-8A46-17BF2A713C34]@0xffffff7f8e9d1000->0xffffff7f8ea6bfff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9D1FF279-C4A2-3344-902F-E0B22B508689]@0xffffff7f8ce45000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f8c731000
            dependency: com.apple.driver.AppleSMC(3.1.9)[601E041E-0A5C-3B6A-A4E6-45ECC7F48A2B]@0xffffff7f8ce57000
         com.apple.driver.AppleAPIC(1.7)[B20F3195-7BF4-3B68-831E-2C180BE8CA78]@0xffffff7f8f6b8000->0xffffff7f8f6bafff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f8c731000
         com.alexandred.VoodooI2C(2.4)[A94B3BB8-4863-36C1-9022-7695E0448D79]@0xffffff7f8fc15000->0xffffff7f8fc3dfff
            dependency: com.alexandred.VoodooI2CServices(1)[E1E325A5-365B-3FF0-96A8-CF330377D2BE]@0xffffff7f8f771000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9D1FF279-C4A2-3344-902F-E0B22B508689]@0xffffff7f8ce45000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[C0E7E4BE-56E8-38D5-93D5-1474FEC39BD6]@0xffffff7f8cad3000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f8c731000
            dependency: org.coolstar.VoodooGPIO(1.1)[151B0B3A-772E-3818-B59B-409AE0EB2401]@0xffffff7f8fbf0000

BSD process name corresponding to current thread: kernel_task
Boot args: agdpmod=vit9696 darkwake=1 

Mac OS version:
19E287

Kernel version:
Darwin Kernel Version 19.4.0: Wed Mar  4 22:28:40 PST 2020; root:xnu-6153.101.6~15/RELEASE_X86_64
Kernel UUID: AB0AA7EE-3D03-3C21-91AD-5719D79D7AF6
Kernel slide:     0x000000000ba00000
Kernel text base: 0xffffff800bc00000
__HIB  text base: 0xffffff800bb00000
System model name: MacBookPro13,3 (Mac-A5C67F76ED83108C)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 8534727004805
last loaded kext at 3613074791013: com.vmware.kext.vmioplug.18.7.0	18.7.0 (addr 0xffffff7f8fc57000, size 36864)
last unloaded kext at 8436147821456: ru.joedm.SMCSuperIO	1.1.2 (addr 0xffffff7f8fc09000, size 49152)
loaded kexts:
me.kishorprins.VoodooInput	1.0.5
as.acidanthera.BrcmPatchRAM3	2.5.2
[... 123 lines cut ...]
>!ACredentialManager	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.101.3
|IOSCSIArchitectureModel!F	422.101.1
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	840.3
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

dreamwhite, 2020-04-21:
Another bug found on the v2.4 version of VoodooI2C. It seems that after putting laptop in sleep the mouse doesn't wake up :'( 

Infos about my machine are here

Goshin, 2020-04-21:
@gujiangjiang same cause. Does the problem exist in polling mode?

No idea what was happening, maybe this would help. 
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4508232/Archive.zip)

dreamwhite, 2020-04-21:
Found that after putting laptop in sleep mode, the trackpad doesn't wake up. This happened with vanilla VoodooI2C 2.4 + VoodooI2CHID. @alexandred what can we do in order to fix it? 
[stopped_trackpad.txt](https://github.com/alexandred/VoodooI2C/files/4512991/stopped_trackpad.txt)

gujiangjiang, 2020-04-22:
@Goshin 
Thanks for your help.
With this Archive.zip it was no problem with sleep and wake.
It works well with no kernel panic.

[image]
Thanks.

suzuke, 2020-04-23:
> Found that after putting laptop in sleep mode, the trackpad doesn't wake up. This happened with vanilla VoodooI2C 2.4 + VoodooI2CHID. @alexandred what can we do in order to fix it?
> [stopped_trackpad.txt](https://github.com/alexandred/VoodooI2C/files/4512991/stopped_trackpad.txt)

I have the same issue too.

LeeBinder, 2020-04-23:
I have the issue of trackpad not responsive after wake from sleep only very very sporadic, maybe after 15 sleep/wake cycles or so. Mojave, VoodooI2C 2.4 + VoodooI2CHID loaded from /L/E.

edd1024, 2020-04-23:
I think (because it is so seldom) it happens here as well. What helps is putting the laptop back to sleep and than wake it up again. So far there are really no issues at this end, I just cannot thank you enough!

LeeBinder, 2020-04-23:
ditto, for me also only a minor issue. Thanks for the hint with one more sleep/ wake cycle - I'll try that (if there is a) next time.

kprinssu, 2020-04-25:
@LeeBinder  can you try the latest version (2.4.1) of VoodooI2C and see if that helps?

LeeBinder, 2020-04-25:
@kprinssu thanks for the pointer, I just did - so far so good, I'll observe. If you don't hear from me, that's a good sign :)

lersy, 2020-04-26:
@kprinssu I do not want to open a new issue but with latest version (2.4.1) laptop does not sleep and wake instantly. if I remove anything related to voodooi2c , it sleeps properly.

dreamwhite, 2020-04-26:
Hi there, today I got my first panic using VoodooI2C 2.4.
Below the panic window log I got:

[log.txt](https://github.com/alexandred/VoodooI2C/files/4536518/log.txt)

I've updated to VoodooI2C 2.4.1 right now and everything seems working:

- trackpad in macOS recovery
- gestures and force touch

I'll test the sleep and edit this comment if everything is working

kprinssu, 2020-04-26:
@lersy Please create a new issue with the usual troubleshooting documentation.

edd1024, 2020-04-27:
Please make sure to run `log show -last 3boot | grep -i "wake reason"` and provide the results. I had the same problem with several laptops, you need to make sure GPRW/UPRW is set up properly which includes ACPI patches and corresponding aml-s. My experience is that I2C wil rather cause a KP than an instant wake. I guess your USB configuration is missing something.

dreamwhite, 2020-04-27:
> `log show -last boot | grep -i "wake reason"`

`2020-04-27 08:07:35.835023+0200 0x633      Default     0x0                  223    0    airportd: systemWokenByWiFi: System Wake Reason not found
2020-04-27 08:17:15.796421+0200 0x104e     Default     0x0                  223    0    airportd: systemWokenByWiFi: System Wake Reason not found
2020-04-27 08:17:15.817545+0200 0x13f2     Default     0x0                  223    0    airportd: systemWokenByWiFi: System Wake Reason not found
2020-04-27 08:17:18.053424+0200 0x13f0     Default     0x0                  223    0    airportd: systemWokenByWiFi: System Wake Reason not found`

If I show instead the last 120 minutes, I see a lot of those "Power Wake reason", but I don't think I have to worry about it, as I always resume from sleep using the Power Button.

`2020-04-27 08:03:46.260835+0200 0x74       Default     0x0                  0      0    kernel: (AppleACPIPlatform) AppleACPIPlatformPower Wake reason: PWRB XDCI GLAN (User)`

> you need to make sure GPRW/UPRW is set up properly which includes ACPI patches and corresponding aml-s

Is there any documentation you can suggest me, in order to setup GPRW/UPRW? 
Here you are my [origin folder](https://github.com/dreamwhite/dell-inspiron-5370-hackintosh/tree/master/CLOVER/ACPI/origin)

> My experience is that I2C will rather cause a KP than an instant wake.

Yep I think so, as the panic reason is related to VoodooI2C

> I guess your USB configuration is missing something.

My USB configuration is properly setup (correctly mapped USB 2.0, USB 3.0 and USB-C 3.0), and I've also added USBX device through SSDT-USBX.aml

[image]

I apologize if USBMap can't be used as debugging method, but it clearly says that my EC and USBX device are properly setup.

[image]

Maybe I don't have to worry about it?

edd1024, 2020-04-27:
I strongly assume there are other sources for your problem - i.e. make sure you have deactivated Wake for Ethernet network access, and also Bluetooth wakes. GLAN is clearly connected to the network. 

Since this is all off-topic here, I suggest you move over to Tonymac and seek for help on a more general platform. Another experiences I have gathered while trying to fix sleep:
* Deactivating AppleALC also leads sometimes to success
* Switching all USB3 ports off show if any problems are rather with USB than with I2C
* Crawl through RehabMan posts and files to find what you need. More or less everything is already there.

Good luck!

kprinssu, 2020-04-27:
@dreamwhite Please use 2.4.1 and the logs are irrevelant to the sleep/wake as it shows bluetooth to be the cause. @edd1024 @dreamwhite please take non-VoodooI2C related discussions out of this PR.

dreamwhite, 2020-04-27:
> @dreamwhite Please use 2.4.1 and the logs are irrevelant to the sleep/wake as it shows bluetooth to be the cause. You should probably apply the USB GPRW patch, but that is irrelevant to VoodooI2C. @edd1024 @dreamwhite please take non-VoodooI2C related discussions out of this PR.

Oke np

gujiangjiang, 2020-04-28:
Update to 2.4.1 but still reboot after wake...

[image]

Now only this one with no problem with wake.
https://github.com/alexandred/VoodooI2C/issues/284#issuecomment-616995064

Goshin, 2020-04-28:
@gujiangjiang The changes above have not been merged yet.

lvs1974, 2020-04-29:
@gujiangjiang, if you observe a kernel panic in macOS, a kernel panic report is useless by default, since all symbols are stripped. To provide a full kernel panic report you should use **keepsyms=1** in boot-args.

dreamwhite, 2020-04-30:
After updating successfully to VoodooI2C 2.4.2, after a sleep/wake cycle, my trackpad stopped working. I attach the output of 
`log show --last 30m | grep VoodooI2C`

[stopped_trackpad.txt](https://github.com/alexandred/VoodooI2C/files/4558301/stopped_trackpad.txt)

kprinssu, 2020-04-30:
@dreamwhite it looks your device is "waking" up from sleep, but perhaps the I2C commands are failing. What is your touchpad, it looks like it's a USB device?

marianopela, 2020-04-30:
> @dreamwhite it looks your device is "waking" up from sleep, but perhaps the I2C commands are failing. What is your touchpad, it looks like it's a USB device?

It is an MSFT0001, like mine. I2CHID is the only satellite

Anyways, I haven't had any problems with sleep/wake lately, on 10th gen

kprinssu, 2020-04-30:
@marianopela did you not also have sleep/wake issues with earlier versions of VoodooI2C?

marianopela, 2020-04-30:
> @marianopela did you not also have sleep/wake issues with earlier versions of VoodooI2C?

I had, and solved earlier in this Issue, [here]. Never had problems of this kind since then. (https://github.com/alexandred/VoodooI2C/issues/284#issuecomment-614470027)

edd1024, 2020-04-30:
Here too, however there are casual KP's, but nohing driving me mad so far. In most cases when I leave the laptop unattended, attempting to go to sleep by itself after 15min or so.

Goshin, 2020-04-30:
Might be addressed in https://github.com/alexandred/VoodooGPIO/pull/7 and https://github.com/alexandred/VoodooI2CHID/pull/33 with more robust toggling of the interrupt.

marianopela, 2020-04-30:
> Might be addressed in [alexandred/VoodooGPIO#7](https://github.com/alexandred/VoodooGPIO/pull/7) and [alexandred/VoodooI2CHID#33](https://github.com/alexandred/VoodooI2CHID/pull/33) with more robust toggling of the interrupt.

In @dreamwhite 's case, he is running in polling mode

dreamwhite, 2020-04-30:
@Goshin here you are my infos about the trackpad:

**DeviceName**: `TPD0`
**IOName**: `MSFT0001`
**Mode**: `Polling`
**APIC Pin**: `0x33`
**I2C Controller**: `VoodooGPIOSunrisePointLP`

You can find my two DSDTs, the origin and the patched version:
- [origin/DSDT.aml](https://github.com/dreamwhite/dell-inspiron-5370-hackintosh/tree/master/CLOVER/ACPI/origin/DSDT.aml)
- [patched/DSDT.aml](https://github.com/dreamwhite/dell-inspiron-5370-hackintosh/tree/master/CLOVER/ACPI/patched/DSDT.aml)

I noticed also that the issue about the trackpad which stops working after a sleep-wake cycle doesn't happen everytimes. I've tested it for 5 times, and only 2 times, the trackpad stopped working. 

Do you think that this is due to the polling mode?
Many thanks for your contribute <3

Goshin, 2020-04-30:
> @Goshin here you are my infos about the trackpad:
>
> **DeviceName**: `TPD0`
> **IOName**: `MSFT0001`
> **Mode**: `Polling`
> **APIC Pin**: `0x33`
> **I2C Controller**: `VoodooGPIOSunrisePointLP`
>
> You can find [...]

I am not sure about the cause. Try to build with the changes mentioned in https://github.com/alexandred/VoodooI2C/issues/284#issuecomment-621912393, as it will completely disable and unregister the polling timer before sleeping, and restore when waking up.

dreamwhite, 2020-11-13:
@all can I close this issue? At the moment I'm still on v.2.4.3 waiting for ticket #388 bug fix

kprinssu, 2020-11-13:
@dreamwhite Please do not use `@ all`. Please close the issue if it's been resolved.

closed 2020-11-13 by kprinssu