VoodooI2C/VoodooI2C #285: RGB Control

ModMike, 2020-04-14:
Can this be used on MB that uses I2C to control RGB lights, such as Aura on Z390?

There is an application that already controls USB but its missing I2C RGB header control.

DO you have a support site where we can ask questions?

ben9923, 2020-04-14:
In theory it should be possible :)
Someone will obviously need to develop a satellite kext for VoodooI2C that will interface with the app.

Check out the Gitter channel linked in the README.

ModMike, 2020-04-14:
Thank you for the reference, I am forwarding to the developer of macAura.

kprinssu, 2020-04-17:
@ModMike feel free to hop onto Gitter, https://gitter.im/alexandred/VoodooI2CDev (developer channel). It's much easier to get in touch with myself and the other VoodooI2C devs :).

I am going to close this issue as it's not really an issue.

closed 2020-04-17 by kprinssu