VoodooI2C/VoodooI2C #286: Gesture Recognition

nikhil19rathore99, 2020-04-15:
Launchpad gesture(3 fingures with thumb) is not working correctly in 2.4 version 
VodooI2C FTE1001 Asus Laptop

kprinssu, 2020-04-16:
Please attach troubleshooting archive.

nikhil19rathore99, 2020-04-17:
[image]
[image]

williambj1, 2020-04-17:
@nikhil19rathore99 

Please delete your screenshots and follow the instructions here, https://voodooi2c.github.io/#Troubleshooting/Troubleshooting

YOU ARE WASTING THE DEVELOPERS' TIME!
https://github.com/williambj1/GenI2C#terms-of-use

nikhil19rathore99, 2020-04-22:
[image]

kprinssu, 2020-04-22:
@nikhil19rathore99 Please **read** https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.

I am closing this issue until you post the troubleshooting archive.

closed 2020-04-24 by nikhil19rathore99