VoodooI2C/VoodooI2C #289: Bug: VoodooI2CHID Enable Dragging buggy with 2.4

Andrw0830, 2020-04-20:
Hello, thanks for the new update as my Trackpad now works in Recovery mode which never did with a previous update! One bug I've noticed though with the new update is with my Dell Inspiron 7567 which uses HID which maps to I2C1, the Enable Dragging option in Accessibility hardly works. If you double-tap the trackpad, 9 times out of 10, the trackpad won't allow you to drag anything in Mac. I am running 10.15.4 with OpenCore 0.5.7.

I am fully up-to-date on the VoodooPS2 and VoodooInput kexts as well which was running alongside VoodooI2C for awhile without causing issues.

Please let me know if you need anything else or if anyone has suggestions on what I can do to fix.

kprinssu, 2020-04-20:
Please see https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and attach a troubleshooting archive containing the above logs.

Andrw0830, 2020-04-20:
> Please see https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and attach a troubleshooting archive containing the above logs.

Thanks man! Attached is my IOReg. Let me know if you need anything else?
[IOReg-2020-04-20.zip](https://github.com/alexandred/VoodooI2C/files/4503855/IOReg-2020-04-20.zip)

Andrw0830, 2020-04-21:
Saw a new commit added for an option to disable force touch emulation as apparently has issues with Drag, but I can't build VoodooI2C to try. Keep getting "ERROR: Failed to bootstrap SDK with code 1!" when running the bootstrap.sh command from the build environment document. I can run the bootstrap directly from VoodooInput Scripts folder which succeeded so I know it's not my build environment, but no matter what I do, I can't satisfy the missing VoodooInput include files to build.

Could someone please build or guide me through so I can test that to confirm if that's what is causing this issue? I am not that good in code so can't figure it out.

Goshin, 2020-04-21:
The commit https://github.com/acidanthera/VoodooInput/commit/66c6f47350916a750dac9e762096af7a21947a7a might fix your issue. You need to build VoodooInput first and put it into `Dependencies/VoodooInput/Debug` and `Dependencies/VoodooInput/Release`.

Andrw0830, 2020-04-21:
> The commit [acidanthera/VoodooInput@66c6f47](https://github.com/acidanthera/VoodooInput/commit/66c6f47350916a750dac9e762096af7a21947a7a) might fix your issue. You need to build VoodooInput first and put it into `Dependencies/VoodooInput/Debug` and `Dependencies/VoodooInput/Release`.

Thanks as that was the missing link. Question, when I built VoodooI2C using Xcode 11.4.1, all the .kexts are separated. For example, VoodooI2CHID.kext is in the "VoodooI2C Satellites\VoodooI2CHID\build" folder same with the VoodooI2CServices. Never seen this behaviour before as usually when the build succeedes, the VoodooI2CServices.kext is in the VoodooI2C.kext\Contents\Plugins. Any idea?

Goshin, 2020-04-21:
`/Users/yourname/Library/Developer/Xcode/DerivedData/VoodooI2C-xxxx/Build/Products/Release/`

You only need VoodooI2C.kext and VoodooInput.kext. Check out the details in the documentation.

Andrw0830, 2020-04-21:
> `/Users/yourname/Library/Developer/Xcode/DerivedData/VoodooI2C-xxxx/Build/Products/Release/`
> 
> You only need VoodooI2C.kext and VoodooI2CHID.kext. Check out the details in the documentation.

Thanks. Sorry as my last response made it seem like VoodooI2C.kext and VoodooI2CHID.kext were in the same kexts, but I realize they are 2 separate kexts you copy.

What my Xcode is doing, when you run the VoodooI2C build, there are build folders in each Dependancy and Satellite folders, not in the output DeprivedData folder like it normally is. In the past and this is including the 2.4 Release download, the VoodooI2C.kext contains VoodooI2CServices.kext, VoodooGPIO.kext and VoodooInput.kext. Mine only had the VoodooInput.kext listed in the Plugins folder within it. I've tried the 'Clean Build folder' option and Xcode's DeprivedData setting is set for the Default location so something is wrong with building this project with my Xcode. What I did to get by this is I copied those kexts and put them in the VoodooI2C.kext/Contents/Plugins manually. Building VoodooInput and VoodooPS2 projects didn't have this issue. I have the newest VoodooI2C, VoodooI2CHID, VoodooInput and VoodooPS2Controller that I've built myself and I still have drag issues.

Again sorry as wasn't quite clear. Hope that clears it up. Do you think the Force Touch Emulation is too blame? I can try disabling it in the Info.plist like the newest commit said?

Andrw0830, 2020-04-21:
Update: I've read the original thread about ForceTouchEmulation which describes my issue, but removing the checkbox in the Trackpad Sys Prefs 'Force Click and Haptic Feedback doesn't do anything, I still have drag and selecting text issues.

Goshin, 2020-04-21:
Not related to that PR if you are using double-tap to drag. I don't fully understand your problem sry, but you can try my build, [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4512602/Archive.zip)
and I personly suggest you try three-finger drag as it can greatly eliminate the click delay.

Andrw0830, 2020-04-21:
> Not related to that PR if you are using double-tap to drag. I don't fully understand your problem sry, but you can try my build, [...]

OK perfect that seemed to work! Not sure what happened with my build, but appreciate it! One last question, since VoodooInput is now included in the VoodooI2C.kext, do I need to still have the VoodooInput.kext separate in OpenCore or can I just point the config.plist to the new location?

kprinssu, 2020-04-21:
No, you do not need VoodooInput as separate kext. Thanks @Goshin. Closing as it's fixed in master, please wait for the next release of VoodooI2C.

closed 2020-04-21 by kprinssu