VoodooI2C/VoodooI2C #290: Stuck in drag mode after releasing physical button

jaromeyer, 2020-04-21:
Same as #263 because I'm pretty sure it is a bug and not intended. I know that there are many open issues about drag but as far as I know, this one is different. On my XPS 9570 (one button trackpad) with VoodooI2C 2.4 and different builds from #259, I drag something by physically pressing the trackpad and when I release the physical button but don't completely lift my finger off, it is stuck in dragging mode. Only after completely lifting off my finger it finishes the drag action. In 2.3 similar things happened, but after releasing the physical button it was only stuck in drag mode for a few seconds. I have just tested it on a real Macbook and confirmed that you only have to release the physical button without lifting of your finger to exit drag mode. Everything else is working fine.

Goshin, 2020-04-21:
You are right. I can replicate the bug. It should be reported to VoodooInput, but I can still give you a quick fix for testing.
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/4510074/Archive.zip)

jaromeyer, 2020-04-21:
@Goshin thank you so much! it completely fixed the problem and force touch emulation is working properly too and can be toggled from the system preferences. may i know what caused that problem? and should i still report it to VoodooInput or can you fix it for them?

closed 2020-04-21 by jaromeyer

Goshin, 2020-04-21:
The fix just resets the finger state in MT2Simulator when releasing the button. Further testing is needed as it might introduce other new bugs. Sure you can open a discussion there and see if it is necessary to fix or we can get a better solution for it.

TheRandMan, 2020-05-14:
I was experiencing the same issue on a SYNA0000 with the Synaptics satellite kext. Replacing VoodooI2C.kext with the attached one has resolved the issue for me as well.

alikazemkhanloo, 2020-05-15:
@Goshin  It worked for me too. I'm using Elan satellite. can you provide a poll request for it?

Goshin, 2020-05-15:
@alikazemkhanloo it is only a temporary fix. Some touch data from real MT2 or MacBook is required to complete the fix. Raise an issue in the VoodooInput project for further discussion.

> I have just tested it on a real Macbook and confirmed that you only have to release the physical button without lifting of your finger to exit drag mode. Everything else is working fine.

Btw, @jaromeyer did you test on a MacBook 2015+ with a force touch trackpad?

alikazemkhanloo, 2020-05-15:
@Goshin VoodooInput has disabled the issues. There is no issue tab there.

Goshin, 2020-05-15:
> @Goshin VoodooInput has disabled the issues. There is no issue tab there.

https://github.com/acidanthera/bugtracker/issues

jaromeyer, 2020-05-15:
> Btw, @jaromeyer did you test on a MacBook 2015+ with a force touch trackpad?
No, it was a late 2012 with the old trackpad

kprinssu, 2020-05-15:
@alikazemkhanloo Do not comment on closed issues unless the original issue also directly effects you. Additionally, Elan has polling support, 2.4.2 has support.