VoodooI2C/VoodooI2C #293: The new version loads and restarts

liyi-debug, 2020-04-25:
panic(cpu 0 caller 0xffffff800574064a): "thread_invoke: preemption_level 1, possible cause: blocking while holding a spinlock, or within interrupt context"@/AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-6153.120.27/osfmk/kern/sched_prim.c:2205
Backtrace (CPU 0), Frame : Return Address
0xffffff81c41c7860 : 0xffffff800571e6dd 
0xffffff81c41c78b0 : 0xffffff8005857b75 
0xffffff81c41c78f0 : 0xffffff800584972e 
0xffffff81c41c7940 : 0xffffff80056c4a40 
0xffffff81c41c7960 : 0xffffff800571dda7 
0xffffff81c41c7a60 : 0xffffff800571e197 
0xffffff81c41c7ab0 : 0xffffff8005ec12dc 
0xffffff81c41c7b20 : 0xffffff800574064a 
0xffffff81c41c7b90 : 0xffffff800573ed3f 
0xffffff81c41c7be0 : 0xffffff800572fb63 
0xffffff81c41c7c20 : 0xffffff8005e2bcbf 
0xffffff81c41c7c60 : 0xffffff8005e2cdd3 
0xffffff81c41c7ca0 : 0xffffff7f88b0a9b6 
0xffffff81c41c7ce0 : 0xffffff7f88b0b4a3 
0xffffff81c41c7d10 : 0xffffff8005e2e6a8 
0xffffff81c41c7d70 : 0xffffff7f88b0b475 
0xffffff81c41c7d90 : 0xffffff7f88b051f6 
0xffffff81c41c7dc0 : 0xffffff8005e2e8f5 
0xffffff81c41c7e20 : 0xffffff7f88b0519a 
0xffffff81c41c7e40 : 0xffffff7f88b58dc2 
0xffffff81c41c7e70 : 0xffffff8005e2e8f5 
0xffffff81c41c7ed0 : 0xffffff8005e503f3 
0xffffff81c41c7f10 : 0xffffff7f884cfa9d 
0xffffff81c41c7f50 : 0xffffff7f873e388c 
0xffffff81c41c7f60 : 0xffffff7f873ec547 
0xffffff81c41c7f80 : 0xffffff8005848f7d 
0xffffff81c41c7fd0 : 0xffffff80056c4bed 
0xffffff81c42d38d0 : 0xffffff8005814d18 
0xffffff81c42d38f0 : 0xffffff8005736249 
0xffffff81c42d39e0 : 0xffffff8005df2f20 
0xffffff81c42d3b90 : 0xffffff8005e77782 
0xffffff81c42d3c10 : 0xffffff8005c2bfc1 
0xffffff81c42d3e70 : 0xffffff8005749d9f 
0xffffff81c42d3fa0 : 0xffffff80056c413e 
      Kernel Extensions in backtrace:
         com.apple.driver.AppleACPIPlatform(6.1)[41A3E552-1E29-3CA6-A6A3-263EDEC92A1B]@0xffffff7f873dc000->0xffffff7f87476fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9483DF76-7217-3152-9908-49526931D984]@0xffffff7f864a2000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[95D2EB08-9875-3B09-AFB5-511D7EDFB574]@0xffffff7f864b4000
            dependency: com.apple.driver.AppleSMC(3.1.9)[E93667E5-4F4F-313E-B2C2-C0DEFEBB0963]@0xffffff7f864f3000
         com.apple.driver.AppleAPIC(1.7)[7C697FBC-5DEE-34CD-B29E-D9AB39204AB9]@0xffffff7f884ce000->0xffffff7f884d0fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[95D2EB08-9875-3B09-AFB5-511D7EDFB574]@0xffffff7f864b4000
         com.alexandred.VoodooI2C(2.4.1)[7FD6B011-3B4E-3310-84BA-170C03186495]@0xffffff7f88b03000->0xffffff7f88b2ffff
            dependency: com.alexandred.VoodooI2CServices(1)[DBFA5EBE-2783-30ED-8E39-5829C42D6832]@0xffffff7f88ad8000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9483DF76-7217-3152-9908-49526931D984]@0xffffff7f864a2000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[7C4E3493-B1E5-39CC-9455-C9D2B366B13C]@0xffffff7f860a6000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[95D2EB08-9875-3B09-AFB5-511D7EDFB574]@0xffffff7f864b4000
            dependency: org.coolstar.VoodooGPIO(1.1)[31865491-FC22-3BA0-B115-BBE17DDB021A]@0xffffff7f88ae3000
         com.alexandred.VoodooI2CHID(1.0)[A0A110F9-6B36-3E64-A4A3-1EB5CD376161]@0xffffff7f88b52000->0xffffff7f88b64fff
            dependency: com.alexandred.VoodooI2C(2.4.1)[7FD6B011-3B4E-3310-84BA-170C03186495]@0xffffff7f88b03000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9483DF76-7217-3152-9908-49526931D984]@0xffffff7f864a2000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[7C4E3493-B1E5-39CC-9455-C9D2B366B13C]@0xffffff7f860a6000

BSD process name corresponding to current thread: kernel_task
Boot args: -v darkwake=8

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 19.5.0: Sun Apr  5 21:30:12 PDT 2020; root:xnu-6153.120.27~16/RELEASE_X86_64
Kernel UUID: 8DB5A43B-B2C4-35AD-B415-12A494014D41
Kernel slide:     0x0000000005400000
Kernel text base: 0xffffff8005600000
__HIB  text base: 0xffffff8005500000
System model name: MacBookPro15,3 (Mac-1E7E29AD0135F9BC)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization

System uptime in nanoseconds: 2911020659
last loaded kext at 1920681125: as.acidanthera.voodoo.driver.PS2Keyboard	2.1.3 (addr 0xffffff7f88b9f000, size 45056)
loaded kexts:
as.acidanthera.voodoo.driver.PS2Keyboard	2.1.3
com.alexandred.VoodooI2CHID	1
com.insanelymac.RealtekRTL8111	2.2.2
ru.joedm.SMCSuperIO	1.1.2
com.alexandred.VoodooI2C	2.4.1
as.acidanthera.voodoo.driver.PS2!C	2.1.3
org.coolstar.VoodooGPIO	1.1
com.hieplpvip.AsusSMC	1.2.2
as.vit9696.SMCProcessor	1.1.2
com.alexandred.VoodooI2CServices	1
xyz.racermaster.NoTouchID	1.0.3
as.vit9696.!AALC	1.4.8
ru.usrsse2.SMCBatteryManager	1
as.vit9696.VirtualSMC	1.1.2
as.vit9696.WhateverGreen	1.3.8
as.vit9696.Lilu	1.4.3
>!AFileSystemDriver	3.0.1
>!AVirtIO	1.0
@filesystems.hfs.kext	522.100.5
@!AFSCompression.!AFSCompressionTypeDataless	1.0.0d1
@BootCache	40
@!AFSCompression.!AFSCompressionTypeZlib	1.0.0
@filesystems.apfs	1412.120.1
>!AAHCIPort	341.0.2
>!ASmartBatteryManager	161.0.0
>!AACPIButtons	6.1
>!ARTC	2.0
>!AACPIEC	6.1
>!AHPET	1.8
>!ASMBIOS	2.1
>!AAPIC	1.7
$!AImage4	1
@nke.applicationfirewall	303
$TMSafetyNet	8
@!ASystemPolicy	2.0.0
|EndpointSecurity	1
>!UMergeNub	900.4.2
|IOAudio!F	300.2
@vecLib.kext	1.2.0
|IOSerial!F	11
|IOSurface	269.11
@filesystems.hfs.encodings.kext	1
>!AXsanScheme	3
|IONVMe!F	2.1.0
|IOAHCIBlock!S	316.100.5
|IOAHCI!F	290.0.1
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>usb.!UHostPacketFilter	1.0
|IOUSB!F	900.4.2
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
|IOHID!F	2.0.0
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
|IOSMBus!F	1.1
>DiskImages	493.0.0
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!AKeyStore	2
>!UTDM	489.120.1
|IOSCSIBlockCommandsDevice	422.120.2
>!ACredentialManager	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.120.3
|IOSCSIArchitectureModel!F	422.120.2
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	840.3
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

konrad11901, 2020-04-26:
Same issue here. VoodooI2C 2.4.0 worked (mostly) fine, but 2.4.1 makes my system unbootable. In some **very** rare cases, macOS will boot, but any attempt to use the touchscreen will cause a reboot.

Using Dell Inspiron 7373 (touchscreen - ELAN2097, touchpad - DELL07EB) - OpenCore files can be found [here](https://github.com/konrad11901/Inspiron7373-macOS).

kprinssu, 2020-04-26:
This seems to be an issue with @Goshin's changes in https://github.com/alexandred/VoodooI2CHID/commit/9d7a966f394e8f3b7bc26efdeebef38bd37a77c8.

The issue is that the control flow is accessing a lock somewhere within an interrupt context. Let me build a kext with the above changes reverted.

Goshin, 2020-04-26:
The backtrace shows the thread is trying to get blocked within the direct interrupt context.
I might know the cause of the panic. Are you using the APIC interrupt?

konrad11901, 2020-04-26:
> The backtrace shows the thread is trying to get blocked within the direct interrupt context.
> I might know the cause of the panic. Are you using the APIC interrupt?

Yes, in my case, touchscreen is handled using APIC interrupts (touchpad is using the GPIO).

Goshin, 2020-04-26:
So it is caused by the direct interrupt introduced in https://github.com/alexandred/VoodooI2CHID/commit/9d7a966f394e8f3b7bc26efdeebef38bd37a77c8.

I didn't think of the case of the APIC interrupt controller. I had to use "direct interrupt" in https://github.com/alexandred/VoodooI2CHID/commit/9d7a966f394e8f3b7bc26efdeebef38bd37a77c8 because currently VoodooGPIO does not report the interrupt type correctly. It is okay because the interrupt handler will actually run in VoodooGPIO's workloop thread, which is in the indirect interrupt context, if you are using GPIO.

will fix it very soon.

Goshin, 2020-04-26:
@konrad11901  Plz test this. [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4536349/Archive.zip)

Related changes:
https://github.com/Goshin/VoodooGPIO/commit/f7431bc0678606cf9813fb800db2255e6ccc5247
https://github.com/Goshin/VoodooI2CHID/commit/e7413a2632c96d8e87b63fb844edc7dbee8fdebf

konrad11901, 2020-04-26:
@Goshin Problem solved - with your version macOS boots correctly and the touchscreen works as in VoodooI2C 2.4.0. Thanks a lot! :)

kprinssu, 2020-04-26:
@Goshin feel free to create a PR, will merge.

Goshin, 2020-04-26:
> @Goshin feel free to create a PR, will merge.

https://github.com/alexandred/VoodooGPIO/pull/6
https://github.com/alexandred/VoodooI2CHID/pull/32

closed 2020-04-27 by kprinssu