VoodooI2C/VoodooI2C #296: New version causes problems

nikhil19rathore99, 2020-04-29:
when i updated to 2.4.2 then cursor shows ridiculous behaviour cusor flickering, not working properly, cursor jumping from a point to point without touching the trackpad.
trackpad: FTE1001
Asus X556UQK
kext used: vodooi2c.kext, vodooi2cFTE.kext

ben9923, 2020-04-29:
You'll need to attach a troubleshooting archive.
How does 2.4.1 behave on your machine?

nikhil19rathore99, 2020-04-29:
how do i get turbo-shooting files. yeah it was works fine but launchpad gestures works only sometimes not everytime and my cursor speed is too fast even at 1 step increased in initial value.

nikhil19rathore99, 2020-04-29:
[turboshooting.zip](https://github.com/alexandred/VoodooI2C/files/4552122/turboshooting.zip)

kprinssu, 2020-04-29:
Please attach *all* logs, read https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. It is very hard to understand what's going on from just the IORegistry file.

kecinzer, 2020-04-29:
For me, 2.4.2 is finally fully functional version. Touchpad works for me great with all multifinger gestures like on real MacBook. 👍

nikhil19rathore99, 2020-04-30:
[logs.zip](https://github.com/alexandred/VoodooI2C/files/4556208/logs.zip)

nikhil19rathore99, 2020-04-30:
I hope this is all from my side please help if possible ,Thanks for the concern

nikhil19rathore99, 2020-04-30:
[image]
this is left

nikhil19rathore99, 2020-04-30:
[log.zip](https://github.com/alexandred/VoodooI2C/files/4556233/log.zip)

kprinssu, 2020-04-30:
You are getting multiple `VoodooI2CFTETouchpadDriver::FTE1001 Invalid report` log events. Are you certain that your device is a FTE based device?

nikhil19rathore99, 2020-04-30:
> You are getting multiple `VoodooI2CFTETouchpadDriver::FTE1001 Invalid report` log events. Are you certain that your device is a FTE based device?

In windows i checked the device properties it shows deviceDescription:  \dev\FTE1001

kprinssu, 2020-04-30:
That is not a good of way checking if you have a FTE device or not. Check which drivers are loading on Windows and Linux. Preferrably Linux as it is verbose and can easily tell you which kernel module has loaded for your trackpad.

nikhil19rathore99, 2020-04-30:
> That is not a good of way checking if you have a FTE device or not. Check which drivers are loading on Windows and Linux. Preferrably Linux as it is verbose and can easily tell you which kernel module has loaded for your trackpad.

can i send the window trackpad device properties screenshot to show ??

kprinssu, 2020-04-30:
Yes, I don't see why you shouldn't?

nikhil19rathore99, 2020-04-30:
> Yes, I don't see why you shouldn't?
[image]

[image]

kprinssu, 2020-04-30:
What is your laptop model?

nikhil19rathore99, 2020-05-01:
> What is your laptop model?

Asus X556UQK, I've mentioned a txt file about laptop above

nikhil19rathore99, 2020-05-01:
> [logs.zip](https://github.com/alexandred/VoodooI2C/files/4556208/logs.zip)

this log file contains system file,
Asus X556UQK,
Processor: i5-7200U, KabyLake 7th Gen,
GPU: intel hd620

problems:
1. Cursor moves too fast even at slow speed, 
2. 3 fingures with thumg laucnhpad gesture not working properly,
3.Cursor movement is not so smooth like windows

Please consider this problems, Thanks for your concern in this matter!!

kprinssu, 2020-05-01:
I want you to double check if you are using the correct satellite kext. As I suspect that you do not have a FTE based device rather a Elan based one. An easy way to check is to boot up your device using a live Linux USB and checking which kernel modules load.

Additionally, here's an archived repository for your device, https://github.com/systemfaliure/ASUS-X556UQK-hackintosh/tree/master/CLOVER/kexts/10.13. Notice that this user is using VoodooI2CElan. 

I suggest you read https://voodooi2c.github.io/#Satellite%20Kexts/Satellite%20Kexts.

nikhil19rathore99, 2020-05-01:
> I want you to double check if you are using the correct satellite kext. As I suspect that you do not have a FTE based device rather a Elan based one. An easy way to check is to boot up your device using a live Linux USB and checking which kernel modules load.
> 
> Additionally, here's an archived repository for your device, https://github.com/systemfaliure/ASUS-X556UQK-hackintosh/tree/master/CLOVER/kexts/10.13. Notice that this user is using VoodooI2CElan. I also have yet to come across an Asus laptop that does not use a Elan device.
> 
> I suggest you read https://voodooi2c.github.io/#Satellite%20Kexts/Satellite%20Kexts.

which command i use to check the trackpad

nikhil19rathore99, 2020-05-01:
I: Bus=0018 Vendor=0b05 Product=0101 Version=0100
N: Name="Asus TouchPad"
P: Phys=i2c-FTE1001:00
S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-5/i2c-FTE1001:00/0018:0B05:0101.0001/input/input7
U: Uniq=
H: Handlers=mouse0 event7 
B: PROP=5
B: EV=b
B: KEY=e520 10000 0 0 0 0
B: ABS=661800011000003

nikhil19rathore99, 2020-05-01:
[image]

nikhil19rathore99, 2020-05-01:
[image]

edd1024, 2020-05-01:
`lsmod`
There are also i2c tools, check this out: https://it-explain.com/how-to-scan-i2c-bus-in-linux/

nikhil19rathore99, 2020-05-02:
> `lsmod`
> There are also i2c tools, check this out: https://it-explain.com/how-to-scan-i2c-bus-in-linux/

i install that i2c tools but how to scan this i mean i provide the xinput output above which confirms that my trackpad is i2c-FTE:1001

> [image]

nikhil19rathore99, 2020-05-02:
> I want you to double check if you are using the correct satellite kext. As I suspect that you do not have a FTE based device rather a Elan based one. An easy way to check is to boot up your device [...]

and talkin about his specs it was old mine was new 2017 model, so there is hardware difference

nikhil19rathore99, 2020-05-02:
> I want you to double check if you are using the correct satellite kext. As I suspect that you do not have a FTE based device rather a Elan based one. An easy way to check is to boot up your device [...]

It's been 3 days since i posted the issue yet it's not been resolved althoughi provided all the information needeed to fix but the problem is not solved yet, Please help!!

Goshin, 2020-05-02:
Seems the changes of VoodooI2CHID in the new version have not yet been ported to the satellite you are using. You may want to raise an issue in the VoodooI2CFTE project.

kprinssu, 2020-05-02:
> It's been 3 days since i posted the issue yet it's not been resolved althoughi provided all the information needeed to fix but the problem is not solved yet, Please help!!

@nikhil19rathore99 you do realize that myself and the other contributors and maintainers are doing this for free right? We are taking time out of busy schedules to help you, for free to add.

As Goshin mentioned mentioned the changes have to be ported, as seen here https://github.com/kprinssu/VoodooI2CELAN/commit/6897677af23c1e647760ebfb4e23af22883fc995.

closed 2020-05-02 by kprinssu

nikhil19rathore99, 2020-05-02:
> > > I want you to double check if you are using the correct satellite kext. As I suspect that you do not have a FTE based device rather a Elan based one. An easy way to check is to boot up your device using a live Linux USB and checking which kernel modules load.
> > > Additionally, here's an archived repository for your device, https://github.com/systemfaliure/ASUS-X556UQK-hackintosh/tree/master/CLOVER/kexts/10.13. Notice that this user is using VoodooI2CElan. I also have yet to come across an Asus laptop that does not use a Elan device.
> > > I suggest you read https://voodooi2c.github.io/#Satellite%20Kexts/Satellite%20Kexts.
> > 
> > 
> > It's been 3 days since i posted the issue yet it's not been resolved althoughi provided all the information needeed to fix but the problem is not solved yet, Please help!!
> 
> @nikhil19rathore99 you do realize that myself and the other contributors and maintainers are doing this for free right?
> 
> As Goshin mentioned mentioned the changes have to be ported, as seen here [kprinssu/VoodooI2CELAN@6897677](https://github.com/kprinssu/VoodooI2CELAN/commit/6897677af23c1e647760ebfb4e23af22883fc995). Will be fixed, until then please wait or you can try attempting the port yourself.

So Sorry,  Thanks for your support, I think I am new in GitHub, please accept my apology, Sorry and again Thanks for the reply!!