VoodooI2C/VoodooI2C #297: touchpad slower As the usage time increases

packyan, 2020-04-29:
As the usage time increases, the touchpad will have a significant delay. The touchpad is very sensitive within a few minutes after booting, and you can’t feel any delay, but after using it for a period of time, you will feel a significant delay. The kernel task occupancy rate reaches 30-40%. After waiting for tens of seconds, the touchpad can return to the sensitive state.

ben9923, 2020-04-29:
You'll need to provide more details.
Which hardware, kext version (make sure it's latest), when did it start, which satellites are you using for each device.

Attach a troubleshooting archive (ACPI, IOReg, kernel log).

packyan, 2020-04-30:
> You'll need to provide more details.
> Which hardware, kext version (make sure it's latest), when did it start, which satellites are you using for each device.
> 
> Attach a troubleshooting archive (ACPI, IOReg, kernel log).

Already update details, plz check.

marianopela, 2020-04-30:
> using latest 2.4.2 kext.(only copy VoodooI2C.kext and VoodooI2CHID.kext to efi, it's that wrong? Because I saw my original efi have these two files )

VoodooInput.kext is now bundled with VoodooI2C. If you are on Clover you can just delete VoodooInput.kext. If you are on OpenCore well, you can replace VoodooInput with the one insieme of VoodooI2C/Plugins or just delete it and redefine the path in your config (in Kernel section) like so
[image]

packyan, 2020-04-30:
> > 使用最新的2.4.2 kext。（仅将VoodooI2C.kext和VoodooI2CHID.kext复制到efi，这是错误的吗？因为我看到我的原始efi有这两个文件）
> 
> VoodooInput.kext现在与VoodooI2C捆绑在一起。如果您在Clover上，则可以删除VoodooInput.kext。如果您使用的是OpenCore，则可以用VoodooI2C / Plugins内部的一个替换VoodooInput，也可以删除它，然后在配置（在内核部分）中重新定义路径，如下所示
> 
> [image]

I'm using Clover, and I have done what your said, then it's working fine, no more lagging, but system log are still tons of GPIO 28 Error. This Error can be fixed? Thanks for your reply.

Default 0x0 0 0 kernel: (kernel) VoodooGPIOCannonLakeLP::Failed getting hardware pin for GPIO pin 28

marianopela, 2020-04-30:
system log are still tons of GPIO 28 Error. This Error can be fixed? Thanks for your reply.
> 
> Default 0x0 0 0 kernel: (kernel) VoodooGPIOCannonLakeLP::Failed getting hardware pin for GPIO pin 28

Seems like 28 is not the right pin for you
Please read [here](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning) carefully

packyan, 2020-04-30:
> 系统日志仍然是GPIO 28错误吨。这个错误可以解决吗？感谢您的回复。
> 
> > 默认值0x0 0 0内核：（内核）VoodooGPIOCannonLakeLP :::无法获取GPIO引脚28的硬件引脚
> 
> 似乎28不是适合您的引脚
> 请仔细阅读[此处](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning)

I am a newbie in hackintosh, so when I saw that page, I gave up.

closed 2020-04-30 by packyan