VoodooI2C/VoodooI2C #299: Weird behavior with hard click

hieplpvip, 2020-05-04:
Hi. I get weird behavior on v2.4.2 when hard clicking (pressing the trackpad). It works find on v2.4.0

Expected behavior: Only one click is triggered

Actual behavior: Multiple clicks are triggered in a very short time. The number of clicks is not fixed. This breaks dragging and makes it really annoyed to use the trackpad.

Goshin, 2020-05-04:
Have you tried to disable `force click and haptic feedback` in trackpad pref?

hieplpvip, 2020-05-04:
@Goshin Yes, I have.

I forgot to mention that v2.4.2 worked fine for the last few days, then suddenly that problem appears after I sleep my laptop. I tried rebooting but that didn't fix it, so I reverted back to v2.4.0.

Btw, do you have any idea why trackpad stops working randomly in polling mode?

Goshin, 2020-05-04:
I don't know for sure, but maybe you can try this out: https://github.com/alexandred/VoodooI2CHID/pull/33, to disable interrupt source (timer or GPIO interrupt) completely before sleep.

nikhil19rathore99, 2020-05-05:
> I don't know for sure, but maybe you can try this out: [alexandred/VoodooI2CHID#33](https://github.com/alexandred/VoodooI2CHID/pull/33), to disable interrupt source (timer or GPIO interrupt) completely before sleep.

Can u help me with the issue of vodooi2cFTE invalid reports(0) log errors every time on touches the trackpad??

Goshin, 2020-05-05:
> Can u help me with the issue of vodooi2cFTE invalid reports(0) log errors every time on touches the trackpad??

I am afraid I cannot, it is off topic but I am not familiar with the device and quite busy these days. Why not use the working version before someone can fix it?

nikhil19rathore99, 2020-05-05:
> I am afraid I cannot, it is off topic but I am not familiar with the device and quite busy these days. Why not use the working version before someone can fix it?

sorry to disturb but this problem is from the begining any of the version can't fix this problem

Goshin, 2020-05-06:
> sorry to disturb but this problem is from the begining any of the version can't fix this problem

It is confusing that you said the new version causes the problem, so it could be a device-specific compatibility issue. Consider reporting it to VoodooI2CFTE, because it is off-topic here.

kprinssu, 2020-05-08:
@nikhil19rathore99 Please consider this a warning. Please do not spam other Github issues asking for help as it is not constructive and quite derailing for the conversation at hand.

kprinssu, 2020-05-22:
@hieplpvip does this issue still occur?

hieplpvip, 2020-05-23:
@kprinssu Currently everything works fine. I've been using v2.4.2 for the last few weeks. Sometimes the problem appears out of nowhere, then it disappears. I don't change anything.

I will close the issue. Will reopen if it happens again.

closed 2020-05-23 by hieplpvip