VoodooI2C/VoodooI2C #300: Which VoodooInput.kext I should use?

kecinzer, 2020-05-04:
I just looking at VoodooInput.kext bundled with VoodooI2C and VoodooInput.kext bundled with VoodooPS2 and both is 1.0.5 version, but VoodooI2C one is bigger (94 vs 74 kB).
Which version is better for use?

nikhil19rathore99, 2020-05-05:
> I just looking at VoodooInput.kext bundled with VoodooI2C and VoodooInput.kext bundled with VoodooPS2 and both is 1.0.5 version, but VoodooI2C one is bigger (94 vs 74 kB).
> Which version is better for use?

Use vodooinput.kext bundled with vodooi2c.kext.

ben9923, 2020-05-08:
If you only use VoodooInput with one kext (i.e your only have a PS2 *keyboard*), delete/disable VoodooInput on the other.

Otherwise, either should work (Unless there's a breaking change in VoodooInput). Use the more recent one if there's a difference, I guess.

closed 2020-05-08 by ben9923