VoodooI2C/VoodooI2C #301: Asking about additing controller support

m0d16l14n1, 2020-05-04:
Hello! I am trying to make my touchpad working, but it`s connected to AMI0010 controller (because it`s AMD CPU based laptop). Touchpad is ELAN

Is it possible to add support for that kind of controller? I can give all information or anything you need

kprinssu, 2020-05-08:
@m0d16l14n1 Please read https://voodooi2c.github.io/#Installation/Installation and https://voodooi2c.github.io/#Satellite%20Kexts/Satellite%20Kexts. I am closing this issue as it is not an issue. Please use Gitter for community support.

closed 2020-05-08 by kprinssu

kprinssu, 2020-05-08:
@m0d16l14n1 I am very sorry for closing your issue earlier. No AMD controllers are not supported. 

You or someone else will have to extend VoodooGPIO to support AMD's chipsets.