VoodooI2C/VoodooI2C #302: [OC 0.5.8] Dell Inspiron 5370 problems with trackpad activation

dreamwhite, 2020-05-06:
Hi there, I'm planning to migrate to OpenCore 0.5.8, from my actual working Clover r5115 installation. 
I'm facing some difficulties activating the trackpad using my custom DSDT.aml
My infos about trackpad are below described:
- VoodooI2C + VoodooI2CHID (PNP0C50, matching property) 
- Device name: TPD0
- APIC Pin: 0x33
- IOName: MSFT0001
- Mode: Polling

I've noticed that if I apply the same patch, as depicted below (which is pretty removing the following lines):
[image]

my system hangs on "Previous shutdown cause: "
Is this a malfunction of VoodooI2C or am I doing something wrong? Do I need any SSDT such as SSDT-XOSI?

Many thanks
dreamwhite
P.S.

I've tried removing the custom DSDT and the system boots, but if I go checkin the IOName, it says eNUB (tf?)

ben9923, 2020-05-08:
Seems like a bad ACPI configuration.

Use something similar to SSDT-AWAC in OpenCore example ACPI tables.
Set `SDM0` to `Zero` inside when `_OSI` is `Darwin`. That will enable GPIO interrupts.

For `XOSI`, I'd use the familiar `XOSI` method, but return the fake OS only when `_OSI` returns `Darwin`, otherwise return the original `_OSI` value to not break anything.
OpenCore team might have a better suggestion for the OS patch.

dreamwhite, 2020-05-08:
Uhm, It seems that adding simply SSDT-XOSI.aml, enables the trackpad. I'm not using any custom DSDT as I don't need it. Do you think it's better using GPIO Interrupts?

kprinssu, 2020-05-08:
@dreamwhite Interrupts are generally better than polling mode. It results in smoother gesture and mouse movement when using interrupts. However, you will need to figure the GPIO pin following the pinning guide (which may be difficult for certain devices and hardware configurations).

dreamwhite, 2020-05-08:
Uh okay, I've found that my GPIO pin is 0x1B (Interrupt Specifier: 0x33, SunrisePointLP controller). Do I have to test with SMD0 set to Zero?

kprinssu, 2020-05-08:
You probably want to do something similar (this is pseudocode):

```
if XOSI == 'Darwin':
  Return (ConcatenateResTemplate (SBFB, SBFG))
else:
  original DSDT/SSDT function code
```

The above is only applicable if you are running multiple OSes, you can simply modify the _CRS function to return `Return (ConcatenateResTemplate (SBFB, SBFG))`. That wis what @ben9923 is suggesting you do.

dreamwhite, 2020-05-08:
Okay, many thanks ❤️

kprinssu, 2020-05-08:
No problem, please ping either me or Ben if you require any assistance.

dreamwhite, 2020-05-09:
Dear @kprinssu, do I need SSCN and FMCN? In my logs I can't see them but the trackpad is working. Many thanks <3

ben9923, 2020-05-09:
If it's working you _probably_ don't need them.

kprinssu, 2020-05-14:
@dreamwhite do you have any further issues? Otherwise, please close this issue.

dreamwhite, 2020-05-14:
Well ATM nope. I can close this issue

closed 2020-05-14 by dreamwhite