VoodooI2C/VoodooI2C #303: Optimization: Fixing memory leak and using DMA to reduce CPU usage

tiger511, 2020-05-07:
After Goshin's patch ,things have improved a lot.Particularly, kernel_task CPU usage.But CPU Usage  can be reduced further by avoiding allocating buffer in interrupt handler or similar function.Consider the following code Line 151-154 @  VoodooI2CHIDDevice.cpp :
```C++
void VoodooI2CHIDDevice::getInputReport() {
    IOBufferMemoryDescriptor* buffer;
    IOReturn ret;
    unsigned char* report = (unsigned char *)IOMalloc(hid_descriptor.wMaxInputLength);
```
getInputReport gets called when we use the trackpad at around 20-50/second times.Since the size of the memory needed for this allocation is fixed,we can allocate memory in  `VoodooI2CHIDDevice::handleStart` and reuse the same buffer.See solution below.

Similiarly Line 174 @ VoodooI2CHIDDevice.cpp :

```C++
void VoodooI2CHIDDevice::getInputReport() {
    ...
    buffer = IOBufferMemoryDescriptor::inTaskWithOptions(kernel_task, 0, return_size);
    ...
}
```
This `IOBufferMemoryDescriptor::inTaskWithOptions`  should not be called from interrupt context according to [https://developer.apple.com/documentation/kernel/iobuffermemorydescriptor/1813824-intaskwithoptions?language=objc](url) which states 

>  This function may block and so should not be called from interrupt level or while a simple lock is held.

Memory leak:

```C++
void VoodooI2CHIDDevice::getInputReport() {
   ....
   int return_size = report[0] | report[1] << 8;

    if (!return_size) {
        // IOLog("%s::%s Device sent a 0-length report\n", getName(), name);
        command_gate->commandWakeup(&reset_event);
        goto exit;
    }

    if (!ready_for_input)
        goto exit;

    if (return_size > hid_descriptor.wMaxInputLength) {
        // IOLog("%s: Incomplete report %d/%d\n", getName(), hid_descriptor.wMaxInputLength, return_size);
        goto exit;
    }

.....
    exit:
        return;
}
```
We are not freeing up memory when the return_size is invalid or  report is incomplete. Similiarly i've found quite a few places where memory is either allocated poorly or not freed up properly when an error encounters. We should not allocate memory in the interrupt workloop in the first place

**Solution:**

Allocate memory when the handler start:

```C++
 bool VoodooI2CHIDDevice::handleStart(IOService* provider) {
    if (!IOHIDDevice::handleStart(provider)) {
        return false;
    }
    report = (unsigned char *)IOMalloc(hid_descriptor.wMaxInputLength); 
    buffer = IOBufferMemoryDescriptor::inTaskWithOptions(kernel_task, 0, hid_descriptor.wMaxInputLength);
```

Reuse this buffers in `VoodooI2CHIDDevice::getInputReport()` and other functions.We can safely use the same  buffer throughout this class because it works in workloop and they are syncronized.we can release allocating memory in `VoodooI2CHIDDevice::releaseResources`

```C++
void VoodooI2CHIDDevice::releaseResources() {
    buffer->release();
    buffer=NULL;
    IOFree(report, hid_descriptor.wMaxInputLength);
    ....
```

Further Optimization:

We can further optimize CPU usage  from 5-6% to 2-3%( According to my testing) by avoiding copying report data Line 175 :

```C++
void VoodooI2CHIDDevice::getInputReport() {
    ...
    buffer = IOBufferMemoryDescriptor::inTaskWithOptions(kernel_task, 0, return_size);
    buffer->writeBytes(0, report + 2, return_size - 2);
    ...
```

Instead we can do this:
```C++
void VoodooI2CHIDDevice::getInputReport() {    
    unsigned char* report=(unsigned char *)(buffer->getBytesNoCopy(););
    api->readI2C(report, hid_descriptor.wMaxInputLength);
....
   ret = handleReport(buffer +2, kIOHIDReportTypeInput); // <<< This does not work.because buffer is not pointer.
```

We should be able to do something like this but I cant seem to modify IOBufferMemoryDescriptor buffer pointer to ignore the first two bytes.If we don't ignore the first two bytes, `handleReport()` can't seem to recognize our buffer.Maybe we can avoid writing the first two bytes of the report in the first place by modifying "VoodooI2CControllerDriver" .Any ideas ?

Goshin, 2020-05-08:
> This `IOBufferMemoryDescriptor::inTaskWithOptions` should not be called from interrupt context according to [...]

In fact, the handler here is in the indirect interrupt context, which is handled by the `IOInterruptEventSource` and is called in the workloop thread, so not to worry it is fine to allocate memory and buffer here.

---

>Further Optimization:
>
>We can further optimize CPU usage from 5-6% to 2-3%( According to my testing) by avoiding copying report data Line 175 :
>
>We should be able to do something like this but I cant seem to modify IOBufferMemoryDescriptor buffer pointer to ignore the first two bytes.If we don't ignore the first two bytes, handleReport() can't seem to recognize our buffer.Maybe we can avoid writing the first two bytes of the report in the first place by modifying "VoodooI2CControllerDriver" or offset the first 2 bytes somehow.Any ideas ?

Would the 3% CPU be consumed by the `handleReport()` here, rather than the `buffer->writeBytes()`?
I can see a similar improvement with the touchpad disabled (in which `handleInterruptReport()` returns immediately).

I agree with the idea to use pre-allocated buffer variables in the interrupt handler. Though I don't think it can help a lot reduce the CPU load, it might be helpful for reducing latency and potential leaks.

Please correct me.

tiger511, 2020-05-08:
> Would the 3% CPU be consumed by the `handleReport()` here, rather than the `buffer->writeBytes()`?
> I can see a similar improvement with the touchpad disabled (in which `handleInterruptReport()` returns immediately).

Did you disable touchpad with hotkey? There is another problem related to this.I2C service does not pause/detach when trackpad is disabled.If i put a finger on my trackpad,cpu usage spikes meaning trackpad is generating interrupt.

> I agree with the idea to use pre-allocated buffer variables in the interrupt handler. Though I don't think it can help a lot reduce the CPU load, it might be helpful for reducing latency and potential leaks.
> 
> Please correct me.

Yes you are correct.I've just tested it out.I have also preallocated buffer in VoodooInput.I didnt notice any performance improvements.

Goshin, 2020-05-08:
> Did you disable touchpad with hotkey? There is another problem related to this.I2C service does not pause/detach when trackpad is disabled.If i put a finger on my trackpad,cpu usage spikes meaning trackpad is generating interrupt.

Yes. VoodooI2CHID just ignores the finger touch data from the touchpad when disabled, so it actually does not disable the device when being set disabled. It is not a problem for me. I meant to say the 3% CPU load you mention above might be not caused by copying report data.

kprinssu, 2020-05-08:
@tiger511 Excellent work. Only thing to note when using preallocated buffers, we will have ensure that the access to the buffer is synchronized.

kprinssu, 2020-05-09:
@tiger511 Please feel create a PR, I will review it and merge it in.

tiger511, 2020-05-09:
Please take a loot at these changes:
https://github.com/tiger511/VoodooI2CHID/commit/46e23375a3594221eb83360bc0fa9171ebf335f4
https://github.com/tiger511/VoodooInput/commit/e315ce7f57df45630cfbf4fbcc25e7cc47fe2eeb

kprinssu, 2020-05-09:
Can you create PRs? Best to follow Github flows as it makes collaboration and change tracking easier.

tiger511, 2020-05-09:
> Can you create PRs? Best to follow Github flows as it makes collaboration and change tracking easier

I've creted PR.

> Couple of things:
> 
> * why are allocating a IOMemoryDescriptor and then releasing it right away ([tiger511/VoodooInput@e315ce7#diff-9c0f831ad415aca9affec569d2f390ebR281](https://github.com/tiger511/VoodooInput/commit/e315ce7f57df45630cfbf4fbcc25e7cc47fe2eeb#diff-9c0f831ad415aca9affec569d2f390ebR281))

I didn't release it right away. IOMemoryDescriptor is released in releaseResources()
> * move lines 269-271 to the end of the function as the interrupts and polling mode is still running (potential use-after-free, https://github.com/tiger511/VoodooI2CHID/blob/46e23375a3594221eb83360bc0fa9171ebf335f4/VoodooI2CHID/VoodooI2CHIDDevice.cpp#L269)

> * it's best to use OSSafeReleaseNull to release pointers as it will NULL the pointer and cause a crash if it were to be used again ([tiger511/VoodooI2CHID@46e2337#diff-879e63994a863a726fb58e4c54bd1fcaR269](https://github.com/tiger511/VoodooI2CHID/commit/46e23375a3594221eb83360bc0fa9171ebf335f4#diff-879e63994a863a726fb58e4c54bd1fcaR269))
> * when creating the PR, can you remove out the commented code please
> * minor/pedantic stylistic comment, [tiger511/VoodooI2CHID@46e2337#diff-879e63994a863a726fb58e4c54bd1fcaR161](https://github.com/tiger511/VoodooI2CHID/commit/46e23375a3594221eb83360bc0fa9171ebf335f4#diff-879e63994a863a726fb58e4c54bd1fcaR161) should have a space between `)` and `{`
> * Did you try releasing the report before calling IOFree, addressing this https://github.com/tiger511/VoodooI2CHID/blob/46e23375a3594221eb83360bc0fa9171ebf335f4/VoodooI2CHID/VoodooI2CHIDDevice.cpp#L557
> * Good practice to NULL pointers on declaration, [tiger511/VoodooI2CHID@46e2337#diff-032fd783182b578b17394d65fe0b3272R70](https://github.com/tiger511/VoodooI2CHID/commit/46e23375a3594221eb83360bc0fa9171ebf335f4#diff-032fd783182b578b17394d65fe0b3272R70)

All done.

kprinssu, 2022-02-27:
I am closing this as there has not been much movement on this issue for nearly 2 years. Please create a PR if you want the changes to be merged.

closed 2022-02-27 by kprinssu