VoodooI2C/VoodooI2C #304: trackpad is lagging on xiaomi mi notebook pro 15.6

boo50, 2020-05-08:
Hello, I've installed Catalina on my Mi notebook pro and have issue with trackpad.
It worked jerky, or laggy.
I've same system on another notebook - mi notebook air 12.5 and there touchpad work very smooth, like on real mac and it use 
Voodool2C.kext
Voodool2CSynaptics.kext
VoodoolInput.kext
VoodoolPS2Controller.kext

On Mi notebook pro I use latest versions of:
VoodooI2C.kext
VoodooI2CHID.kext
VoodooInput.kext
VoodooPS2Controller.kext

Also tried version from mi air (2CSynaptics), but touchpad not working at all with it.

I record a video explaining a problem
https://www.youtube.com/watch?v=Z3PzXIVod98
I move very smooth but window is not moving smooth, but jumping, twitches.

nikhil19rathore99, 2020-05-08:
> Hello, I've installed Catalina on my Mi notebook pro and have issue with trackpad.
> It worked jerky, or laggy.
> I've same system on another notebook - mi notebook air 12.5 and there touchpad work very [...]

Same problem with mine newer version is too laggy and not smooth cursor!!

Goshin, 2020-05-08:
@boo50 GPIO or polling mode? Try to remove VoodooInput.kext and use the one buddled in VoodooI2C.kext.

boo50, 2020-05-08:
@Goshin don't know about GPIO or polling mode, where to check ?

kprinssu, 2020-05-08:
@boo50 Can you please attach a troubleshooting archive? Please take a look at https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. Please include your DSDT and SSDT files are as well.

boo50, 2020-05-09:
@kprinssu acrhive in attachment.
I don't know where to get DSDT, everything else is in archive.
I use EFI from https://github.com/daliansky/XiaoMi-Pro-Hackintosh with clover bootloader if it make sense.

[mi notebook pro troubleshooting.zip](https://github.com/alexandred/VoodooI2C/files/4602927/mi.notebook.pro.troubleshooting.zip)

kprinssu, 2020-05-09:
@boo50 it looks VoodooInput is failing to initialise properly. Did you try following @Goshin's suggestion of removing the extraneous VoodooInput kext?

VoodooInput is currently distributed within VoodooI2C so you **should not** have it by self.

nikhil19rathore99, 2020-05-11:
> get

go to EFI/Clover/ACPI/patched folder u'll get the DSDT.aml

kprinssu, 2020-05-14:
@boo50 this is most likely related to the recent changes introduced by @Goshin. There are multiple other users with similar issues. @Goshin can we spend time together on debugging these issues?

Goshin, 2020-05-14:
@kprinssu Sure, but this one in @boo50 's case, he got the problem since v2.2 as he described in https://github.com/daliansky/XiaoMi-Pro-Hackintosh/issues/388#issuecomment-625620890. I tend to think it could be a configuration-related issue.

kprinssu, 2020-05-14:
@boo50 can you please try redoing your configuration from scratch? Also the video you linked to here, https://github.com/daliansky/XiaoMi-Pro-Hackintosh/issues/388#issuecomment-625626599 doesn't show anything abnormal. Please use fingemgmt when demoing as it shows the issue much more clearly.

tiger511, 2020-05-14:
> @Goshin don't know about GPIO or polling mode, where to check ?

What's the output of the following command: `log show --last=boot|grep Voodoo`

boo50, 2020-05-15:
@tiger511 
2020-05-10 10:11:33.877000+0700 0xb8       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOSunrisePointLP::Loading GPIO Data for SunrisePoint-LP
2020-05-10 10:11:33.877082+0700 0xb8       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOSunrisePointLP::VoodooGPIO Init!
2020-05-10 10:11:33.877084+0700 0xb8       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 0
2020-05-10 10:11:33.877129+0700 0xb8       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 1
2020-05-10 10:11:33.877157+0700 0xb8       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 2
2020-05-10 10:11:33.877259+0700 0x76       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOSunrisePointLP::GPIO Controller is already awake! Not reinitializing.
2020-05-10 10:11:42.109236+0700 0x266      Default     0x0                  112    0    hidd: (IOKit) [com.apple.iohid:default] Service added: IOHIDService name:VoodooI2CPrecisionTouchpadHIDEventDriver id:0x100000397 primaryUsagePage:0x1 primaryUsage:0x2 transport:I2C reportInterval:8000 batchInterval:0 events:0 mask:0x0
2020-05-10 10:12:16.123326+0700 0xa0b      Default     0x0                  0      0    kernel: (kernel) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0

About versions - I'm just installed macOs in this notebook, and tried different verstions of VoodooI2C.kext and results are the same.
On the video probably it's not so clear, but as I mentioned before, I've another mi air 12.5 with catalina and working with it's touchpad is a pleasure compared to mine, it works much more accurately and responsively, like on a real macbook.

stevezhengshiqi, 2020-05-19:
@Goshin Since XiaoMi-Pro's IOInterruptSpecifier is `10`(smaller than `2F`), I think there's little thing to do with the configuration. Both GPIO and APIC interrupts work for XiaoMi-Pro natively and @boo50 could try to use the APIC one.

@boo50 

> On the video probably it's not so clear, but as I mentioned before, I've another mi air 12.5 with catalina and working with it's touchpad is a pleasure compared to mine

It doesn't help if only mention the laptop model. To help developers better track the problem,  you also need to tell which satellite you are using and the trackpad model. For example, XiaoMi-Pro's trackpad model is  `ETD2303`.

boo50, 2020-05-19:
Can you point me please to modes, how to set APIC mode instead of GPIO?

Regarding mi air touchpad:
Notebook model is 2017 Mi air 12.5 with m7y30 cpu, is it possible to get trackpad model from macOs?
And I'm use latest versions of VoodooI2C.kext with VoodooI2CSynaptics.kext satellite there.

stevezhengshiqi, 2020-05-19:
@boo50 After doing some tests, I found out that I mistakenly thought XiaoMi-Pro has APIC Interrupt mode. It uses GPIO interrupt all the time. BTW, XiaoMi-Pro still doesn’t need configuration for GPIO pinning because APIC Pin number is 10.

tiger511, 2020-05-19:
> @boo50 After doing some tests, I found out that I mistakenly thought XiaoMi-Pro has APIC Interrupt mode. It uses GPIO interrupt all the time. BTW, XiaoMi-Pro still doesn’t need configuration for GPIO pinning because APIC Pin number is 10.

So the trackpad works out of the box without Voodooi2C?

stevezhengshiqi, 2020-05-19:
@tiger511 See https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning

> If your hexadecimal APIC pin number is less than or equal to 0x2F then you are good to go and can skip straight to the section Installing the kext.

ben9923, 2020-12-26:
Does it still happen with latest release of VoodooI2C?

ama31337, 2020-12-30:
just installed a new version, can't say it's as smooth as on 12.5 mi, but first impression is what it's become better.
probably it's a hw related because touchpads are different and after real mac you'll see difference for sure.

ben9923, 2020-12-31:
> just installed a new version, can't say it's as smooth as on 12.5 mi, but first impression is what it's become better.
> probably it's a hw related because touchpads are different and after real mac you'll see difference for sure.

Can you attach a troubleshooting archive made with OC-gen-debug or RunMe.app?

Can you also try polling mode with the `-vi2c-force-polling` boot arg?

ama31337, 2021-01-06:
I'm using Clover, not OC.

ben9923, 2021-01-06:
> I'm using Clover, not OC.
> Archive from RunMe in attachment
> [mi15.4.zip](https://github.com/VoodooI2C/VoodooI2C/files/5774105/mi15.4.zip)

We don't support setups adding a fake device, it's just too prone to errors, and is just unnecessary.
You also seem to not have an OS patch, which could be needed.
Please make your setup not use one, then attach a new archive. Notice IOReg is not included.
As for kernel logs, add the `msgbuf=1048576` boot arg and attach the output of the `dmesg` command.

Did you also try polling mode, as requested?

ama31337, 2021-01-07:
Can you please explain a a little ?
what does mean fake device as well as OS patch.

ben9923, 2021-01-08:
@ama31337 Your SSDT-TPD0 just adds a whole trackpad device to your ACPI tables, why?
OS patch - emulating an OS other than 'Darwin' to make ACPI work as intended on non-Windows systems.

I can't explain ACPI patching to you, being familiar with it is a requirement to use VoodooI2C...
Proper trackpad setup is _almost never_ plug & play.

Didn't mean to close the issue, tho.

kprinssu, 2021-01-08:
@ben9923 sorry to take over the conversation here.

@ama31337  Please read up on ACPI specification as @ben9923 is pretty much setting up your configuration from scratch. Please properly set up your Hackintosh configuration before asking for help. 

It's very tedious on our end to provide support for wide array of devices and it makes lives much easier if you understand what is required. Additionally, please consider OpenCore over Clover, as Clover is starting to show its age and has numerous issues with loading/injecting kexts.

kprinssu, 2022-02-27:
Please test with 2.7, if there is no response within 3 days, I will be closing this issue.

closed 2022-07-24 by kprinssu