VoodooI2C/VoodooI2C #305: Newer version 2.4.2 cause lagginess

nikhil19rathore99, 2020-05-08:
I finally figured mine problem was lagginess with newer version all other issues were resolved
Asus X55UQK
sattelite: vodoi2cFTE
vodooi2c 2.4.2 version

kprinssu, 2020-05-08:
Frankly speaking, I am not sure how many times I can point this out to you. Please *read* https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. Please *attach* the troubleshooting archive.

nikhil19rathore99, 2020-05-08:
> Frankly speaking, I am not sure how many times I can point this out to you. Please **read** https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. Please **attach** the troubleshooting archive.
> 
> As mentioned in the previous issue you created, this most likely related to the interrupt changes not being ported to the FTE satellite. As I do not have a FTE device on hand, I cannot easily port and test the changes. PRs are welcome and you are welcome to try the port yoursel

> Frankly speaking, I am not sure how many times I can point this out to you. Please **read** https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. Please **attach** the troubleshooting archive.
> 
> As mentioned in the previous issue you created, this most likely related to the interrupt changes not being ported to the FTE satellite. As I do not have a FTE device on hand, I cannot easily port and test the changes. PRs are welcome and you are welcome to try the port yourself.

Thanks to you, the only problem left was cursor lagging and cursor smoothness, the interrupt issue was resolved in 2.4.2 version, i'll be posting the log files,  thanks again for you kind support..

kprinssu, 2020-05-08:
As previously mentioned, these changes (https://github.com/kprinssu/VoodooI2CELAN/commit/6897677af23c1e647760ebfb4e23af22883fc995) are required to be added in FTE kext. I am quite certain that is the issue you are facing.

As I mentioned before, please wait until the changes are added into the FTE kext. Please **do not** spam other issues or Gitter requesting for help.

nikhil19rathore99, 2020-05-08:
> fo

can u suggest the method how to port changes from one satellite to another??

kprinssu, 2020-05-08:
No. I am not trying to be rude but if you are asking me that question then you shouldn't be attempting the patch. I commented on the issue you created with the original creator of the FTE kext here, https://github.com/prizraksarvar/VoodooI2CFTE/issues/3.

If they are kind enough to apply the changes then I can merge it into the upstream fork, until then please wait and please do not harass them.

nikhil19rathore99, 2020-05-08:
> No. I am not trying to be rude but if you are asking me that question then you shouldn't be attempting the patch. I commented on the issue you created with the original creator of the FTE kext here, [prizraksarvar/VoodooI2CFTE#3](https://github.com/prizraksarvar/VoodooI2CFTE/issues/3).
> 
> If they are kind enough to apply the changes then I can merge it into the upstream fork, until then please wait and please do not harass them.

okay i think i am a mess so sorry i should be waited and so sorry i think i am not a contributor to github that's why i didn't know how much you are working to make these things possible, sorry my bad, thank you. i'll close the issue

kprinssu, 2020-05-08:
No worries, but please do not spam the Github issues. It's quite hard to keep track of multiple issues as is.

I am reopening the ticket so we can track of it.

nikhil19rathore99, 2020-05-10:
he is getting success for fixing logs and bugs in 2.4.1 version and completely works for me but 2.4.2 getting lagging issue is 2.4.2 version more optimised than 2.4.1?? if not then i am happy to use 2.4.1 version!!

nikhil19rathore99, 2020-05-11:
Now everything is works beautifully, like real mac, you can close the issue,
Again Thankyou so much to put your effort and time in this,
I attached a file below to just to look
[image]

kprinssu, 2020-05-11:
Fixed by https://github.com/VoodooI2C/VoodooI2C/pull/312. Thanks @prizraksarvar.

closed 2020-05-11 by kprinssu