VoodooI2C/VoodooI2C #306: ELAN 1200 works, but not displayed in settings

snitron, 2020-05-08:
He﻿llo everyone! I made my﻿ trackpad working with VoodooI2C.kext and VoodooI2CHID.kext v 2.4.2 . Now it's working, but doesn't visible in﻿ settings (there is "Wait for a Bluetooth trackpad to be discovered") and doesn't work taps (only physical clicks). It ﻿works in Pooling mode. What should I do?

---

ASUS VivoBook X571GT-BQ037

(Intel Core i7 9750H, nVidia GeForce GTX 1650 (Intel UHD 630 in use), 16 GB RAM, OpenCore 0.5.8)

kprinssu, 2020-05-08:
Please **read** https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and **attach** the troubleshooting archive.

Your screenshot and config.plist is useless as it gives no insight into your configuration and hardware.

snitron, 2020-05-08:
> Please **read** https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and **attach** the troubleshooting archive.
> 
> Your screenshot and config.plist is useless as it gives no insight into your configuration and hardware.

Data is in archive. I didn't patch my DSDT. Only added SSDTs (in archive too) and _OSI to XOSI OpenCore patch. 

ASUS VivoBook X571GT-BQ037

(Intel Core i7 9750H, nVidia GeForce GTX 1650 (Intel UHD 630 in use), 16 GB RAM, OpenCore 0.5.8, macOS 10.15.3)
[data.zip](https://github.com/alexandred/VoodooI2C/files/4601225/data.zip)

stevezhengshiqi, 2020-05-08:
@snitron Also better check with the battery driver. If the battery percentage can't be driven, the trackpad panel won't work.

kprinssu, 2020-05-08:
@snitron everything looks good, not sure why it is not showing up. Can you try @stevezhengshiqi's suggestion?

snitron, 2020-05-09:
> @snitron everything looks good, not sure why it is not showing up. Can you try @stevezhengshiqi's suggestion?

Of course I will try, but can you give me any guide how to do this?

kprinssu, 2020-05-09:
@snitron Dortania is a good guide to follow https://1revenger1.gitbook.io/laptop-guide/. You can also use Rehabman’s guides, please note that these guides are out of the scope of VoodooI2C and I will not be able to provide support for it.

kprinssu, 2020-05-22:
Closing for now as I suspect is not related to VoodooI2C. Please reopen if you have issues.

closed 2020-05-22 by kprinssu