VoodooI2C/VoodooI2C #307: Accessing SMBus from userspace - RGB device control

CalcProgrammer1, 2020-05-08:
Does VoodooI2C provide an equivalent to /dev/i2c-X on Linux that could be accessed similarly to the i2c-tools programs (i2cdump, i2cdetect, etc)?

I'm the lead developer of OpenRGB, an open source library and application for controlling RGB lighting on PC components and peripheral devices.  Many RGB devices are controlled using SMBus or I2C, mainly RGB RAM modules and some RGB-enabled motherboards (Asus Aura and Gigabyte RGB Fusion).  My project implements the drivers for these devices in userspace, which is ideal for portability as I can use a custom i2c driver implementation on Windows using inpout32 and share the same code.  I'm interested in porting to MacOS and need a way to access I2C/SMBus adapters from userspace.  Your project looks like just what I need, but I don't see any example userspace apps such as i2c-tools, only kernel extensions, so it seems I'm attempting to use it in a different way than most.

Would you be able to provide documentation on how to go about this?  I'm a complete MacOS noob unfortunately.

My project:

https://gitlab.com/CalcProgrammer1/OpenRGB

alexandred, 2020-05-08:
Unfortunately we don’t provide user space frontends to the I2C controllers as Linux does (although it is a long term aspiration of mine to add this feature in at some point).

Also, having had a quick cursory glance at your project, I am not sure if VoodooI2C supports the I2C controllers you use. We support the I2C controllers that are baked into the Intel SoC that are found on most laptop motherboards today. If I am not mistaken the I2C controllers you need support for are plug and play ones? (Or do the hardware components connect directly to the built in I2C bus on the motherboard?

CalcProgrammer1, 2020-05-08:
The controller I'm interested in for my Hackintosh test bench is i2c-i801, which is the SMBus controller built into Intel's chipsets (Z270 in my case).  The DDR SPD slots connect to this bus and the RGB controllers on the RAM use the same bus as the SPD EEPROMs.

I may be able to help in adding a userspace interface if I can familiarize myself with how that would be done in MacOS.  Eventually I would want to get i2c-piix4 ported over as well, as that driver gives access to the SMBus controllers on AMD chipsets, also used for SPD and RGB on memory modules.

alexandred, 2020-05-08:
Ah right I see, then you’ll want VoodooSMBUS (https://github.com/leo-labs/VoodooSMBus). That project is based on VoodooI2C and so also doesn’t have a userspace component but it does support the controller you are targeting. 

I have no experience when it comes to the user land side of things so I’m not sure if I can be of any help with that unfortunately. Apple’s documentation is pretty good when it comes to this kind of thing though.

You’re also more than welcome to join the VodoooI2C dev channel (gitter.im/alexandred/VoodooI2CDev) for questions about the kernel land part of VoodooI2C. I wasn’t involved in VoodooSMBUS but I imagine much of it is the same.

closed 2020-05-21 by kprinssu

ben9923, 2020-05-23:
Try having a looks at this one too (directly use Apple's SMBUS kexts?)