VoodooI2C/VoodooI2C #308: Issues with macOS Catalina 10.15.4

Halelluja12, 2020-05-09:
Seen voodooI2C new release not compatibility with Interrupt mode, but polling mode not working too, so sad

kecinzer, 2020-05-09:
I'm on macOS 10.15.4 and both modes works fine :). You muset have something with your patches.

Halelluja12, 2020-05-09:
> I'm on macOS 10.15.4 and both modes works fine :). You muset have something with your patches.

I have already GPIO patch,  override _OSI Mehtod patches, and modify _CRS Method, some case not working , i check with IO Reg, maclog VoodooI2C is work fine, but trackpad preferences panel still empty

kprinssu, 2020-05-09:
@kecinzer Please don’t comment unless you have something constructive.  

@Halelluja12 please read https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. Please follow those steps and then post a troubleshooting archive.

kprinssu, 2020-05-15:
Closing as there has been no reply from @Halelluja12. Please reopen if you upload the troubleshooting archive.

closed 2020-05-15 by kprinssu