VoodooI2C/VoodooI2C #309: V2.4.1 and above has buggy clicking

aspring14, 2020-05-09:
When I use the kext versions 2.4.1 or above, when clicking the trackpad it will act like a long press even if I click it really fast, making it not usable unless tap to click this on.
I'm on a Lenovo Ideapad S340 81N8003TUS. V2.4 there is sometimes random stopping and it caused a kernel panic in sleep mode once.

Goshin, 2020-05-09:
Try to disable `force click and haptic feedback` in trackpad pref.

kprinssu, 2020-05-09:
@aspring14 Please read https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. Please follow those steps and then post a troubleshooting archive.

aspring14, 2020-05-09:
Disabling Force Click and haptic feedback fixed it, updated to latest 2.4.2, I'll update if I get more kernel panics caused by VoodooI2C.

closed 2020-05-09 by aspring14