VoodooI2C/VoodooI2C #313: 2.4 breaks trackpad on dell xps 7590

shadowed87, 2020-05-13:
As per title, the last working version I can use on dell xps (7590 in my case) is 2.3, everything newer gives me no trackpad at all

kprinssu, 2020-05-14:
Please **read** https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and then please **attach** a troubleshooting archive.

Goshin, 2020-05-14:
> Please **read** https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and then please **attach** a troubleshooting archive.

Maybe you can create an issue template to remind people to follow the procedure.

kprinssu, 2020-05-14:
Believe me, I wish I could; I myself I am getting tired of repeating the same message. I don't have access to the settings to do so. I am waiting on @alexandred to set it up.

Goshin, 2020-05-14:
Well, but wouldn't it only require you to create a file `ISSUE_TEMPLATE.md` in `/.github/` or `/docs/`?

kprinssu, 2020-05-14:
You are indeed correct. Let's carry the conversation over on at Gitter.

tiger511, 2020-05-14:
I think it's the timeout issue in ResetHID().@shadowed87  Disable touchscreen form BIOS and try this:
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4631072/Archive.zip)

OldDream, 2020-05-15:
I have a XPS 9570 with the same issue, I block touchscreen in SSDT, 

> I think it's the timeout issue in ResetHID().@shadowed87 Disable touchscreen form BIOS and try this:
> [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4631072/Archive.zip)
> 
> If it still doens't work,post the output of `log show --last=boot|grep Voodoo`

I have a XPS 9570 with the same issue, I block touchscreen in SSDT, after using your kext, the trackPad still didn't work

Logs:
2020-05-15 17:27:18.989105+0800 0xc7       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::Loading GPIO Data for CannonLake-H
2020-05-15 17:27:18.989109+0800 0xc7       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH:: Creating new workloop
2020-05-15 17:27:18.989435+0800 0xc7       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::VoodooGPIO Init!
2020-05-15 17:27:18.989704+0800 0xc7       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::VoodooGPIO Initializing Community 0
2020-05-15 17:27:18.990366+0800 0xc7       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::VoodooGPIO Initializing Community 1
2020-05-15 17:27:18.990559+0800 0xc7       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::VoodooGPIO Initializing Community 2
2020-05-15 17:27:18.990776+0800 0xc7       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::VoodooGPIO Initializing Community 3
2020-05-15 17:27:18.991555+0800 0x76       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH::GPIO Controller is already awake! Not reinitializing.
2020-05-15 17:27:19.206288+0800 0xff       Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,a369 Starting I2C controller
2020-05-15 17:27:19.209639+0800 0xff       Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,a369 Set PCI power state D0
2020-05-15 17:27:19.213043+0800 0xff       Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,a369 Publishing nub
2020-05-15 17:27:19.217178+0800 0x116      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a369 Probing controller
2020-05-15 17:27:19.219182+0800 0x116      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a369 Found valid Synopsys component, continuing with initialisation
2020-05-15 17:27:19.229808+0800 0x116      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a369 Got bus configuration values
2020-05-15 17:27:19.233685+0800 0x116      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a369 Publishing device nubs
2020-05-15 17:27:19.239991+0800 0x116      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a369 Found I2C device: SYNA2393
2020-05-15 17:27:19.243594+0800 0x116      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeH
2020-05-15 17:27:32.092073+0800 0x28e      Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOCannonLakeH:: Returning existing workloop
2020-05-15 17:27:32.113252+0800 0x28e      Default     0x0                  130    0    hidd: (IOKit) [com.apple.iohid:default] Service added: IOHIDService name:VoodooI2CPrecisionTouchpadHIDEventDriver id:0x1000003b2 primaryUsagePage:0x1 primaryUsage:0x2 transport:I2C reportInterval:8000 batchInterval:0 events:0 mask:0x0
2020-05-15 17:27:53.419919+0800 0xb18      Default     0x0                  0      0    kernel: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0

kprinssu, 2020-05-15:
@tiger511 Please do not provide support without the troubleshooting archive as it's very difficult to diagnose without information.

@OldDream Please see https://github.com/VoodooI2C/VoodooI2C/issues/313#issuecomment-628552186.

tiger511, 2020-05-15:
> I have a XPS 9570 with the same issue, I block touchscreen in SSDT, after using your kext, the trackPad still didn't work

Thats weird.I have the same device with same trackpad.It works just fine for me.Can you share share your SSDT patch for trackpad ?

OldDream, 2020-05-16:
> > I have a XPS 9570 with the same issue, I block touchscreen in SSDT, after using your kext, the trackPad still didn't work
> 
> Thats weird.I have the same device with same trackpad.It works just fine for me.Can you share share your SSDT patch for trackpad ?

I found the problem, I didn't add VoodooI2C.kext/Contents/PlugIns/VoodooInput.kext to opencore setting , touchpad works after adding it.

OldDream, 2020-05-16:
> I found the problem, I didn't add VoodooI2C.kext/Contents/PlugIns/VoodooInput.kext to opencore setting , touchpad works after adding it.

well....works in 2.4 ,break in 2.4.2...

kprinssu, 2020-05-16:
@OldDream Once again, read https://github.com/VoodooI2C/VoodooI2C/issues/313#issuecomment-628552186. Support will not be given without the troubleshooting archive.

tiger511, 2020-05-16:
@OldDream  Try the Archive.zip i shared i above.

shadowed87, 2020-05-16:
Sorry for this late answer (got problems with the quarantene). 

I am deeply sorry if I didn't put the troubleshooting archive! This is the first time for me doing an hackintosh (and I still don't get 100%), I'll do it asap and post it here.

In the meantime I tried the archive shared by @tiger511 and everything works (without disabling touchscreen), what is the difference between the archive and 2.4.2 release? (it seems the same version)

P.s. I added VoodooPS2Controller to get the keyboard working.

shadowed87, 2020-05-16:
This should be the archive you asked, I hope there is everything. 
https://drive.google.com/file/d/1d5GQ65BsifKnOo73YSIGaJ_UJjOLM9k7/view

Moreover here there is my whole efi (with voodoo 2.3):
https://github.com/shadowed87/Hackintosh-XPS-7590

kprinssu, 2020-05-16:
@shadowed87 it looks to be a code issue. Most likely with @Goshin's changes. @tiger511 will have to figure out what the issue is and create a PR.

shadowed87, 2020-05-16:
thanks a lot for your support! 

I am still using @tiger511  archive.zip very reliably without any issue.

Thank you, and feel free to ask anything if you need me to try something else.

OldDream, 2020-05-18:
this is my SSDT and ioreg... already rolle back to 2.3
https://drive.google.com/file/d/14moih082gPKmfEtsvByI3TpFwD0hnh_U/view?usp=sharing

tctien342, 2020-06-17:
@shadowed87 did you still facing with this error ? mine is Ins7591 version have this bug too. Could you try set fastboot to `thorough` and test again ?

MuntashirAkon, 2020-08-11:
Possibly this commit fixed the issue: https://github.com/tiger511/VoodooI2C/commit/36696b3df06749eacc0affdd6ad66f2a78102328

Goshin, 2020-08-11:
> Possibly this commit fixed the issue: [tiger511@36696b3](https://github.com/tiger511/VoodooI2C/commit/36696b3df06749eacc0affdd6ad66f2a78102328)

I don't think it is related.

This reset timeout issue should be already fixed since v2.4.3. Can anyone confirm it?

MuntashirAkon, 2020-08-11:
> This reset timeout issue should be already fixed since v2.4.3. Can anyone confirm it?

I'll try and see if it works.

MuntashirAkon, 2020-08-11:
> This reset timeout issue should be already fixed since v2.4.3. Can anyone confirm it?

It's working for me. I've also noticed that I've previously used VoodooPS2's VoodooInput instead of VoodooI2C's which was probably causing the problem.

OldDream, 2020-08-11:
> > Possibly this commit fixed the issue: [tiger511@36696b3](https://github.com/tiger511/VoodooI2C/commit/36696b3df06749eacc0affdd6ad66f2a78102328)
> 
> I don't think it is related.
> 
> This reset timeout issue should be already fixed since v2.4.3. Can anyone confirm it?

In XPS 9570， touchpad and touchscreen work well in GPIO mode .

pmdevita, 2020-08-18:
XPS 7590, can confirm the newest release is working for me on the newest stable release of Catalina

shadowed87, 2020-08-18:
I am using opencore (with catalina) and if update all the kexts with the newest stable release, the trackpad stops working... Only the kexts in the archive shared here by @tiger511 works for me... How come?

Goshin, 2020-08-18:
@shadowed87 have you added VoodooInput in your OC config?

shadowed87, 2020-08-18:
Here is my voodoo related oc config

`
<dict>
				<key>BundlePath</key>
				<string>VoodooI2C.kext/Contents/PlugIns/VoodooInput.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<false/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/VoodooInput</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
			<dict>
				<key>BundlePath</key>
				<string>VoodooI2C.kext/Contents/PlugIns/VoodooGPIO.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/VoodooGPIO</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
			<dict>
				<key>BundlePath</key>
				<string>VoodooI2C.kext/Contents/PlugIns/VoodooI2CServices.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/VoodooI2CServices</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
			<dict>
				<key>BundlePath</key>
				<string>VoodooI2C.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/VoodooI2C</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
			<dict>
				<key>BundlePath</key>
				<string>VoodooI2CHID.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/VoodooI2CHID</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
			<dict>
				<key>BundlePath</key>
				<string>VoodooPS2Controller.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/VoodooPS2Controller</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
			<dict>
				<key>BundlePath</key>
				<string>VoodooPS2Controller.kext/Contents/PlugIns/VoodooPS2Keyboard.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/VoodooPS2Keyboard</string>
				<key>MaxKernel</key>
				<string></string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
`

MuntashirAkon, 2020-08-18:
@Goshin In <= v2.3, trackpad didn't work properly now and then after sleep but it eventually worked. But after upgrading to v2.4.4, it completely stops working. This is very very rare case. I just thought you should know that.

shadowed87, 2020-08-18:
I just triend every 2.4.x versions... Still not working for me. Does anyone see anything funny in my oc config?!?

Goshin, 2020-08-18:
@shadowed87 @MuntashirAkon can you provide some log please?

MuntashirAkon, 2020-08-18:
> @MuntashirAkon can you provide some log please?

As I've said earlier, it's extremely rare case. I'll try to catch some logs the next time it occurs.

MuntashirAkon, 2020-09-25:
@Goshin just to let you know that the problem is fixed after updating to 10.15.6.

ben9923, 2020-12-26:
Closing as issue reporter didn't reply for a while.
Feel free to re-open, with a troubleshooting archive made using latest VoodooI2C.

closed 2020-12-26 by ben9923