VoodooI2C/VoodooI2C #316: Persistent multi-touch implementation

gimdh, 2020-05-16:
I really love this implementation and I can tell this is as good as genuine magic trackpad.

However, the issue of most laptops is that their touchpad is nowhere close to the size of real macbooks'.

While performing gesture input, let's say 3-finger drag, it works just fine if I keep my fingers separated. But if each fingers are g
athered close enough, than touch input just freaks out and jumps over everywhere. 

While I cannot confirm this is not a device issue, at least Windows precision driver handles gestures with no such issue. So I guess some buggy code generates jittery readouts. 

In case of this is a hardware issue, my suggestion is an addition of algorithm that when multi-touch gestures are detected, sends multi-touch command first to initiate multi-touch 'mode', then only send a coordiate of median or average of all sensed touch inputs or, just coordiate of single finger until fingers are off the touchpad.

The idea is 3 and 4 finger gestures work fine with only one finger as long as corresponding n-finger multi touch mode is somehow fired by initial multi touch input.

Goshin, 2020-05-16:
The driver mainly parses and reports the finger coordinates from the touchpad input. Gesture handling is not its concern.

kprinssu, 2020-05-16:
Due to the generic nature of how we are supporting multiple different hardware configurations and integrating with macOS's own touch handling system this will be quite the monumental task to implement. This also sounds very configuration specific and I unfortunately have to say that we will not be implementing this feature request for the time being.

Additionally, this will feels outside the scope of VoodooI2C, as VoodooI2C should only read the data that the hardware provides and send it to macOS's stack (via VoodooInput). 

If you are willing to put the effort, please do so. PRs are always welcome. Otherwise, if you still think that it's a hardware issue, please read https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and attach a troubleshooting archive.

gimdh, 2020-05-16:
I see this is not a primary concern of this project. However, considering touchpad works just fine in linux and Windows, it is likely that VoodooI2C or VoodooI2CHID satellite driver have some issue.

My immediate guess is there could be some issue with ordering fingers(coordinate/n-th finger mappings are not consistent), and somehow coordinates of each fingers are not reported consistently to VoodooInput, which may result movement of cursor. With 3-finger drag gesture, I was able to figure out that when cursor jumps, they jumps back and forth between set of 2~4 coordinates. If it were a simple jitter caused by hardware, I would expect more random behavior. 

And even it's a hardware specific parsing issue, I have synaptics precision touchpad and other syaptics devices would not be far apart. Anyways, I see if jitter filtering algorithms should be added, then VoodooInput is better place to do so.

I'm new to macOS development, and I do not know much about drivers. However, I do have C/C++ experience and willing to contribute codebase. It would take a while. Currently my biggest concern is how should I test kext without replacing compiled kexts in /C/K/O and rebooting to load it, but I hope there is an easier way.

kprinssu, 2020-05-22:
@gimdh I am closing this ticket for now. There is now ongoing discussion in VodooInput to get palm rejection in place. You can follow/join the conversation here, https://github.com/acidanthera/bugtracker/issues/932#issuecomment-632795017

closed 2020-05-22 by kprinssu