VoodooI2C/VoodooI2C #317: the multitouch gesture on the touchscreen conflicts with the stylus pen input (reopen issue of #188)

Jamesxxx1997, 2020-05-16:
the latest version of voodooi2c still address the same issue as #188.
machine model : Miix 520-121KB (Type 81CG）
CPU model ： i7-8550u
OS version : 10.15.4
VoodooI2C version : 2.4.2
bug description : the multitouch gesture on the touchscreen conflicts with the stylus pen input. The stylus pen input doesn't conflict with single touch on touchscreen ; However , multitouch gesture on touchscreen will make conflict with stylus pen input , making the touchscreen cannot sense stylus pen until single touch on touchscreen.
files
video.zip : https://drive.google.com/open?id=1UmtAULEyhycypHZOL5_OgTMxO8abO0v6
debug file : https://drive.google.com/open?id=1DNngAUVuDj-LnNbj7wiOmt63_6ZobZy1

kprinssu, 2020-05-16:
The issue with how we track the stylus and fingers. A better option would be to track them separately (allocate a stylus struct instead of using the transducers array). @blankmac what do you think?

Jamesxxx1997, 2020-05-30:
@blankmac @kprinssu I am sincerely thankful for your work to give us the touchscreen function that apple don't give us ; however , I have to honestly say that this issue is so annoying and I eventually give up using my miix-520 as the daily driver. If this issue is unfixable , **may you mention this issue on the voodooi2c readme page?** I think that whoever plan to build a touchscreen hackintosh as a daily driver , should aware of this issue , thank you.

By the way , since voodooi2c not only support i2c but also usb touchscreen , I also use voodooi2c to implement native touchscreen multitouch gesture on my **Thinkpad** x1 yoga 2018. As far as I know , the **Thinkpad** series only use **usb** rather than i2c touchscreen(the laptop implement touchscreen function via usb device "pen and multitouch sensor") , **can also use voodooi2c to implement touchscreen gesture**. Moreover , the usb touchscreen **don't** have this stylus-finger conflict. I provide some information about my Thinkpad x1 yoga 2018 information below , hope this help.

[image]
[image]

[thinkpad x1 yoga 2018 , usb touchscreen ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/4704918/thinkpad.x1.yoga.2018.usb.touchscreen.ioreg.zip)

blankmac, 2020-06-04:
@kprinssu Anything is worth a try, you would know better than me!  :D  I've tried to reproduce the problem on my device but it takes quite a few tries to pop up for me.

Jamesxxx1997, 2020-09-11:
Hi! @xeijix has this problem been fixed? I am willing to test the fix of the issue , if you can , please send me the compiler version of the driver, thank you

Jamesxxx1997, 2020-09-11:
Sorry to bothering , but @kprinssu @Goshin is this issue able to fix ? I’m worrying that this issue is not possible to fix. For this issue makes the touchscreen unbearable to use , I want to sell it to pay my tuition , thank you

Goshin, 2020-09-11:
@Jamesxxx1997 there is a PR for this issue https://github.com/VoodooI2C/VoodooI2CHID/pull/36.

Also, can you give a try to this fix? (changes are mentioned in https://github.com/VoodooI2C/VoodooI2CHID/pull/36#issuecomment-643612068)
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5208597/Archive.zip)

Jamesxxx1997, 2020-09-13:
@Goshin the PR solve the multitouch conflict issue , however it introduce another issue in this
video (https://youtu.be/GhMk6r_6gPw). The pointer will shift back to some fixed place when the pen nip is moved away from touchscreen.

Jamesxxx1997, 2020-09-19:
> @Goshin the PR solve the multitouch conflict issue , however it introduce another issue in this
> video (https://youtu.be/GhMk6r_6gPw). The pointer will shift back to some fixed place when the pen nip is moved away from touchscreen.

Now this problems is found in same scenario as I mentioned above:  the multitouch gesture on the touchscreen conflicts with the stylus pen input. The stylus pen input doesn't conflict with single touch on touchscreen ; However , multitouch gesture on touchscreen will make conflict with stylus pen input , making the pointer shift back to the fixed place when the pen nip is moved away from touchscreen , this issues remains exist until single touch on touchscreen.

Jamesxxx1997, 2020-10-09:
@xeijx @zhen-zen can you guys help fixing this issue? thanks!