VoodooI2C/VoodooI2C #318: Touchpad randomly not loaded during boot phase...

profzei, 2020-05-16:
**Describe the bug**
During the last 3 weeks randomly the touchpad doesn't load... so it didn't work then.
This issue occurs both with and without the use of SSDT-GPI0.aml and SSDT-TPL1.aml made with the auxilium of the app GenI2C (however they are the same as made by hand following instructions present at "chapter" GPIO pinning).

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes, but I found nothing useful

**System Environment**
1. Operating system: Mojave 10.14.6 Supplemental Security Update 2
2. I'm using VoodooI2C and VoodooI2CHID both at v. 2.4.2
3. Hardware configuration: laptop Huawei Matebook X Pro (2018) i7-8550U Ram 8GB, 3k monitor, Clover v.5116 with touchscreen (mapped at I2C0.TPD0) and touchpad (mapped at I2C1.TPL1). I'm using in config.plist two kernel patches "com.apple.driver.AppleIntelLpssI2C" and "com.apple.driver.AppleIntelLpssI2CController" even if these two "old" issues should be fixed with VoodooI2C v. 2.4 (GenI2C app in "Diagnose" section still checked the presence of these two patches)

**Troubleshooting Archive**
[DSDT copy.dsl.zip](https://github.com/VoodooI2C/VoodooI2C/files/4638610/DSDT.copy.dsl.zip)
[SSDT-TPL1.dsl.zip](https://github.com/VoodooI2C/VoodooI2C/files/4638612/SSDT-TPL1.dsl.zip)
[SSDT-GPI0.dsl.zip](https://github.com/VoodooI2C/VoodooI2C/files/4638614/SSDT-GPI0.dsl.zip)
[MBXP_ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/4638616/MBXP_ioreg.zip)
[boot_log_with_issue.txt](https://github.com/VoodooI2C/VoodooI2C/files/4638631/boot_log_with_issue.txt)
[boot_log_without_issue.txt](https://github.com/VoodooI2C/VoodooI2C/files/4638632/boot_log_without_issue.txt)

kprinssu, 2020-05-16:
From your Gitter conversation with @Goshin, this was fixed with https://github.com/VoodooI2C/VoodooI2CHID/commit/67d477da75332b74bfde66de6890e5d303e50609.

Please wait for a new version of VoodooI2C as it's fixed in master.

Goshin, 2020-05-16:
@kprinssu actually it is a kext loading issue according to the log.

```
kernel: kxld[com.alexandred.VoodooI2CServices]: The super class vtable '__ZTV9IOService' for vtable '__ZTV15VoodooI2CLogger' is out of date. Make sure your kext has been built against the correct headers.
kernel: kxld[com.alexandred.VoodooI2CServices]: The super class vtable '__ZTV9IOService' for vtable '__ZTV17VoodooI2CServices' is out of date. Make sure your kext has been built against the correct headers.
kernel: Can't load kext com.alexandred.VoodooI2CServices - link failed.
kernel: Failed to load executable for kext com.alexandred.VoodooI2CServices.
kernel: Kext com.alexandred.VoodooI2CServices failed to load (0xdc008016).
kernel: Failed to load kext com.alexandred.VoodooI2CServices (error 0xdc008016).
kernel: kxld[org.coolstar.VoodooGPIO]: The super class vtable '__ZTV9IOService' for vtable '__ZTV10VoodooGPIO' is out of date. Make sure your kext has been built against the correct headers.
kernel: kxld[org.coolstar.VoodooGPIO]: The __ZTV21VoodooGPIOCannonLakeH was not patched because its parent, the __ZTV10VoodooGPIO, was not patchable.
kernel: kxld[org.coolstar.VoodooGPIO]: The __ZTV22VoodooGPIOCannonLakeLP was not patched because its parent, the __ZTV10VoodooGPIO, was not patchable.
kernel: kxld[org.coolstar.VoodooGPIO]: The __ZTV23VoodooGPIOSunrisePointH was not patched because its parent, the __ZTV10VoodooGPIO, was not patchable.
kernel: kxld[org.coolstar.VoodooGPIO]: The __ZTV24VoodooGPIOSunrisePointLP was not patched because its parent, the __ZTV10VoodooGPIO, was not patchable.
kernel: Can't load kext org.coolstar.VoodooGPIO - link failed.
kernel: Failed to load executable for kext org.coolstar.VoodooGPIO.
kernel: Kext org.coolstar.VoodooGPIO failed to load (0xdc008016).
kernel: Failed to load kext org.coolstar.VoodooGPIO (error 0xdc008016).
kernel: Couldn't alloc class "VoodooGPIOSunrisePointLP"
kernel: Kext com.alexandred.VoodooI2C - library kext com.alexandred.VoodooI2CServices not found.
kextd: (IOKit) [com.apple.kext:kextlog] Kext id com.alexandred.VoodooI2C not found; removing personalities from kernel.
```

profzei, 2020-05-16:
...but I loaded VoodooI2C/VoodooI2CHID@67d477d which @Goshin attached in Gitter Conversation and I got the same issue!

Goshin, 2020-05-16:
@profzei that kext I provided in Gitter is built on 10.15 so it won't work on 10.14. Did you get the log boot_log_with_issue.txt with the kext from 2.4.2 official release? Are you using Clover or Opencore?

profzei, 2020-05-16:
@Goshin Sorry, my fault!

I run Mojave 10.14.6 with Clover r5116

I got the following file with the kext from 67d477d: 
[boot_log_with_issue.txt](https://github.com/VoodooI2C/VoodooI2C/files/4638794/boot_log_with_issue.txt)

I got the following file with the kext from the official release 2.4.2
[sys_log_voodoo_not_working_3.txt](https://github.com/VoodooI2C/VoodooI2C/files/4638800/sys_log_voodoo_not_working_3.txt)

Goshin, 2020-05-16:
> I got the following file with the kext from 67d477d:
> [boot_log_with_issue.txt](https://github.com/VoodooI2C/VoodooI2C/files/4638794/boot_log_with_issue.txt)

The kext is not compatible with 10.14, so the log `boot_log_with_issue.txt` doesn't mean anything. It seems related to the issue with touchpad and touchscreen. You may want to build the latest master branch yourself and give a test or you have to wait for the next release.

profzei, 2020-05-16:
Thank you! How could I build the kext? I guess I need to install Xcode but then I don't know...
I'm a noob but I'm fast in learning and this issue with touchpad is very annoying for me!

Goshin, 2020-05-16:
> Thank you! How could I build the kext? I guess I need to install Xcode but then I don't know...
> I'm a noob but I'm fast in learning and this issue with touchpad is very annoying for me!

https://voodooi2c.github.io/#Build%20Environment/Build%20Environment

profzei, 2020-05-17:
@Goshin  I don't know whether it could be useful, but I attach the result of command
`log show --last boot | grep "Voo"`
This time it occurred with official kext from release 2.4.2
[boot_log_242_with_issue.txt](https://github.com/VoodooI2C/VoodooI2C/files/4640361/boot_log_242_with_issue.txt)

Goshin, 2020-05-17:
@profzei boot log is sometimes incomplete. Try to manually load it with `kextload` and collect the log.

https://voodooi2c.github.io/#Build%20Environment/Build%20Environment
4.4. Manually loading VoodooI2C

ben9923, 2020-08-01:
Any update on this one?
Have you tried latest release, rather than a custom build?

Make sure you rebuild kextcache, etc.

profzei, 2020-08-02:
Hi @ben9923 , the issue very occasionally persists... now I'm using official v. 2.4.3 release

Goshin, 2020-08-09:
@profzei can you provide the log?

profzei, 2020-08-12:
Hi @Goshin : here there is a boot log at the end of which touchpad is not working...
[sys_bootlog_touchpad_not_working.txt](https://github.com/VoodooI2C/VoodooI2C/files/5061317/sys_bootlog_touchpad_not_working.txt)

Goshin, 2020-08-12:
> (VoodooI2CHID) VoodooI2CHIDDevice::SYNA2393 Timeout waiting for device to complete host initiated reset
(VoodooI2CHID) VoodooI2CHIDDevice::SYNA1D31 Timeout waiting for device to complete host initiated reset

Refer to https://github.com/VoodooI2C/VoodooI2C/issues/319.

profzei, 2020-08-12:
Hi @Goshin 
- I "think" touchpad and touchscreen are both working in interrupt mode with the following patch (which is currently in use):
[SSDT-TPXX.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/5061431/SSDT-TPXX.aml.zip)
- I "think" the same patch also disable the builtin TPD0 & TPL1 I2C devices

Do I need to increase the three parameters and recompile kext? ...In such case, I need to admit I'm not able in recompiling stuff...

Goshin, 2020-08-12:
Try the kexts in https://github.com/VoodooI2C/VoodooI2C/issues/319#issuecomment-634958211.

profzei, 2020-08-12:
@Goshin : Ok, now (when the issue manifested itself) I'm using official 2.4.4 release (8 August)... I'll try the kexts in #319 even if they are old (27 may) compared to your latest release...

profzei, 2020-08-25:
@Goshin : I tried extensively the kexts in [#319](https://github.com/VoodooI2C/VoodooI2C/issues/319) you pointed me to:
the situation is dramatically improved!
- during boot phase the issue is gone!
- BUT occasionally after sleep this issue is still there...

panda2k, 2020-08-26:
I'm also encountering this issue

profzei, 2020-09-26:
Just a sad **update**: I'm still using `VoodooI2C` + `VoodooI2CHID` "modified version" from [#319 ](https://github.com/VoodooI2C/VoodooI2C/issues/319) as suggested by @Goshin...

Last night, during sleep phase, machine got the following **KP**:

```
panic(cpu 4 caller 0xffffff80112469aa): Kernel trap at 0xffffff8011792945, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0xffffff8127a67000, CR3: 0x0000000028c61000, CR4: 0x00000000003626e0
RAX: 0x7c61a6dbce62f054, RBX: 0xffffff8011c23318, RCX: 0x0000000000000001, RDX: 0x0000000000000007
RSP: 0xffffff91389abcd0, RBP: 0xffffff91389abce0, RSI: 0xffffff8011c23318, RDI: 0xffffff805007f0c0
R8:  0xffffff8011aa4480, R9:  0x0000000000000000, R10: 0x0000000000000088, R11: 0x0000000000000001
R12: 0xffffff8038e2e900, R13: 0xffffff8011c23318, R14: 0xffffff805007f0c0, R15: 0x0000000000000055
RFL: 0x0000000000010286, RIP: 0xffffff8011792945, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff8127a67000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff8010f53220 : 0xffffff801111a65d 
0xffffff8010f53270 : 0xffffff8011254a75 
0xffffff8010f532b0 : 0xffffff80112465fe 
0xffffff8010f53300 : 0xffffff80110c0a40 
0xffffff8010f53320 : 0xffffff8011119d27 
0xffffff8010f53420 : 0xffffff801111a117 
0xffffff8010f53470 : 0xffffff80118c1a6c 
0xffffff8010f534e0 : 0xffffff80112469aa 
0xffffff8010f53660 : 0xffffff80112466a8 
0xffffff8010f536b0 : 0xffffff80110c0a40 
0xffffff8010f536d0 : 0xffffff8011792945 
0xffffff91389abce0 : 0xffffff7f95ec777e 
0xffffff91389abd20 : 0xffffff801182d7bf 
0xffffff91389abd50 : 0xffffff801182b33c 
0xffffff91389abd80 : 0xffffff801182e9d8 
0xffffff91389abde0 : 0xffffff7f95f3d263 
0xffffff91389abe00 : 0xffffff7f95f3d6a4 
0xffffff91389abe30 : 0xffffff8011813794 
0xffffff91389abea0 : 0xffffff801181359a 
0xffffff91389abec0 : 0xffffff801115c605 
0xffffff91389abf40 : 0xffffff801115c131 
0xffffff91389abfa0 : 0xffffff80110c013e 
      Kernel Extensions in backtrace:
         org.coolstar.VoodooGPIO(1.1)[D4C4E8FC-D51D-3CCE-BBD1-B67A39DC9E7A]@0xffffff7f95ec4000->0xffffff7f95edefff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2956198D-24F2-3790-A9B2-1EAB9434B906]@0xffffff7f923da000
         com.alexandred.VoodooI2CHID(1.0)[EDA918D7-CD1A-3E2A-8C98-5B455F2CA788]@0xffffff7f95f36000->0xffffff7f95f48fff
            dependency: com.alexandred.VoodooI2C(2.4.2)[297E8E68-706B-3272-961C-910CC1A66996]@0xffffff7f95efa000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2956198D-24F2-3790-A9B2-1EAB9434B906]@0xffffff7f923da000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[9643D692-8F23-37B2-9B89-4AA85D94BEE3]@0xffffff7f92067000

BSD process name corresponding to current thread: kernel_task
Boot args: -wegnoegpu -cdfon -igfxmlr darkwake=1 dart=0 gfxrst=1

Mac OS version:
19H2

Kernel version:
Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64
Kernel UUID: 05D51A3D-3A87-3FF0-98C3-9CF3827A3EDD
Kernel slide:     0x0000000010e00000
Kernel text base: 0xffffff8011000000
__HIB  text base: 0xffffff8010f00000
System model name: MacBookPro15,2 (Mac-827FB448E656EC26)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 6408734287576
```

profzei, 2020-10-19:
Update: using VoodooI2C 2.5.2 + satellite VoodooI2CHID touchpad is correctly recognised during boot, but there are still random problems:
- sometimes it is not recognised after wake
- or sometimes KP happens like the following

```
panic(cpu 0 caller 0xffffff8013c469aa): Kernel trap at 0xffffff8014192945, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0xffffff812a53307f, CR3: 0x000000002b74f000, CR4: 0x00000000003626e0
RAX: 0x91f61b3a961c380f, RBX: 0xffffff8014623318, RCX: 0x0000000000000000, RDX: 0x0000000000000007
RSP: 0xffffff913bc1bcd0, RBP: 0xffffff913bc1bce0, RSI: 0xffffff8014623318, RDI: 0xffffff805178a660
R8:  0xffffff80144a4480, R9:  0x0000000000000000, R10: 0x0000000000000088, R11: 0x0000000000000001
R12: 0xffffff803b916600, R13: 0xffffff8014623318, R14: 0xffffff805178a660, R15: 0x0000000000000055
RFL: 0x0000000000010286, RIP: 0xffffff8014192945, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff812a53307f, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 0

Backtrace (CPU 0), Frame : Return Address
0xffffff8013951220 : 0xffffff8013b1a65d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff8013951270 : 0xffffff8013c54a75 mach_kernel : _kdp_i386_trap + 0x155
0xffffff80139512b0 : 0xffffff8013c465fe mach_kernel : _kernel_trap + 0x4ee
0xffffff8013951300 : 0xffffff8013ac0a40 mach_kernel : _return_from_trap + 0xe0
0xffffff8013951320 : 0xffffff8013b19d27 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff8013951420 : 0xffffff8013b1a117 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff8013951470 : 0xffffff80142c1a6c mach_kernel : _panic + 0x54
0xffffff80139514e0 : 0xffffff8013c469aa mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff8013951660 : 0xffffff8013c466a8 mach_kernel : _kernel_trap + 0x598
0xffffff80139516b0 : 0xffffff8013ac0a40 mach_kernel : _return_from_trap + 0xe0
0xffffff80139516d0 : 0xffffff8014192945 mach_kernel : __ZN15OSMetaClassBase12safeMetaCastEPKS_PK11OSMetaClass + 0x15
0xffffff913bc1bce0 : 0xffffff7f989c64f4 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO19unregisterInterruptEi + 0xb2
0xffffff913bc1bd20 : 0xffffff801422d7bf mach_kernel : __ZN22IOInterruptEventSource11setWorkLoopEP10IOWorkLoop + 0x19f
0xffffff913bc1bd50 : 0xffffff801422b33c mach_kernel : __ZN10IOWorkLoop13_maintRequestEPvS0_S0_S0_ + 0x22c
0xffffff913bc1bd80 : 0xffffff801422e9d8 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0x138
0xffffff913bc1bde0 : 0xffffff7f98a351f5 com.alexandred.VoodooI2CHID : __ZN18VoodooI2CHIDDevice13stopInterruptEv + 0x63
0xffffff913bc1be00 : 0xffffff7f98a3563a com.alexandred.VoodooI2CHID : __ZN18VoodooI2CHIDDevice13setPowerStateEmP9IOService + 0xc0
0xffffff913bc1be30 : 0xffffff8014213794 mach_kernel : __ZN9IOService19driverSetPowerStateEv + 0x184
0xffffff913bc1bea0 : 0xffffff801421359a mach_kernel : __ZN9IOService15pmDriverCalloutEPS_ + 0x2a
0xffffff913bc1bec0 : 0xffffff8013b5c605 mach_kernel : _thread_call_delayed_timer + 0xec5
0xffffff913bc1bf40 : 0xffffff8013b5c131 mach_kernel : _thread_call_delayed_timer + 0x9f1
0xffffff913bc1bfa0 : 0xffffff8013ac013e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         org.coolstar.VoodooGPIO(1.1)[FB3BE4D0-E5F6-30B4-9797-DE6ADDEC2F90]@0xffffff7f989c3000->0xffffff7f989e1fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2956198D-24F2-3790-A9B2-1EAB9434B906]@0xffffff7f948a2000
         com.alexandred.VoodooI2CHID(1.0)[D892AED5-6F7F-3496-B3DA-E014475EB2CB]@0xffffff7f98a2e000->0xffffff7f98a56fff
            dependency: com.alexandred.VoodooI2C(2.5.2)[46356335-D26B-38BF-ACE3-43565C70A2C8]@0xffffff7f989f4000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2956198D-24F2-3790-A9B2-1EAB9434B906]@0xffffff7f948a2000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[9643D692-8F23-37B2-9B89-4AA85D94BEE3]@0xffffff7f944a6000

BSD process name corresponding to current thread: kernel_task
Boot args: keepsyms=1 msgbuf=1048576 -wegnoegpu -igfxnorpsc=1 -cdfon -igfxonln=1 -igfxmlr darkwake=1 dart=0 gfxrst=1

Mac OS version:
19H2

Kernel version:
Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64
Kernel UUID: 05D51A3D-3A87-3FF0-98C3-9CF3827A3EDD
Kernel slide:     0x0000000013800000
Kernel text base: 0xffffff8013a00000
__HIB  text base: 0xffffff8013900000
System model name: MacBookPro15,2 (Mac-827FB448E656EC26)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 5962780201211
```

ben9923, 2020-12-22:
Does it still happen with latest release?
Please attach a troubleshooting archive for the case it stops working after wake, and an updated panic log with 2.6.1.

profzei, 2021-01-04:
@ben9923 Issue fixed! Thanks a lot!

closed 2021-01-04 by profzei