VoodooI2C/VoodooI2C #319: ELAN0406 / Razer Blade Advanced Mid 2019 bus timeout after 2.2

boyuanx, 2020-05-17:
**Describe the bug**
ELAN0406 will not function with VoodooI2C versions after 2.2 due to various timeouts during boot (detailed logs below).

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes, there was a few mentions for an Asus having similar issues by `Kamil Kasperski @Ressetkk Nov 16 2019 13:58`.

**System Environment**
Please answer:
1. What is your operating system?
-  macOS Catalina 10.15.4 (OpenCore)
2. What version of the drivers (including drivers) are you using? 
- VoodooI2C 2.2 (2.2 works, 2.2+ don't)
3. What is your hardware configuration?
- Razer Blade Advanced Mid 2019:
- i7-9750H with ELAN0406 precision touchpad

**Troubleshooting Archive**
- Please note that boot logs are useless because for some reason it does NOT record all VoodooI2C logs, as reported here: https://www.insanelymac.com/forum/topic/341552-how-can-i-get-full-boot-log-in-verbose-mode/
- IORegistryExplorer file is attached (this is booting under 2.2 aka working trackpad).
[IOReg.zip](https://github.com/VoodooI2C/VoodooI2C/files/4639571/IOReg.zip)
- Due to the logging issue mentioned above, I had to record a video and manually pick out the relevant lines:
```
VoodooI2CPCIController::pci8086,a368 Starting I2C controller
VoodooI2CPCIController::pci8086,a368 Set PCI power state D0
VoodooI2CPCIController::pci8086,a368 Publishing nub
VoodooI2CPCIController::pci8086,a368 Probing controller
VoodooI2CPCIController::pci8086,a368 Found valid Synopsys component, continuing... (cutoff from video)
VoodooI2CPCIController::pci8086,a368 Got bus configuration values
VoodooI2CPCIController::pci8086,a368 Publishing device nubs
VoodooI2CDeviceNub::Got GPI0 Controller! VoodooGPIoCannonLakeH
VoodooGPI0CannonLakeH::Registering hardware pin 0x45 for GPI0 IRQ pin 0x52
VoodooI2CControllerDriver::pci8086,a368 Timeout waiting for bus to accept transfer request
VoodooI2CControllerDriver::pci8086,a368 Timed out waiting for bus to change state
VoodooI2CPrecisionTouchpadHIDEventDriver::ELAN0406 Putting device into Precision Touchpad Mode
VoodooI2CControllerDriver::pci8086,a368 Timeout waiting for bus not to be busy
```

**Additional context**
This precise issue is very likely caused by the commits in September 2019 adding timeout checks during boot. However, for whatever reason ELAN0406 will trigger a timeout and despite being correctly loaded as a Precision Trackpad, it will not function at all.

kprinssu, 2020-05-20:
This may be fixed by @Goshin. Please see https://github.com/VoodooI2C/VoodooI2C/commit/9afff08fd5a451efb6ee259fd5da1854d77ab968. 

For now, please test the archive found there. if it doesn't work please ping me and Goshin.

boyuanx, 2020-05-21:
@kprinssu Yup, the archive seems to be working! Rebooted 5 times and the trackpad works wonderfully. Really appreciate everyone's hard work!

kprinssu, 2020-05-21:
Great, the changes are in master and will be in the next release.

boyuanx, 2020-05-21:
@kprinssu Actually starting this morning it started to not work again. I did 10 reboots and the trackpad only worked 3 out of the 10 tries. Same error messages as above: bus timeouts. Reverting to v2.2 fixed it.

kprinssu, 2020-05-21:
That sounds like a kext cache issue. Please clear and rebuild it and try again.

boyuanx, 2020-05-21:
@kprinssu I tried rebuilding kext cache after every boot + switching up the loading order of kexts, still no joy after 5 reboots.

boyuanx, 2020-05-21:
Here's an IOReg under v2.2:
[IOReg_I2C_working.zip](https://github.com/VoodooI2C/VoodooI2C/files/4665174/IOReg_I2C_working.zip)

kprinssu, 2020-05-23:
I suspect that the built copy of master should work. I'll attach some pre-built binaries. Please ping me if I do not so within the next 24 hours.

kprinssu, 2020-05-23:
@boyuanx Please try these at your earliest convenience
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4672675/Archive.zip)
.

boyuanx, 2020-05-23:
@kprinssu 
1st reboot (after replacing kexts): doesn’t work
2nd reboot (after kext cache rebuild): works
3rd reboot: doesn’t work 
4th reboot (after kext cache rebuild): works
5th reboot: doesn’t work
6th reboot: works
7th reboot: works
8th reboot: doesn’t work
9th reboot: doesn’t work
10th reboot: doesn’t work 

And the error messages are still the same: bus timeouts.

kprinssu, 2020-05-24:
It seems like something is broken with initialising VoodooGPIO. @Goshin do you perhaps know why this may be happening?

Goshin, 2020-05-24:
@boyuanx have you tried the polling mode? If it works then the problem is in VoodooGPIO.

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4673202/Archive.zip)
Maybe this one will help. I adjust some timeout deadlines.

boyuanx, 2020-05-24:
@kprinssu @Goshin this version is much better. Only 1 failure out of 20 reboots. So it seems to be a deadline issue?

Goshin, 2020-05-24:
@boyuanx Testing the kexts in https://github.com/VoodooI2C/VoodooI2C/issues/319#issuecomment-633153172 in polling mode can help identify the problem. If polling mode doesn't work in your case, we can be sure the issue is related to the timeout settings.

boyuanx, 2020-05-25:
> @boyuanx Testing the kexts in [#319 (comment)](https://github.com/VoodooI2C/VoodooI2C/issues/319#issuecomment-633153172) in polling mode can help identify the problem. If polling mode doesn't work in your case, we can be sure the issue is related to the timeout settings.

@Goshin @kprinssu I tried using polling mode (confirmed by boot logs "cannot find interrupt using polling instead") and the behavior is still inconsistent. Sometimes it works, other times it tells me the bus timed out.

Goshin, 2020-05-27:
It seems like the i2c bus is not responsive when initializing. v2.2 works well because it waits endlessly. Could it be a hardware-specific issue, or related to the other kexts you have installed? I can give you a temp fix that removes the timeout limits.
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4691982/Archive.zip)

boyuanx, 2020-05-28:
@Goshin Thank you. 

I don't think it's related to my other kexts since they are all pretty standard. Here is a list:
- AppleALC
- CPUFriend
- CPUFriendDataProvider
- Lilu
- NoTouchID
- SMCBatteryManager
- SMCLightSensor
- SMCProcessor
- SMCSuperIO
- USBPorts
- VirtualSMC
- VoodooI2C
- VoodooI2CHID
- WhateverGreen

apocolipse, 2020-06-17:
Hi I’m using a 2020 model Razer Blade Advanced, it has the same model trackpad (ELAN0406) and I believe the same GPIO pins though I can’t confirm since I can’t get GPIO mode working, I believe I’m experiencing the same issue here. With Polling mode I’m able to get trackpad working about 50+% of the time, some boots it works some boots it doesn’t, haven’t been able to pin down the difference.  

When working or not, even when not working with GPIO pins set, it fully populates IOReg tree and I can see precision trackpad device there, just doesn’t work.

apocolipse, 2020-06-19:
@Goshin @boyuanx I'm able to get GPIO pinned mode working reliably every time on my 2020 model (same GPIO pin as the 2019 models) by adjusting some of those timeouts.  Specifically in `VoodooI2CHID/VoodooI2CHIDDevice.cpp:` 
```diff
@@ -331,7 +331,7 @@ IOReturn VoodooI2CHIDDevice::resetHIDDeviceGated() {

     // Device is required to complete a host-initiated reset in at most 6 seconds.

-    nanoseconds_to_absolutetime(6000000000, &absolute_time);
+    nanoseconds_to_absolutetime(12000000000, &absolute_time);
```

As well as these 2 in `VoodooI2C/VoodooI2C/VoodooI2CController/VoodooI2CControllerDriver.cpp`
```diff
@@ -170,7 +170,7 @@ IOReturn VoodooI2CControllerDriver::prepareTransferI2C(VoodooI2CControllerBusMes
      *   10ms is required, for example, when reading the HID descriptor at the first time.
      *   Timeout is set to 100ms (10ms x 10 times)
      */
-    nanoseconds_to_absolutetime(100000000, &abstime);
+    nanoseconds_to_absolutetime(1000000000, &abstime);
     clock_absolutetime_interval_to_deadline(abstime, &deadline);
     sleep = command_gate->commandSleep(&bus_device.command_complete, deadline, THREAD_INTERRUPTIBLE);

@@ -523,7 +523,7 @@ void VoodooI2CControllerDriver::stop(IOService* provider) {
 }

 IOReturn VoodooI2CControllerDriver::toggleBusState(VoodooI2CState enabled) {
-    int timeout = 500;
+    int timeout = 1000;
```

I'm not sure if those are ideal timeouts, but I now have a reliably working pinned trackpad 100% of the time compared to never working pinned and 50% of the time working polled.

boyuanx, 2020-06-19:
@apocolipse Nice work. FYI Goshin already posted a version without timeouts a few posts above :)

apocolipse, 2020-06-19:
> @apocolipse Nice work. FYI Goshin already posted a version without timeouts a few posts above :)

Yeah but I'm sure it didn't include the recent fixes for new comet lake i2c device ID's I had to make ;) So i had to figure out what those changes were heh, also I had an additional timeout that you didn't list in your logs

kprinssu, 2020-06-19:
@apocolipse that is pretty much a hack. Your hardware is not responding to the I2C read/writes within the default timeouts. Not a good approach as it may break functionality for other hardware.

tylernguyen, 2020-06-22:
Hello everyone,

I am the owner of https://github.com/tylernguyen/razer15-hackintosh

I also had this problem but have since fixed it. The solution is as followed:
1. Disable the builtin `TPD0` I2C device by loading a custom value to `GPHD`. See my patch [SSDT-OCGPI0-GPHD.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-OCGPI0-GPHD.dsl)
2. Create a new I2C device and move the contents of `TPD0` to it while modifying its contents to load blind in macOS. Also see my patch here: [SSDT-OCI2C-TPXX-RAZER15.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-OCI2C-TPXX-RAZER15.dsl). In my case, I called this device `TPXX` but you are welcome to edit my patch and call the device anything. Make sure have the support patch also: [SSDT-I2CxConf.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-I2CxConf.dsl)
*Please note that my `SSDT-OCI2C-TPXX-RAZER15.dsl` is pinned to my model's value, yours may be different. If my patch does not work, please double check this pin value.
3. Patch AWAC per [Dortania's guide](https://dortania.github.io/Getting-Started-With-ACPI/Universal/awac-methods/manual.html), see mine here: [SSDT-AWAC.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-AWAC.dsl)
4. In addition, please make sure that your battery patches are in order, including patching all Mutex values to 0.

boyuanx, 2020-06-23:
> Hello everyone,
> 
> I am the owner of https://github.com/tylernguyen/razer15-hackintosh
> 
> I believe this is a hackintosh issue, not a `VoodooI2C` one.
> 
> I also had this problem but have since fixed it. The solution is as followed:
> 
> 1. Disable the builtin `TPD0` I2C device by loading a custom value to `GPHD`. See my patch [SSDT-OCGPI0-GPHD.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-OCGPI0-GPHD.dsl)
> 2. Create a new I2C device and move the contents of `TPD0` to it while modifying its contents to load blind in macOS. Also see my patch here: [SSDT-OCI2C-TPXX-RAZER15.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-OCI2C-TPXX-RAZER15.dsl). In my case, I called this device `TPXX` but you are welcome to edit my patch and call the device anything. Make sure have the support patch also: [SSDT-I2CxConf.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-I2CxConf.dsl)
>    *Please note that my `SSDT-OCI2C-TPXX-RAZER15.dsl` is pinned to my model's value, yours may be different. If my patch does not work, please double check this pin value.
> 3. Patch AWAC per [Dortania's guide](https://dortania.github.io/Getting-Started-With-ACPI/Universal/awac-methods/manual.html), see mine here: [SSDT-AWAC.dsl](https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-AWAC.dsl)
> 4. In addition, please make sure that your battery patches are in order, including patching all Mutex values to 0.
> 
> Your trackpad should now load everytime on the newest `VoodooI2C` version.
> If you still have issues, first check my patch's pinning. After that, try using my `EFI-OpenCore` and check if the issue is fixed. If it is, `diff` your EFI against mine.

Unfortunately, your patch did not work for me. I can see TPXX in IOReg but it gives me the same timeout error during boot. Also interestingly your battery patch also doesn't work for me. Weird, considering we have basically the same laptop aside from your optical keyboard.

tylernguyen, 2020-06-24:
@boyuanx I've taken a look at your EFI posted on your repo. The most likely reason is because of your embedded controller. You renamed EC0 to EC. This is bad per OpenCore guidelines, see https://github.com/acidanthera/OpenCorePkg/blob/master/Docs/AcpiSamples/SSDT-EC.dsl.

I recommend you remove your rename patch and create a fake EC device instead. My battery patch is designed for that fake EC controller, not the renamed one. See my example here: https://github.com/tylernguyen/razer15-hackintosh/blob/master/patches/SSDT-EC-USBX-LAPTOP.dsl

About my I2C patched, it is most likely a pinning difference, just as I stated before. Please check your own pin configuration.

boyuanx, 2020-06-24:
> @boyuanx I've taken a look at your EFI posted on your repo. The most likely reason is because of your embedded controller. You renamed EC0 to EC. This is bad per OpenCore guidelines, see [...]

Interesting, thanks. I'll take a look.
(GPI0 pin is the same =))

tylernguyen, 2020-06-24:
> Interesting, thanks. I'll take a look.
> (GPI0 pin is the same =))

If your GPI0 pin is the same, then it is an entire patch difference. You would have to create your own fake I2C device patch by moving the contents of your old I2C device and editing it. Make sure you disable your old I2C device, it may depends on `GPHD` , `GPEN`, or `SBRG`, please reference your own `DSDT`.

ben9923, 2020-08-01:
@boyuanx @Goshin is this resolved in latest release?

Goshin, 2020-08-01:
@ben9923 according to @tylernguyen, it is a configuration issue.

boyuanx, 2020-08-01:
@ben9923 @Goshin @tylernguyen Even when I straight-up used his EFI folder, I still got the same timeout error.

tylernguyen, 2020-08-01:
> @ben9923 @Goshin @tylernguyen Even when I straight-up used his EFI folder, I still got the same timeout error. 

You have a different model than mine so you shouldn't use my patches and EFI. I've posted what I did above so you should try creating your own patches. Also, you may wish to look into VoodooRMI as an alternative I2C handler.

boyuanx, 2020-08-01:
@tylernguyen From what I've researched, the only difference between our models is the optical keyboard - everything else related to I2C is identical, down to the GPI0 pins.

tylernguyen, 2020-08-01:
> @tylernguyen From what I've researched, the only difference between our models is the optical keyboard - everything else related to I2C is identical, down to the GPI0 pins.

Right. But the patch contents comes from the actual I2C device itself within DSDT. Additionally, I2C trackpad relies a lot of battery status and EC patching to make things work smoothly, which differs greatly across CPUs and model generations.

boyuanx, 2020-08-01:
@tylernguyen I understand, but all the things you've mentioned are working perfectly under your patches, just like my previous patch. We also share the same motherboard and CPU.

tylernguyen, 2020-08-01:
> @tylernguyen I understand, but all the things you've mentioned are working perfectly under your patches, just like my previous patch. We also share the same motherboard and CPU.

My recommendation is that you diff the two DSDT and you will see how different they are. 

Again, you should look into VoodooRMI as an alternative I2C handler. You may find that kext easier to config.

kprinssu, 2020-10-03:
@boyuanx Can you check if 2.5 fixes your issue as it contains an increased timeout (thanks @apocolipse).

boyuanx, 2020-10-03:
> @boyuanx Can you check if 2.5 fixes your issue as it contains an increased timeout (thanks @apocolipse).

I recently got a hold of someone else's I2C SSDTs and it ended up fixing the timeout issue even on older versions. 2.5 continues to work perfectly.

The I2C SSDT I used to use (which had the timeout issue): https://github.com/tylernguyen/razer15-hackintosh/tree/master/EFI-OpenCore/EFI/OC/ACPI

The new I2C SSDT I use now that solves the timeout issue: https://github.com/Errrneist/Hackintosh-Razer-Blade-RZ09-0301/tree/master/EFI/11.00.0/3.1/OC/ACPI

FYI I'm on Big Sur and everything is working great, including the latest 2.5 version.

Thanks again!

kprinssu, 2020-10-03:
Awesome, please reopen the ticket if you have the issue.

closed 2020-10-03 by kprinssu