VoodooI2C/VoodooI2C #321: ELAN1200 stops working after a period of use

Jie2GG, 2020-05-19:
**Describe the bug**
ELAN1200 was not responsive after a period of use, so it stopped working. My Touchpad working in GPIO model

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes

**System Environment**
Please answer:
Operation System: macOS Catalina 10.15.4
Device Version: VoodooI2C + VoodooI2CHID (2.4.2)

|Name|Model|Note
:-:|:-:|:-:|
|CPU|Intel Core i5 8265U @ 1.6GHz
|Main board|X530FN_S5300FN|BIOS Version: 305
|RAM|16G x 2
|SSD|WDC PC SN520 (512GB)
|HDD|WDC WD20SPZX-22CRAT0 (2TB)
|WLAN|BCM94350ZEN (DW1820A)
|GPU|Intel UHD Graphics 620
|Touchpad|ELAN1200

**Troubleshooting Archive**
This is my laptop DSDT, SSDT, ioreg and logs
[VoodooI2C.zip](https://github.com/VoodooI2C/VoodooI2C/files/4650279/VoodooI2C.zip)

**Additional context**

I used FingerMgmt to detect the finger state and found that there is always a finger on the Touchpad (in fact, there is no finger on the Touchpad). 
This is the first time working has stopped. Before this, once in the while the mouse pointer was not sensitive (but pressing the keyboard can restore normal).
It is worth noting that waking from sleep can restore working.

kprinssu, 2020-05-20:
Can you upload a screen capture of the problem? It's kind of hard to make out the issue from your description.

Jie2GG, 2020-05-20:
> Can you upload a screen capture of the problem? It's kind of hard to make out the issue from your description.

Since this problem appears very few times, it may take me a long time to reproduce the problem and take a screenshot. (It only happened once since I upgraded to VoodooI2C 2.4.2).

kprinssu, 2020-05-22:
Please try to capture the behaviour via a screen recording if you can. It has hard to understand the problem from your description.

Jie2GG, 2020-05-26:
> Please try to capture the behaviour via a screen recording if you can. It has hard to understand the problem from your description.

Hi, I via a screen recording get VoodooI2C stop working of situation. 
This time, FingerMgmt detected two fingers resting on the trackpad. (in fact, there is no finger on the Touchpad).

*  [Video.zip](https://github.com/VoodooI2C/VoodooI2C/files/4679753/Video.zip)

kprinssu, 2020-05-26:
It looks like the driver just stopped reporting the gestures half way through. Looks to be another code issue with VoodooI2CHID.

Jie2GG, 2020-05-26:
> It looks like the driver just stopped reporting the gestures half way through. Looks to be another code issue with VoodooI2CHID.

There is one notable detail. When I just opened FingerMgmt, the touchpad did not show my finger until I touched the touchpad and the above picture appeared

Jie2GG, 2020-05-26:
> It looks like the driver just stopped reporting the gestures half way through. Looks to be another code issue with VoodooI2CHID.

If you need me to provide other relevant information, please contact me directly

kprinssu, 2020-05-26:
I would need you need to run debug versions of VoodooI2C and VoodooI2CHID (one which prints out the finger positions). I suspect it's changes to VoodooGPIO that is causing the issue.

Jie2GG, 2020-05-26:
> I would need you need to run debug versions of VoodooI2C and VoodooI2CHID (one which prints out the finger positions). I suspect it's changes to VoodooGPIO that is causing the issue.
> 
> @Jie2GG off hand, can you test these kexts:
> 
> [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4681894/Archive.zip)
> 
> Those are built from master and are the latest versions of VoodooI2C and VoodooI2CHID.

OK, No problem. But how do I provide that information?

kprinssu, 2020-05-26:
For the time being, please use the above and then wait and see if the issue pops up. Please confirm if the issue is fixed or not while using the above kexts.

Jie2GG, 2020-05-26:
Ok, I have used the debug version VoodooI2C + VoodooI2CHID and clear system log. If any of exceptions, I will commit my system log.

kprinssu, 2020-05-26:
There are no debug log prints in the above kexts, I just want to check if the latest version of master fixes your issue.

Jie2GG, 2020-05-26:
> There are no debug log prints in the above kexts, I just want to check if the latest version of master fixes your issue.

Thank. I will soon test result.

Jie2GG, 2020-05-27:
> There are no debug log prints in the above kexts, I just want to check if the latest version of master fixes your issue.

Hi, After testing, it was found that debug version are still issues

Bronfentrinker, 2020-05-29:
I have the same problem. On similar hardware (Asus x509fa i7-8565u, touchpad elan1200). I tried the kext from the archive, but the problem remained.

kprinssu, 2020-06-04:
@Jie2GG does this issue occur on 2.3?+

Bronfentrinker, 2020-06-04:
> @Jie2GG does this issue occur on 2.3 of VoodooI2C?+
The issue is still with version 2.4.3

hieplpvip, 2020-06-05:
Same for me

Jie2GG, 2020-06-05:
@kprinssu This issues occur on version 2.2 of VoodooI2C+

iDanielSouza, 2020-06-23:
I'm using macOS Catalina 10.15.5 and this works with my trackpad:
Try to install in your clover folder the kext VoodooI2CHID.kext from the release 2.1.6 and VoodooI2C.kext from the latest release. 
This issue doesn't occur to me since I did it 3 days ago.
(My laptop: Asus Vivobook X510UR-BQ378T/Core i5 8250U)

Bronfentrinker, 2020-06-23:
> I'm using macOS Catalina 10.15.5 and this works with my trackpad:
> Try to install in your clover folder the kext VoodooI2CHID.kext from the release 2.1.6 and VoodooI2C.kext from the latest release. [...]
not working for me.  with version 2.1.6 the touchpad does not work at all

soffyo, 2020-07-17:
This is happening for me too (asus Zenbook UX331UN, ELAN 1200). GPIO or polling mode bring the same problem, the trackpad becomes unresponsive at random times. It works again after sleep.

iDanielSouza, 2020-07-24:
Is the issue still happening on Catalina 10.15.6?

Fradri, 2020-08-06:
Same issue for me with ELAN1200. (On Asus Zenbook UX330)
Trackpad randomly stops working, usually a few minutes after starting or sleep.
It lasts 2-3 minutes, i wait, and then it becomes operational again ...

Yesterday I updated to Catalina 10.15.6, for the moment no freezing problem.
But no gesture support with the last version of VoodooI2C (2.4.3). (No scrolling, right clic with two finger ...)

Goshin, 2020-08-06:
Maybe a hardware-specific problem.
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1778087

mishurov, 2020-09-23:
Same issue for me with ELAN1200. (On Asus Zenbook UX310). Polling mode.

Randomly not loading during boot phase was solved with this patch, it allows using the latest version:
https://github.com/VoodooI2C/VoodooI2C/issues/319#issuecomment-646437623

Jie2GG, 2020-09-25:
Yes, ‘ELAN1200’ touchpad working in polling also stop.

technoeunuch, 2021-01-10:
I really wish this one could be sorted out, those random timeouts are really getting on my nerves. Had to use a backup optical mouse most of the time.

mishurov, 2021-01-10:
@technoeunuch the same story, I bought a Microsoft Bluetooth Mobile Mouse 3600 to don't interrupt work during these blackouts.

ben9923, 2021-01-10:
Does it still happen with latest VoodooI2C + VoodooI2CHID?

Jie2GG, 2021-01-10:
Yes, latest VoodooI2C + VoodooI2CHID still will happen this issues.

> Does it still happen with latest VoodooI2C + VoodooI2CHID?

Jie2GG, 2021-01-10:
> > Does it still happen with latest VoodooI2C + VoodooI2CHID?
> 
> ~Yes, latest VoodooI2C + VoodooI2CHID still will happen this issues.~
> Sorry, I didn't notice the latest version 2.6.3, I am currently updating to this version for testing

@ben9923 The latest version 2.6.3 has been tested and there is still a risk of stopping, which usually occurs after the touchpad is before it stops working.

nickcname13241, 2021-01-31:
Hello all.
Asus Zenbook ux410, gpio mode.

It's enough 1-2 min with 5 fingers on touchpad to get a freeze.

Here is 3 video with 5 finger and freeze - https://dropmefiles.com/pkXuH

Sometimes after freeze i got that messages. And whole system is working bad.

`test@MBP-test ~ % sudo dmesg | grep -i voodoo
VoodooI2CControllerDriver::pci8086,9d61 Timeout waiting for bus to accept transfer request
VoodooI2CControllerDriver::pci8086,9d61 Timed out waiting for bus to change state
VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
[IGFB][ERROR  ]  updateSliceConfig Requesting Invalid Slice Count[IGFB][ERROR  ]  updateSliceConfig Requesting Invalid EU CountVoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
[IGFB][ERROR  ]  updateSliceConfig Requesting Invalid Slice Count[IGFB][ERROR  ]  updateSliceConfig Requesting Invalid EU CountVoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy`

kprinssu, 2021-01-31:
@ nickcname13241 Please do not comment on an existing issue unless it's related to the original conversation at hand.

pamrish, 2021-02-05:
Same problem for me on osx 10.15.3
My main system is ubuntu 20.10. The touchpad works perfectly on it out of the box.

kprinssu, 2021-02-05:
@pamrish The issue is with the hardware and it seems like there are not any fixes within Linux kernel module. Hence, we don't have a fix.

Please avoid commenting on this issue if you do not have any issues or anything else to contribute to the discussion at hand.

mishurov, 2021-02-05:
It isn't a hardware issue. It works fine in Linux and in Windows. The issue with random halts is only in MacOS. Still happens with the latest Voodoo by the way.

kprinssu, 2021-02-05:
I've been tracking the issue here, https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1778087

The touchpad randomly stopping/disconnecting seems to be a common problem. 

The fix proposed was https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=670784fb4ebe54434e263837390e358405031d9e. 

I didn't look at the post times, but the lack of new complaints on that issue seems to suggest that the regressions have been fixed.

@mishurov I'll attempt to add the changes above in test build. Would you mind testing it?

Jie2GG, 2021-02-05:
> I've been tracking the issue here, https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1778087
>
> The touchpad randomly stopping/disconnecting seems to be a common problem.
>
> The fix proposed was [...]

No problem. I participated in the test

mishurov, 2021-02-05:
@kprinssu my primary system is Debian, I had no issues with disconnecting in Linux whatsoever. It has some other subtle issues, irrelevant for the topic, but has never random stops. I had no need to reload the module or so.

Yet in MacOS it stops working, becomes irresponsive for a while and then returns to life again. So I have to turn temporarily a BT mouse and use it until the touchpad wakes up.

Goshin, 2021-02-05:
@kprinssu That fix seems to be already present in VoodooGPIO.
https://github.com/VoodooI2C/VoodooGPIO/blob/master/VoodooGPIO/VoodooGPIO.cpp#L318-L321

And @Jie2GG reported in https://github.com/VoodooI2C/VoodooI2C/issues/321#issuecomment-698689982 that polling mode is also problematic, so it might not be GPIO related.

kprinssu, 2021-02-05:
@Goshin Yes, you're right, but few things:

- VoodooI2C's sub module is not up to date in regards to VoodooGPIO
- VoodooGPIO has a partial fix, we need to set the is_offset per chipset (Sunrise, Cannonlake, Icelake all seem to have different offsets)

See:
https://github.com/torvalds/linux/blob/master/drivers/pinctrl/intel/pinctrl-icelake.c#L38 vs https://github.com/VoodooI2C/VoodooGPIO/blob/master/VoodooGPIO/IceLake-LP/VoodooGPIOIceLakeLP.hpp#L35

Goshin, 2021-02-05:
@kprinssu Yes, I agree. It will be better to use `community->is_offset` there.

pamrish, 2021-02-05:
> @pamrish The issue is with the hardware and it seems like there are not any fixes within Linux kernel module. Hence, we don't have a fix.

I have been using ubuntu 20.10 every day since November 2020. In all these months, I have not encountered any problems with the touchpad. I have not tested Windows, but I do not think that there are problems with such thing as touchpad, otherwise there would be a tons of bad reviews on the forums.

jjsmith92, 2021-04-05:
The problem was solved by links and patches from this thread https://bugzilla.kernel.org/show_bug.cgi?id=200663
I applied all patches that i have found related to elan touchpad (it was about ~5 patches). And the problem is gone, no more disconnects for ~12 hours and the touchpad begin works better.
Fun fact: the 5-finger bug has also been was on a Mac, but no one has reported it. Hint: you need to make chips with 5 fingers for 1-2 minutes and the touchpad will disc...
https://dropmefiles.com/JGHkM
In the archive there is a compiled kexts and source code. It is written very dirty (but it works), so whoever wants can test and then rewrite  and make PR.

kprinssu, 2021-04-05:
@jjsmith92 Can you please create a PR? This has been a longstanding issue and I'd very much would like to get this fixed.

mishurov, 2021-04-05:
@kprinssu No, don't close the issue. The fix is only for Coffee Lake and is related to GPIO. Sky Lake and Kaby Lake work well in Linux, via IRQ, in MacOS it supports only the polling mode and the problem still exists.

kprinssu, 2021-04-05:
> @kprinssu No, don't close the issue. The fix is only for Coffee Lake and is related to GPIO. Sky Lake and Kaby Lake work well in Linux, via IRQ, in MacOS it supports only the polling mode and the problem still exists.

I was not planning on closing this issue until it's been completely fixed.

ben9923, 2021-04-10:
@jjsmith92 Pushing the code would be great too, so we can see the diff rather than searching the code manually for changes.
The thread you mentioned references a couple of changes, are you sure all of them are actually needed? Some are already in VoodooI2C anyway.

DanDV4l, 2021-07-11:
I was trying to fix the issue and looks like it worked. Heres the solution that worked for me: (Asus Vivobook X510UR/i5 8250U)

- In recovery mode, disabled SIP with the command in the terminal: csrutil disable
- restart and following a tutorial found in YouTube(mounting "/" on a folder to allow modifications), I placed the latest version of VoodooI2C.kext and VoodooI2CHID.kext in S/L/E.
- I recommend to rebuild kext cache after this
- In my efi folder, added two files in OC/ACPI: SSDT-Trackpad.aml, SSDT-OSYS.aml and activate them using OC Editor.
- Disabled the SSDT-I2CBUS.aml that is in the same folder.
- Save configurations and restart the OS.

I'll attach my EFI folder here for who wants more details about it and try the solution. 
> Please, change the serial numbers in the SMBIOS section. Running fine as MacBook Pro 15,2.
> The only issue with me now is the trackpad stopping to work for about for about 20~40 seconds after fresh starting of OS, but it's rare.
> Testing about 1 week with macOS Big Sur 11.4
Asus Vivobook X510UR-BQ378T / Core i5 8250U / 12GB DDR4 / SSD M.2 120GB / GeForce 930MX / Trackpad Elan 1200
 
EFI folder: 
https://drive.google.com/file/d/1QF-t6V7btN-8faqSgOF5JBUByapFyLtL/view?usp=sharing

Sorry for bad English.

mishurov, 2021-07-12:
It logs
`register name: com.apple.AppleMultitouchTrackpadDomainDidChangeNotification`
[touch.log](https://github.com/VoodooI2C/VoodooI2C/files/6802632/touch.log)

astro-lee-k, 2021-08-04:
I have the same issue with my trackpad on ZenBook Pro Duo 15 UX582L.

TSLARoadster, 2021-08-20:
> The problem was solved by links and patches from this thread https://bugzilla.kernel.org/show_bug.cgi?id=200663
> I applied all patches that i have found related to elan touchpad (it was about ~5 [...]
@jjsmith92 can you reupload the source and/or compiled kexts (preferably to github)? dropmefiles deleted the files as it "exceeded the download limit"

kprinssu, 2022-02-27:
Can anybody with an ELAN1200 please test this with 2.7? I suspect we still have the issue where the device fails to wake from sleep.

Jie2GG, 2022-02-28:
> Can anybody with an ELAN1200 please test this with 2.7? I suspect we still have the issue where the device fails to wake from sleep.
Unwakeable means the trackpad doesn't work when you wake up from sleep?

kprinssu, 2022-02-28:
> > Can anybody with an ELAN1200 please test this with 2.7? I suspect we still have the issue where the device fails to wake from sleep.
> 
> 
> 
> Unwakeable means the trackpad doesn't work when you wake up from sleep?
> 
> 

Can you please check if the original issue still persists with 2.7?

Jie2GG, 2022-02-28:
> > > Can anybody with an ELAN1200 please test this with 2.7? I suspect we still have the issue where the device fails to wake from sleep.
> > 
> > 
> > Unwakeable means the trackpad doesn't work when you wake up from sleep?
> 
> Can you please check if the original issue still persists with 2.7?

I issues is stop working after a period of useing. but it working fine after sleep awakens. There are this issues of Voodoo2.3.

kurtmorris, 2022-03-14:
I have the ELAN1200 on my Asus UX533F, I still have the periodic freezing issue. My trackpad seems to wake from sleep with no issue. When I upgraded to 2.7 from 2.6.5 I experienced no secondary click functionality, all fingers appear on the FingerMgmt app and other gestures seemed to work. Sometimes on 2.6.5 secondary clicking and clicking would register as multiple separate clicks, although after another reboot it seems to subside.

bilijp153, 2022-09-25:
ELAN1200  在（机型 ASUS FL8700JP 处理器 i7-1065G7，GPIO中断模式  EFI地址:https://github.com/bilijp153/ASUS-VivoBook-FL8700JP-icelake-1065G7-Hackintosh） 上  如果多指触控久了会出现无法使用，睡眠唤醒或重启后才恢复正常

JustinCase224, 2022-10-26:
I can also confirm, personally using a asus rog gl703vd, the touchpad when using the voodooi2chid.kext the track pad works but barely tracks any movement then stops working entirely untill sleep

[37 more replies]