VoodooI2C/VoodooI2C #325: Touch display works for a few seconds then stops with GPIO pinning.

fewtarius, 2020-05-21:
**Describe the bug**
With the latest commits in master, my GPIO touch display stops working within 1-2 seconds of a touch event.  It remains non-responsive until the system is rebooted.  The problem does not occur if the touch display is in polling mode but that causes this system to consume around 3w of additional power unnecessarily which was still an improvement over 2.4.2 which was not energy friendly at-all.  I think that's related to polling not allowing the CPU to go into deep sleep.

To recreate it, I can simply enable GPIO on the touch display and touch the display.  It seems like if I'm fast enough I can touch it more than once before I lose it.

[image]
[image]
[image]

Happy to provide any additional data that you need.  I'll also rebuild with debug enabled to see if I can get any more data into the logs.

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors

Yes.

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.

Yes.

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.

Yes, this may be related to the workloop changes happening in #282 but I'm not completely certain.

**System Environment**
Please answer:
macOS Catalina 10.15.4
Latest VoodooI2C built from master.

3. What is your hardware configuration?

Yoga C740
- i5-10210u
- 8GB RAM
- OpenCore 0.5.7
- Microsoft Elan Precision Touchpad
- Wacom Touch Display

**Troubleshooting Archive**

My DSDT is patched via a few renames in my OpenCore config.plist paired with SSDTs where needed.

[Clean DSDT.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/4664075/DSDT.aml.zip)
[OC modified DSDT.zip](https://github.com/VoodooI2C/VoodooI2C/files/4664084/OC.modified.DSDT.zip)

```
$ log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 14:11:11.182939-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 14:11:11.182947-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 14:11:11.182949-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 14:11:11.182995-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 14:11:11.183048-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 14:11:11.183127-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-21 14:11:18.330175-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 14:11:18.330184-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 14:11:25.367500-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 14:11:25.367612-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
```

**Additional context**
Add any other context about the problem here.

fewtarius, 2020-05-21:
Debug build output doesn't seem to shed any additional light.

```
$ log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 14:55:23.227087-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 14:55:23.227096-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 14:55:23.227099-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 14:55:23.227138-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 14:55:23.227172-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 14:55:29.040108-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 14:55:29.040119-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 14:55:35.546799-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 14:55:35.546922-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 14:55:48.658532-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27083490764
2020-05-21 14:55:48.761732-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27186693144
2020-05-21 14:55:48.883080-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27308038148
2020-05-21 14:55:48.970061-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27395013320
2020-05-21 14:55:49.000024-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27424980825
2020-05-21 14:55:49.077275-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27502232567
2020-05-21 14:55:49.159003-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27583955005
2020-05-21 14:55:49.210455-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27635412985
2020-05-21 14:55:49.255236-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27680191834
2020-05-21 14:55:49.264888-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27689834436
2020-05-21 14:55:49.304708-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27729664805
2020-05-21 14:55:49.338760-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27763700598
2020-05-21 14:55:49.448435-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27873396383
2020-05-21 14:55:49.527781-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 27952734528
2020-05-21 14:55:55.959018-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 34383983530
2020-05-21 14:55:56.066352-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 34491312851
2020-05-21 14:55:56.183680-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 34608638534
2020-05-21 14:55:56.267419-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 34692349831
2020-05-21 14:55:56.381075-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 34806035229
2020-05-21 14:55:56.486314-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 34911268685
2020-05-21 14:55:56.972668-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 35397613199
2020-05-21 14:55:57.087575-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 35512530512
2020-05-21 14:55:58.934550-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 37359508566
2020-05-21 14:55:59.005985-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 37430936668
2020-05-21 14:56:06.666526-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 45091456190
2020-05-21 14:56:06.740805-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 45165767186
2020-05-21 14:56:07.017884-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 45442834871
2020-05-21 14:56:07.081749-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 45506695299
[... cut ...]
2020-05-21 14:56:21.630793-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 60055758160
2020-05-21 14:56:23.102066-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 61527021830
2020-05-21 14:56:23.215195-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 61640125199
2020-05-21 14:56:23.402960-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 61827876295
```

Goshin, 2020-05-21:
@fewtarius have you build with the latest VoodooGPIO? You will need to first update the submodule to its upstream master manually.

fewtarius, 2020-05-21:
git pull --recursive won't grab it?  Did not know that, let me test.

fewtarius, 2020-05-21:
Well this is interesting, updated the submodules from upstream master and on reboot I don't have any input at all from the touchpad or display.

```
$ log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 15:19:50.707188-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 15:19:50.707195-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 15:19:50.707197-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 15:19:50.707235-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 15:19:50.707265-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 15:19:56.379252-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::MSFT0001 Timeout waiting for device to complete host initiated reset
2020-05-21 15:19:57.109074-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::MSFT0001 Putting device into Precision Touchpad Mode
2020-05-21 15:19:57.119865-0400  localhost kernel[0]: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
2020-05-21 15:19:58.194860-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 15:19:58.194871-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 15:20:04.799570-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 15:20:04.801376-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 15:20:29.046632-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 39785349023
2020-05-21 15:20:29.121329-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 39860206672
2020-05-21 15:20:29.451682-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 40190556593
2020-05-21 15:20:29.565729-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 40304572473
2020-05-21 15:20:31.370575-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::keyPressed = 42109445219
```

[image]
[image]

Goshin, 2020-05-21:
@fewtarius can you confirm with this build?
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4664345/Archive.zip)

fewtarius, 2020-05-21:
Confirmed, my touchpad and touch display do not work with that build.  I also had a couple of KPs starting up.

```
panic(cpu 0 caller 0xffffff800e64c33a): Kernel trap at 0xffffff7f9274e496, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000024a14000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff803370fe00, RCX: 0xffffff800f088138, RDX: 0x0000000000000001
RSP: 0xffffff812d0fbd10, RBP: 0xffffff812d0fbd30, RSI: 0xffffff800f0232b8, RDI: 0x0000000000000000
R8:  0x0000000000000001, R9:  0x0000000000000004, R10: 0x0000000000000001, R11: 0x0000000000000000
R12: 0xffffff7f927553eb, R13: 0xffffff812d0fbd48, R14: 0x00000001edbcb1e3, R15: 0x0000000000010101
RFL: 0x0000000000010246, RIP: 0xffffff7f9274e496, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff812d0fb770 : 0xffffff800e5215cd 
0xffffff812d0fb7c0 : 0xffffff800e65a405 
0xffffff812d0fb800 : 0xffffff800e64bf8e 
0xffffff812d0fb850 : 0xffffff800e4c7a40 
0xffffff812d0fb870 : 0xffffff800e520c97 
0xffffff812d0fb970 : 0xffffff800e521087 
0xffffff812d0fb9c0 : 0xffffff800ecc2d8c 
0xffffff812d0fba30 : 0xffffff800e64c33a 
0xffffff812d0fbbb0 : 0xffffff800e64c038 
0xffffff812d0fbc00 : 0xffffff800e4c7a40 
0xffffff812d0fbc20 : 0xffffff7f9274e496 
0xffffff812d0fbd30 : 0xffffff7f9274f600 
0xffffff812d0fbd80 : 0xffffff7f8f2eb5a0 
0xffffff812d0fbe10 : 0xffffff7f8f2ea3ad 
0xffffff812d0fbe50 : 0xffffff7f92752dc2 
0xffffff812d0fbe80 : 0xffffff800ec303a5 
0xffffff812d0fbee0 : 0xffffff800ec2ec7d 
0xffffff812d0fbf30 : 0xffffff800ec2d54e 
0xffffff812d0fbf70 : 0xffffff800ec2cb46 
0xffffff812d0fbfa0 : 0xffffff800e4c713e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[F3387125-6876-392D-9FE9-A973FD9763A0]@0xffffff7f8f2da000->0xffffff7f8f379fff
            dependency: com.apple.iokit.IOReportFamily(47)[28F06131-E9AB-3980-9175-6B4EAB4326EE]@0xffffff7f8f2d3000
         com.alexandred.VoodooI2CHID(1.0)[26ED2D8B-D846-38F9-BC08-C402D658D57E]@0xffffff7f9274c000->0xffffff7f9275efff
            dependency: com.alexandred.VoodooI2C(2.4.2)[A186B93A-7992-3199-9CD8-7EBAB462C9D6]@0xffffff7f92710000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2F2C33C8-2BFA-3490-9ACC-1C41AC87BC9A]@0xffffff7f8f64c000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[F3387125-6876-392D-9FE9-A973FD9763A0]@0xffffff7f8f2da000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
19E2269

Kernel version:
Darwin Kernel Version 19.4.0: Wed Mar  4 22:47:48 PST 2020; root:xnu-6153.101.6~16/RELEASE_X86_64
Kernel UUID: 7559A1A8-B300-3BC5-900F-5ED91F3197C7
Kernel slide:     0x000000000e200000
Kernel text base: 0xffffff800e400000
__HIB  text base: 0xffffff800e300000
System model name: MacBookPro15,4 (Mac-53FDB3D8DB8CA971)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 8283727068

```

Logs:

```
$ !log
log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 15:39:20.655672-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 15:39:20.655680-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 15:39:20.655684-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 15:39:20.655751-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 15:39:20.655779-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 15:39:26.494055-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::MSFT0001 Timeout waiting for device to complete host initiated reset
2020-05-21 15:39:26.646307-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::MSFT0001 Putting device into Precision Touchpad Mode
2020-05-21 15:39:26.657383-0400  localhost kernel[0]: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
2020-05-21 15:39:27.012515-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 15:39:27.012523-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 15:39:33.689443-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 15:39:33.689553-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
```

Goshin, 2020-05-21:
@fewtarius can you provide your current ioreg?

fewtarius, 2020-05-21:
Attached.

[slartibartfast.zip](https://github.com/VoodooI2C/VoodooI2C/files/4664425/slartibartfast.zip)

fewtarius, 2020-05-21:
Cannot seem to recreate that panic, but I'll leave keepsyms=1 until I can.

fewtarius, 2020-05-21:
Here's an IOReg from the display being in polling mode.  Interestingly your kexts work fine if the touchpad is in GPIO mode, and the display is in polling mode.

[slartibartfast.zip](https://github.com/VoodooI2C/VoodooI2C/files/4664459/slartibartfast.zip)

Goshin, 2020-05-21:
@fewtarius can you provide the log of loading kext with `kextload`? The logs above are incomplete.

fewtarius, 2020-05-21:
```
slartibartfast:Archive # kextutil VoodooI2C.kext/
Kext with invalid signature (-67050) allowed: <OSKext 0x7f903d430f60 [0x7fff8c4978c0]> { URL = "file:///Users/fewtarius/Downloads/Archive/VoodooI2C.kext/", ID = "com.alexandred.VoodooI2C" }
Kext with invalid signature (-67050) allowed: <OSKext 0x7f903d605c80 [0x7fff8c4978c0]> { URL = "file:///Users/fewtarius/Downloads/Archive/VoodooI2C.kext/Contents/PlugIns/VoodooI2CServices.kext/", ID = "com.alexandred.VoodooI2CServices" }
Kext with invalid signature (-67050) allowed: <OSKext 0x7f903d508520 [0x7fff8c4978c0]> { URL = "file:///Users/fewtarius/Downloads/Archive/VoodooI2C.kext/Contents/PlugIns/VoodooGPIO.kext/", ID = "org.coolstar.VoodooGPIO" }
Code Signing Failure: code signature is invalid
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        Custom MT2 Personality

Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        Custom MT2 Personality

Disabling KextAudit: SIP is off
Invalid signature -67050 for kext <OSKext 0x7f903d430f60 [0x7fff8c4978c0]> { URL = "file:///Users/fewtarius/Downloads/Archive/VoodooI2C.kext/", ID = "com.alexandred.VoodooI2C" }
slartibartfast:Archive # kextutil VoodooI2CHID.kext
Kext with invalid signature (-67050) allowed: <OSKext 0x7fb6d8c30f60 [0x7fff8c4978c0]> { URL = "file:///Users/fewtarius/Downloads/Archive/VoodooI2CHID.kext/", ID = "com.alexandred.VoodooI2CHID" }
/Users/fewtarius/Downloads/Archive/VoodooI2CHID.kext - dependency 'com.alexandred.VoodooI2C' not found.
/Users/fewtarius/Downloads/Archive/VoodooI2CHID.kext - dependency 'com.alexandred.VoodooI2C' not found.
/Users/fewtarius/Downloads/Archive/VoodooI2CHID.kext - dependency 'com.alexandred.VoodooI2C' not found.
Code Signing Failure: code signature is invalid
Dependency Resolution Failures:
    No kexts found for these libraries:
        com.alexandred.VoodooI2C

Diagnostics for /Users/fewtarius/Downloads/Archive/VoodooI2CHID.kext:
Dependency Resolution Failures:
    No kexts found for these libraries:
        com.alexandred.VoodooI2C

slartibartfast:Archive #
```

slartibartfast:Archive # !log
log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 16:22:54.549094-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 16:22:54.549108-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 16:22:54.549111-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 16:22:54.549177-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 16:22:54.549228-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 16:22:54.549696-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-05-21 16:22:54.549762-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-05-21 16:22:54.549940-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-05-21 16:22:54.549942-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-05-21 16:22:54.549947-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Current CPU is Comet Lake, patching...
2020-05-21 16:22:54.549949-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Current CPU is Comet Lake, patching...
2020-05-21 16:22:54.549961-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Successfully patched!
2020-05-21 16:22:54.549963-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Successfully patched!
2020-05-21 16:22:54.549992-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-21 16:22:54.549994-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-05-21 16:22:54.550011-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-05-21 16:22:54.550706-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-05-21 16:22:54.550713-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:22:54.550872-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-05-21 16:22:54.550880-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:22:54.551308-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Got bus configuration values
[... cut ...]
2020-05-21 16:22:54.551348-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found I2C device: WCOM51EF
2020-05-21 16:22:54.551440-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-21 16:22:54.551588-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::VoodooI2CDeviceNub Warning: Incompatible APIC interrupt pin (0x70 > 0x2f) and no GPIO interrupts found; if your chosen satellite implements polling then VoodooI2CDeviceNub will run in polling mode.
```

Will reboot now with the other pin (this is the default) and grab those.

Goshin, 2020-05-21:
```
sudo kextload -d VoodooI2C.kext VoodooI2CHID.kext
```

fewtarius, 2020-05-21:
oof, thanks.  With the default pinning:

```
lartibartfast:Archive $ sudo kextload -d VoodooI2C.kext VoodooI2CHID.kext
Password:
slartibartfast:Archive $ !log
log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 16:32:12.784120-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 16:32:12.784150-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 16:32:12.784159-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 16:32:12.784263-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 16:32:12.784334-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 16:32:12.785733-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-05-21 16:32:12.785735-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-05-21 16:32:12.786265-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-05-21 16:32:12.786276-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-05-21 16:32:12.786277-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Current CPU is Comet Lake, patching...
2020-05-21 16:32:12.786291-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Current CPU is Comet Lake, patching...
2020-05-21 16:32:12.786294-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Successfully patched!
2020-05-21 16:32:12.786309-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Successfully patched!
2020-05-21 16:32:12.786344-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-05-21 16:32:12.786378-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-05-21 16:32:12.786442-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-21 16:32:12.787461-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-05-21 16:32:12.787475-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:32:12.788142-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Got bus configuration values
2020-05-21 16:32:12.788314-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Publishing device nubs
2020-05-21 16:32:12.788331-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found I2C device: WCOM51EF
2020-05-21 16:32:12.788422-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-05-21 16:32:12.788435-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:32:12.789447-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-21 16:32:12.789449-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Got bus configuration values
2020-05-21 16:32:12.789486-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Publishing device nubs
2020-05-21 16:32:12.789498-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found I2C device: MSFT0001
2020-05-21 16:32:12.789841-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-21 16:32:13.292112-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-21 16:32:13.292120-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-21 16:32:13.292143-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-21 16:32:13.292154-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-21 16:32:13.397296-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM51EF Device initiated reset accomplished
2020-05-21 16:32:13.561443-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 16:32:13.561453-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 16:32:13.574138-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
slartibartfast:Archive $
```

fewtarius, 2020-05-21:
With the alternate display pin:

```
slartibartfast:Archive $ sudo kextload -d VoodooI2C.kext VoodooI2CHID.kext
Password:
slartibartfast:Archive $ log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 16:40:25.396260-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 16:40:25.396290-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 16:40:25.396298-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 16:40:25.396396-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 16:40:25.396458-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 16:40:25.397857-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-05-21 16:40:25.397859-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-05-21 16:40:25.398336-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-05-21 16:40:25.398355-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Current CPU is Comet Lake, patching...
2020-05-21 16:40:25.398373-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Successfully patched!
2020-05-21 16:40:25.398396-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-05-21 16:40:25.398407-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Current CPU is Comet Lake, patching...
2020-05-21 16:40:25.398422-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-05-21 16:40:25.398424-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Successfully patched!
2020-05-21 16:40:25.398479-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-05-21 16:40:25.398630-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-21 16:40:25.399555-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-05-21 16:40:25.399568-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:40:25.399701-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-05-21 16:40:25.399715-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:40:25.400649-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Got bus configuration values
2020-05-21 16:40:25.400766-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Publishing device nubs
2020-05-21 16:40:25.400784-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found I2C device: WCOM51EF
2020-05-21 16:40:25.400943-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Got bus configuration values
2020-05-21 16:40:25.400991-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Publishing device nubs
2020-05-21 16:40:25.401007-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found I2C device: MSFT0001
2020-05-21 16:40:25.401830-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-21 16:40:25.401841-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-21 16:40:25.904363-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-21 16:40:25.904375-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-21 16:40:25.904433-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-21 16:40:25.904445-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-21 16:40:26.009733-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM51EF Device initiated reset accomplished
2020-05-21 16:40:26.173500-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 16:40:26.173522-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 16:40:26.191099-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
[... cut ...]
2020-05-21 16:40:32.156121-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::MSFT0001 Putting device into Precision Touchpad Mode
2020-05-21 16:40:32.168155-0400  localhost kernel[0]: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
2020-05-21 16:40:32.170965-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
slartibartfast:Archive $
```

fewtarius, 2020-05-21:
Hmm, it doesn't look like it's using the alternate pin.

```
            Name (SBFG, ResourceTemplate ()
            {
                GpioInt (Level, ActiveLow, Exclusive, PullDefault, 0x0000,
                    "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                    )
                    {   // Pin list
                        0x8600
                    }
            })
```

Goshin, 2020-05-21:
@fewtarius so you got the problem since v2.4.2? I am not 100% sure about the cause. Plz try this one. 
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4664638/Archive.zip)

fewtarius, 2020-05-21:
I'm pretty sure it was working fine on 2.4.2 but it's possible that it wasn't using GPIO on the display.  After I get these logs this I'll go test that.

I got a KP as soon as I loaded the kexts:

```
panic(cpu 0 caller 0xffffff800ec4c33a): Kernel trap at 0xffffff800f22f3bd, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000024f2d000, CR4: 0x00000000003626e0
RAX: 0xffffff804767c030, RBX: 0xffffff803c238300, RCX: 0xffffff803bf9ac00, RDX: 0x0000000000000000
RSP: 0xffffff8102635e50, RBP: 0xffffff8102635e60, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000120, R9:  0xffffff7f92cba600, R10: 0x0000000000000088, R11: 0x0000000000000001
R12: 0xffffff9144c46000, R13: 0x0000000000004000, R14: 0x0000000000000000, R15: 0x000000000000000e
RFL: 0x0000000000010002, RIP: 0xffffff800f22f3bd, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 1, VF: 5

Backtrace (CPU 0), Frame : Return Address
0xffffff81026358b0 : 0xffffff800eb215cd mach_kernel : _handle_debugger_trap + 0x49d
0xffffff8102635900 : 0xffffff800ec5a405 mach_kernel : _kdp_i386_trap + 0x155
0xffffff8102635940 : 0xffffff800ec4bf8e mach_kernel : _kernel_trap + 0x4ee
0xffffff8102635990 : 0xffffff800eac7a40 mach_kernel : _return_from_trap + 0xe0
0xffffff81026359b0 : 0xffffff800eb20c97 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff8102635ab0 : 0xffffff800eb21087 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff8102635b00 : 0xffffff800f2c2d8c mach_kernel : _panic + 0x54
0xffffff8102635b70 : 0xffffff800ec4c33a mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff8102635cf0 : 0xffffff800ec4c038 mach_kernel : _kernel_trap + 0x598
0xffffff8102635d40 : 0xffffff800eac7a40 mach_kernel : _return_from_trap + 0xe0
0xffffff8102635d60 : 0xffffff800f22f3bd mach_kernel : __ZN22IOInterruptEventSource24disableInterruptOccurredEPvP9IOServicei + 0x7d
0xffffff8102635e60 : 0xffffff7f92cb13fe org.coolstar.VoodooGPIO : __ZN10VoodooGPIO26intel_gpio_pin_irq_handlerEj + 0x96
0xffffff8102635ea0 : 0xffffff7f92cb16bb org.coolstar.VoodooGPIO : __ZN10VoodooGPIO17InterruptOccurredEPvP9IOServicei + 0x65
0xffffff8102635ed0 : 0xffffff800f251ea3 mach_kernel : __ZN27IOSharedInterruptController15handleInterruptEPvP9IOServicei + 0x153
0xffffff8102635f10 : 0xffffff7f9281da9d com.apple.driver.AppleAPIC : __ZN28AppleAPICInterruptController15handleInterruptEPvP9IOServicei + 0x11b
0xffffff8102635f50 : 0xffffff7f9181888c com.apple.driver.AppleACPIPlatform : __ZN23AppleACPIPlatformExpert23dispatchGlobalInterruptEi + 0x2e
0xffffff8102635f60 : 0xffffff7f91821547 com.apple.driver.AppleACPIPlatform : __ZN31AppleACPICPUInterruptController15handleInterruptEPvP9IOServicei + 0x6b
0xffffff8102635f80 : 0xffffff800ec4b7dd mach_kernel : _interrupt + 0x17d
0xffffff8102635fd0 : 0xffffff800eac7bed mach_kernel : _hndl_allintrs + 0x11d
0xffffff81228d39a0 : 0xffffff800eb78de1 mach_kernel : _zalloc_async + 0x5f1
0xffffff81228d3ad0 : 0xffffff800eb2e4b4 mach_kernel : _kalloc_canblock + 0xa4
0xffffff81228d3b40 : 0xffffff800f197757 mach_kernel : __ZN8OSObjectnwEm + 0x27
0xffffff81228d3b70 : 0xffffff800f1b5a5a mach_kernel : __ZN8OSNumber10withNumberEyj + 0x1a
0xffffff81228d3ba0 : 0xffffff800f269220 mach_kernel : __ZN10IOReporter18handleCreateLegendEv + 0x50
0xffffff81228d3bd0 : 0xffffff800f26dc88 mach_kernel : __ZN14IOReportLegend17addReporterLegendEP10IOReporterPKcS3_ + 0x38
0xffffff81228d3c10 : 0xffffff800f210ff6 mach_kernel : __ZN9IOService22addInterruptStatisticsEP25IOInterruptAccountingDatai + 0x2e6
0xffffff81228d3cb0 : 0xffffff800f22f13c mach_kernel : __ZN22IOInterruptEventSource4initEP8OSObjectPFvS1_PS_iEP9IOServicei + 0x14c
0xffffff81228d3cf0 : 0xffffff800f22fb94 mach_kernel : __ZN22IOInterruptEventSource20interruptEventSourceEP8OSObjectPFvS1_PS_iEP9IOServicei + 0x74
0xffffff81228d3d30 : 0xffffff7f92cf585c com.alexandred.VoodooI2CHID : __ZN18VoodooI2CHIDDevice11handleStartEP9IOService + 0x112
0xffffff81228d3d50 : 0xffffff7f8f8e794d com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice5startEP9IOService + 0x16d
0xffffff81228d3dc0 : 0xffffff7f92cf5a48 com.alexandred.VoodooI2CHID : __ZN18VoodooI2CHIDDevice5startEP9IOService + 0x16
0xffffff81228d3de0 : 0xffffff800f208d16 mach_kernel : __ZN9IOService14startCandidateEPS_ + 0xf6
0xffffff81228d3e40 : 0xffffff800f208851 mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0xad1
0xffffff81228d3ef0 : 0xffffff800f207bee mach_kernel : __ZN9IOService14doServiceMatchEj + 0x2de
0xffffff81228d3f50 : 0xffffff800f20a3a6 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x186
0xffffff81228d3fa0 : 0xffffff800eac713e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.driver.AppleACPIPlatform(6.1)[46E73EAB-914F-37FF-9DBF-AD27EF57CDEE]@0xffffff7f91811000->0xffffff7f918abfff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2F2C33C8-2BFA-3490-9ACC-1C41AC87BC9A]@0xffffff7f8fc4c000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[6F96761B-C218-30A8-BECC-1EAD32E53D47]@0xffffff7f8f531000
            dependency: com.apple.driver.AppleSMC(3.1.9)[08211E73-BC69-3541-8240-AD3FADB3A982]@0xffffff7f8fc5e000
         com.apple.driver.AppleAPIC(1.7)[DDAEF045-01A0-3B1E-99E7-F5B883F92B0A]@0xffffff7f9281c000->0xffffff7f92820fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[6F96761B-C218-30A8-BECC-1EAD32E53D47]@0xffffff7f8f531000
[... cut ...]
BSD process name corresponding to current thread: kernel_task
Boot args: keepsyms=1

Mac OS version:
19E2269

Kernel version:
Darwin Kernel Version 19.4.0: Wed Mar  4 22:47:48 PST 2020; root:xnu-6153.101.6~16/RELEASE_X86_64
Kernel UUID: 7559A1A8-B300-3BC5-900F-5ED91F3197C7
Kernel slide:     0x000000000e800000
Kernel text base: 0xffffff800ea00000
__HIB  text base: 0xffffff800e900000
System model name: MacBookPro15,4 (Mac-53FDB3D8DB8CA971)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 67895365058

```

fewtarius, 2020-05-21:
After the panic reboot the kexts loaded ok:

```
slartibartfast:Archive-2 $ sudo kextload -d VoodooI2C.kext VoodooI2CHID.kext
Password:
slartibartfast:Archive-2 $ log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 16:54:32.042310-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIOCannonLakeLP::start(IOService*)> VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 16:54:32.042325-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::start(IOService*)> VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 16:54:32.042328-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::start(IOService*)> VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 16:54:32.042407-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::start(IOService*)> VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 16:54:32.042457-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::start(IOService*)> VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 16:54:32.042928-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-05-21 16:54:32.042981-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-05-21 16:54:32.043222-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-05-21 16:54:32.043227-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-05-21 16:54:32.043228-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Current CPU is Comet Lake, patching...
2020-05-21 16:54:32.043232-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Current CPU is Comet Lake, patching...
2020-05-21 16:54:32.043239-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Successfully patched!
2020-05-21 16:54:32.043239-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Successfully patched!
2020-05-21 16:54:32.043261-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-05-21 16:54:32.043273-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-05-21 16:54:32.043410-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::setPowerState(unsigned long, IOService*)> VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-21 16:54:32.043868-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-05-21 16:54:32.043875-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:54:32.043923-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-05-21 16:54:32.043930-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
2020-05-21 16:54:32.044416-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Got bus configuration values
2020-05-21 16:54:32.044430-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Got bus configuration values
2020-05-21 16:54:32.044446-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Publishing device nubs
2020-05-21 16:54:32.044448-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Publishing device nubs
2020-05-21 16:54:32.044454-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found I2C device: WCOM51EF
2020-05-21 16:54:32.044456-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found I2C device: MSFT0001
2020-05-21 16:54:32.044858-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-21 16:54:32.044863-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-21 16:54:32.548439-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::registerInterrupt(int, OSObject*, void (*)(OSObject*, void*, IOService*, int), void*)> VoodooGPIOCannonLakeLP::Registering hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-21 16:54:32.548456-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::registerInterrupt(int, OSObject*, void (*)(OSObject*, void*, IOService*, int), void*)> VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-21 16:54:32.548838-0400  localhost kernel[0]: (VoodooGPIO) <VoodooGPIO`VoodooGPIO::registerInterrupt(int, OSObject*, void (*)(OSObject*, void*, IOService*, int), void*)> VoodooGPIOCannonLakeLP::Registering hardware pin 0x52 for GPIO IRQ pin 0x6E
[... cut ...]
2020-05-21 16:54:32.823283-0400  localhost kernel[0]: (VoodooI2CHID) <VoodooI2CHID`VoodooI2CTouchscreenHIDEventDriver::getFramebuffer()> VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 16:54:32.831715-0400  localhost kernel[0]: (VoodooI2CHID) <VoodooI2CHID`VoodooI2CMultitouchHIDEventDriver::setProperties(OSObject*)> VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
slartibartfast:Archive-2 $
```

No touchpad or touch display at any point while I've been loading these though.

fewtarius, 2020-05-21:
Yeah, it's working just fine on the release version with the default pinning.

Goshin, 2020-05-21:
The latest log shows everything is ok now. Can you try to put the kexts in your EFI and reboot to see if they work?

fewtarius, 2020-05-21:
Nothing when they're in the EFI.  Here's the log:

```
slartibartfast:~ $ !log
log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 17:07:16.711494-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-21 17:07:16.711502-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-21 17:07:16.711505-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-21 17:07:16.711567-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-21 17:07:16.711601-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-21 17:07:22.511964-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::MSFT0001 Timeout waiting for device to complete host initiated reset
2020-05-21 17:07:22.914473-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::MSFT0001 Putting device into Precision Touchpad Mode
2020-05-21 17:07:22.925435-0400  localhost kernel[0]: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
2020-05-21 17:07:24.097453-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 17:07:24.097462-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 17:07:30.745774-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 17:07:30.750051-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
slartibartfast:~ $
```

Goshin, 2020-05-21:
@fewtarius I just revert the workloop related commit. You can see if it helps.
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4664883/Archive.zip)

fewtarius, 2020-05-21:
Hmm, no it does not.  There's less in the log than before as well.  Were there other changes between 2.4.2 and now?  I didn't lose the touchpad until I pulled down the changes from GPIO master, what if we reverted that?

```
slartibartfast:~ $ kextstat | grep -i voodooi2c
   53    1 0xffffff7f84412000 0x10000    0x10000    com.alexandred.VoodooI2CServices (1) E1E325A5-365B-3FF0-96A8-CF330377D2BE <5 3>
   74    1 0xffffff7f84510000 0x2e000    0x2e000    com.alexandred.VoodooI2C (2.4.2) 8E03A734-00FE-33CC-B741-9BA8CC5332A1 <72 66 53 13 12 6 5 3 1>
  118    0 0xffffff7f8454c000 0x13000    0x13000    com.alexandred.VoodooI2CHID (1) 4424DD39-BE91-38A5-A30F-8068754C971D <74 66 12 6 5 3>
slartibartfast:~ $ log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-21 17:34:40.546002-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-21 17:34:40.546014-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-21 17:34:47.083692-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-21 17:34:47.083906-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
```

fewtarius, 2020-05-21:
If I revert VoodooGPIO 8e77a296cf205a7111dc29a499c3471ae8250aa9 back to e1728148549eb6cdc9fa90e774c193ea484793f0 my touchpad works fine, but my touch display does not.

Goshin, 2020-05-22:
@fewtarius which version works all well for both TP and TS in the gpio mode?

Goshin, 2020-05-22:
@fewtarius maybe this one will help.
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4667336/Archive.zip)

fewtarius, 2020-05-22:
Touchpad works, but no display input with this one.  I'll try and load them in the OS next to get data.  With 2.4.2, the touch display works ok but there's that click lag with the touchpad.  Prior to 2.4.2 I didn't have the device pinned, but I did test 2.4.1 and there's no input when in GPIO mode.

fewtarius, 2020-05-22:
Loading the kexts from within macOS, there's no input on either device.  Here are the logs.

```
slartibartfast:Archive $ sudo kextload -d VoodooI2C.kext VoodooI2CHID.kext
slartibartfast:Archive $ !log
log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-22 07:34:42.192722-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-22 07:34:42.192739-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-22 07:34:42.192743-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-22 07:34:42.192823-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-22 07:34:42.192867-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-22 07:34:42.193363-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-05-22 07:34:42.193363-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-05-22 07:34:42.193651-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-05-22 07:34:42.193652-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-05-22 07:34:42.193654-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Current CPU is Comet Lake, patching...
2020-05-22 07:34:42.193658-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Current CPU is Comet Lake, patching...
2020-05-22 07:34:42.193665-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Successfully patched!
2020-05-22 07:34:42.193666-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Successfully patched!
2020-05-22 07:34:42.193683-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-05-22 07:34:42.193695-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-05-22 07:34:42.193762-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-22 07:34:42.194137-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-05-22 07:34:42.194145-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
2020-05-22 07:34:42.194437-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Got bus configuration values
2020-05-22 07:34:42.194513-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Publishing device nubs
2020-05-22 07:34:42.194521-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found I2C device: WCOM51EF
2020-05-22 07:34:42.194731-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-22 07:34:42.194826-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-05-22 07:34:42.194832-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
2020-05-22 07:34:42.195026-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Got bus configuration values
2020-05-22 07:34:42.195043-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Publishing device nubs
2020-05-22 07:34:42.195047-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found I2C device: MSFT0001
2020-05-22 07:34:42.195205-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-22 07:34:42.697262-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-22 07:34:42.697266-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-22 07:34:42.697486-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-22 07:34:42.697489-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-22 07:34:42.801175-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM51EF Device sent a 0-length report
2020-05-22 07:34:42.802326-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM51EF Device initiated reset accomplished
2020-05-22 07:34:42.802905-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::MSFT0001 Device sent a 0-length report
2020-05-22 07:34:42.803948-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::MSFT0001 Device initiated reset accomplished
2020-05-22 07:34:42.948583-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::MSFT0001 Putting device into Precision Touchpad Mode
2020-05-22 07:34:42.959218-0400  localhost kernel[0]: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
2020-05-22 07:34:42.960857-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
```

Goshin, 2020-05-22:
@fewtarius I don't see anything strange in the log. Can you provide the log of loading v2.4.2 release?

fewtarius, 2020-05-22:
Logs from 2.4.2.  Loading the 2.4.2 kexts from the OS, my touchpad does not work...but my touch display does.

```
slartibartfast:VoodooI2C-2 $ sudo chown -R root:wheel *
Password:
slartibartfast:VoodooI2C-2 $ sudo chmod -R 755 *
slartibartfast:VoodooI2C-2 $ !450
sudo kextload -d VoodooI2C.kext VoodooI2CHID.kext
slartibartfast:VoodooI2C-2 $ !log
log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-22 07:40:19.337355-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-22 07:40:19.337426-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-22 07:40:19.337428-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-22 07:40:19.337508-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-22 07:40:19.337557-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-22 07:40:19.338071-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-05-22 07:40:19.338076-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-05-22 07:40:19.338344-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-05-22 07:40:19.338349-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Current CPU is Comet Lake, patching...
2020-05-22 07:40:19.338360-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Successfully patched!
2020-05-22 07:40:19.338383-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-05-22 07:40:19.338408-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-05-22 07:40:19.338413-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Current CPU is Comet Lake, patching...
2020-05-22 07:40:19.338414-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-22 07:40:19.338424-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Successfully patched!
2020-05-22 07:40:19.338448-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-05-22 07:40:19.338975-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-05-22 07:40:19.338982-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
2020-05-22 07:40:19.339186-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-05-22 07:40:19.339193-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
2020-05-22 07:40:19.339526-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Got bus configuration values
2020-05-22 07:40:19.339575-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Publishing device nubs
2020-05-22 07:40:19.339580-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found I2C device: WCOM51EF
2020-05-22 07:40:19.339591-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Got bus configuration values
2020-05-22 07:40:19.339721-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-22 07:40:19.339728-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Publishing device nubs
2020-05-22 07:40:19.339734-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found I2C device: MSFT0001
2020-05-22 07:40:19.339894-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-22 07:40:19.844814-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x54 for GPIO IRQ pin 0x70
2020-05-22 07:40:19.844825-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-22 07:40:20.249571-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::MSFT0001 Timeout waiting for device to complete host initiated reset
2020-05-22 07:40:20.251956-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM51EF Timeout waiting for device to complete host initiated reset
2020-05-22 07:40:20.659926-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::MSFT0001 Putting device into Precision Touchpad Mode
2020-05-22 07:40:20.667422-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-22 07:40:20.667431-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-22 07:40:20.674075-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-22 07:40:20.677299-0400  localhost kernel[0]: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
2020-05-22 07:40:20.688194-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
```

Goshin, 2020-05-22:
@fewtarius but you said the touchpad and touchscreen are ok with v2.4.2, besides the click issue?

fewtarius, 2020-05-22:
I rebooted with your Archive.zip kexts still in my EFI and the touch display did actually accept input for about 2 seconds.

Yes, that's correct, 2.4.2 works fine from the EFI except for the click issue.

marianopela, 2020-05-22:
> I rebooted with your Archive.zip kexts still in my EFI and the touch display did actually accept input for about 2 seconds.
> 
> Yes, that's correct, 2.4.2 works fine from the EFI except for the click issue.

I too have an MSFT0001 trackpad on my Lenovo Ideapad. Although I don't have a touchscreen, my trackpad is working for about a second immediately after getting to login screen, but then it stops until reboot, almost like your touchscreen does. I think we might be looking at related issues. 

CC @kprinssu @hasodikis

Goshin, 2020-05-22:
@fewtarius Do you get the pin error in the log with v2.4.2?

fewtarius, 2020-05-22:
I do not see any pin errors in the logs unless the display is in polling mode.  Everything that gets logged by 2.4.2 logs is what's shown [here](https://github.com/VoodooI2C/VoodooI2C/issues/325#issuecomment-632649234).

fewtarius, 2020-05-22:
Here is the log from 2.4.2 with the display in polling mode.
```
slartibartfast:VoodooI2C-2 $ !470
sudo kextload -d VoodooI2C.kext VoodooI2CHID.kext
Password:
slartibartfast:VoodooI2C-2 $ !log
log show --last boot --predicate 'process == "kernel" AND (eventMessage CONTAINS "VoodooI2C" OR eventMessage CONTAINS "VoodooGPIO")' --style syslog --source
Filtering the log data using "process == "kernel" AND (composedMessage CONTAINS "VoodooI2C" OR composedMessage CONTAINS "VoodooGPIO")"
Skipping info and debug messages, pass --info and/or --debug to include.
Timestamp                       (process)[PID]
2020-05-22 08:21:23.857133-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-05-22 08:21:23.857301-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-05-22 08:21:23.857309-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-05-22 08:21:23.857395-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-05-22 08:21:23.857493-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-05-22 08:21:23.858790-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-05-22 08:21:23.858801-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-05-22 08:21:23.859283-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-05-22 08:21:23.859295-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Current CPU is Comet Lake, patching...
2020-05-22 08:21:23.859317-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Successfully patched!
2020-05-22 08:21:23.859366-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-05-22 08:21:23.859384-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-05-22 08:21:23.859470-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-05-22 08:21:23.859480-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Current CPU is Comet Lake, patching...
2020-05-22 08:21:23.859493-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Successfully patched!
2020-05-22 08:21:23.859528-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-05-22 08:21:23.860641-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-05-22 08:21:23.860654-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
2020-05-22 08:21:23.861308-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Got bus configuration values
2020-05-22 08:21:23.861552-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Publishing device nubs
2020-05-22 08:21:23.861571-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e8 Found I2C device: MSFT0001
2020-05-22 08:21:23.861659-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-05-22 08:21:23.861673-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
2020-05-22 08:21:23.862367-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeLP
2020-05-22 08:21:23.862561-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Got bus configuration values
2020-05-22 08:21:23.862593-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Publishing device nubs
2020-05-22 08:21:23.862604-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,2e9 Found I2C device: WCOM51EF
2020-05-22 08:21:23.862982-0400  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::VoodooI2CDeviceNub Warning: Incompatible APIC interrupt pin (0x70 > 0x2f) and no GPIO interrupts found; if your chosen satellite implements polling then VoodooI2CDeviceNub will run in polling mode.
2020-05-22 08:21:23.864838-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM51EF Warning: Could not get interrupt event source, using polling instead
2020-05-22 08:21:24.069425-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::WCOM51EF Timeout waiting for device to complete host initiated reset
2020-05-22 08:21:24.228382-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active display
2020-05-22 08:21:24.228391-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::Got active framebuffer
2020-05-22 08:21:24.231437-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
2020-05-22 08:21:24.367730-0400  localhost kernel[0]: (VoodooGPIO) VoodooGPIOCannonLakeLP::Registering hardware pin 0x52 for GPIO IRQ pin 0x6E
2020-05-22 08:21:24.573421-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::MSFT0001 Timeout waiting for device to complete host initiated reset
2020-05-22 08:21:24.726779-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::MSFT0001 Putting device into Precision Touchpad Mode
2020-05-22 08:21:24.738835-0400  localhost kernel[0]: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
2020-05-22 08:21:24.741873-0400  localhost kernel[0]: (VoodooI2CHID) VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
slartibartfast:VoodooI2C-2 $
```

Goshin, 2020-05-22:
@fewtarius no idea for now, the log in https://github.com/VoodooI2C/VoodooI2C/issues/325#issuecomment-632646841 shows everything is fine.  In v2.4.2 release, there are some bugs of interrupt toggling and device resetting, as you can see the timeout error in the log. I suggest you use the touchscreen in polling mode with the latest version.

fewtarius, 2020-05-22:
Alright, will do for now.  Thanks for looking into it.

Goshin, 2020-05-24:
@fewtarius Can you test the touchscreen in GPIO mode and touchpad in polling mode?

fewtarius, 2020-05-25:
Unfortunately, the behavior doesn't seem to change.  Touchpad is working in polling mode, the touch display is not working at all.

[image]

[image]

Goshin, 2020-05-25:
So it could be a hardware-specific issue. Does it work normally on Linux?

fewtarius, 2020-05-25:
@Goshin yes, it works perfectly in Windows and in Linux.  Also works with VoodooI2C 2.4.2 as I had mentioned before.  I'll re-test the above scenario using 2.4.2 when I have the chance, today if possible.  In the interim, I've had it disabled via _STA which has significantly improved battery life.

marianopela, 2020-05-27:
@fewtarius Is your trackpad working in GPIO interrupts mode?

fewtarius, 2020-05-27:
@marianopela it does, yes.

marianopela, 2020-05-27:
> @marianopela it does, yes.

That's sick strange. Same trackpad device, same CPU, similar hardware (same Lenovo manufacturer), yet I can't figure out why it's not working for me. Are you using 0x70 pin?

Goshin, 2020-05-27:
@marianopela so your touchpad works well with the kexts in https://github.com/VoodooI2C/VoodooI2C/issues/323#issuecomment-632115743 ?

kprinssu, 2020-05-27:
@marianopela let's track down the issue here. @tiger511's changes do not have some of the upstream patches that were from @Goshin. Something that we merged in is causing the issue.

marianopela, 2020-05-27:
> @marianopela so your touchpad works well with the kexts in [#323 (comment)](https://github.com/VoodooI2C/VoodooI2C/issues/323#issuecomment-632115743) ?

Nope, unfortunately it doesn't. It works for 2 seconds after boot and 1 second after every sleep/wake though, which is very similar to what @fewtarius was describing the title of this issue, so it seems the be just working until it hangs for some reason

marianopela, 2020-05-27:
> @marianopela let's track down the issue here. @tiger511's changes do not have some of the upstream patches that were from @Goshin. Something that we merged in is causing the issue.

Sure. Can you suggest a way to debug this? Maybe by loading VoodooGPIO with log printing

kprinssu, 2020-05-27:
Easiest, way revert commits and build VoodooGPIO until it works. If it's not a VoodooGPIO, reset the project to master and then revert commits in VoodooI2CHID.

This is pretty much brute forcing will be extremely time consuming.

Goshin, 2020-05-27:
I suspect v2.4.2 just works by mistake. Try to revert https://github.com/VoodooI2C/VoodooGPIO/commit/e1728148549eb6cdc9fa90e774c193ea484793f0 only and keep the fix in https://github.com/VoodooI2C/VoodooGPIO/commit/8e77a296cf205a7111dc29a499c3471ae8250aa9 , and see if it works.

marianopela, 2020-05-27:
> I suspect v2.4.2 just works by mistake. Try to revert [VoodooI2C/VoodooGPIO@e172814](https://github.com/VoodooI2C/VoodooGPIO/commit/e1728148549eb6cdc9fa90e774c193ea484793f0) only and keep the fix in [VoodooI2C/VoodooGPIO@8e77a29](https://github.com/VoodooI2C/VoodooGPIO/commit/8e77a296cf205a7111dc29a499c3471ae8250aa9) , and see if it works.

To be honest, 2.4.2 does not “work” with interrupts mode (referring to it not working even for a second)
I’ll try reverting with your suggestions in a few hours, tomorrow, as I had troubles setting up the building environment

Goshin, 2020-05-27:
> To be honest, 2.4.2 does not “work” with interrupts mode (referring to it not working even for a second)
> I’ll try reverting with your suggestions in a few hours, tomorrow, as I had troubles setting [...]

Sry I forgot to ping Fewtarius. He said 2.4.2 works for him.

fewtarius, 2020-05-28:
Thanks @Goshin haven't had much time the last two days, but I'll get at it as soon as I can.

tiger511, 2020-05-28:
@fewtarius Try this:
[MultiGPIO_patch.zip](https://github.com/VoodooI2C/VoodooI2C/files/4697353/MultiGPIO_patch.zip)

I modded this version to work with GPIO Trackpad+ GPIO Touchscreen.Make sure both of them are GPIO pinned.Otherwise you might encounter unknown behaviour. Please report back with your CPU usage while trackpad/touchscreen in use.If you encounter any problem,execute the following command attach the output & your processor model: `log show --last=boot |grep Voodoo`

tiger511, 2020-05-28:
@marianopela  whats the output of this : `log show --last=boot |grep Voodoo`

Also share your Trackpad SSDT patch.Also share a screenshot of TPD0 from ioregistry

marianopela, 2020-05-28:
> @marianopela whats the output of this : `log show --last=boot |grep Voodoo`
> 
> Also share your Trackpad SSDT patch.Also share a screenshot of TPD0 from ioregistry

[SSDT-TPAD.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/4697670/AIR14-TPAD.aml.zip)
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/4697671/log.txt)
[image]

tiger511, 2020-05-28:
> [SSDT-TPAD.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/4697670/AIR14-TPAD.aml.zip)

In _CRS, shoudn't it be `Return (ConcatenateResTemplate (SBFB, SBFG))` ?

marianopela, 2020-05-28:
> In _CRS, shoudn't it be `Return (ConcatenateResTemplate (SBFB, SBFG))` ?

Sorry, I sent you the wrong one, because I was from my desktop. 
It's the same apart from `Return (ConcatenateResTemplate (SBFB, SBFG)` and 0x108 pin. I change those when I'm trying to debug GPIO. 
[SSDT-TPAD.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/4697770/SSDT-TPAD.aml.zip)

Btw I just woke it up from sleep and it worked for a full 3 seconds. I almost thought I was running in polling. Do you think there would be a way to debug where it hangs by adding logs to VoodooGPIO?

[34 more replies]

closed 2020-12-27 by fewtarius