VoodooI2C/VoodooI2C #326: Trackpad not working - Teclast F6 Pro  VooDooI2C  DSDT  TPDX  SYNA3602

Phoenil, 2020-05-25:
Hello everyone! 

The trackpad is being recognized by macOS thanks to batter patch, but it doesn't work in any manner.

The installation has been done through Clover with 10.14.6 running.

The battery and sleep have been fixed with no issue.

Bluetooth, keyboard, audio, everything works smooth and I don't see any problems.

Using VoodooI2C and VoodooI2CHID. Tried also Synaptics, nothing.

DSDT seems well patched by the opinion of many guys who helped me.

The last chance is to ask the developers.

Thanks for the help.

kprinssu, 2020-05-25:
Please I use the GitHub template.

marianopela, 2020-05-28:
@Phoenil Are you sure you carefully read Documentation? You seem to be missing pins and proper return in your CRS method
[image]

They surely are strange ACPIs, but you can give [Gpio Pinning](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning) a try

ben9923, 2020-05-29:
@Phoenil Please try removing the SYNA3602 personality from VoodooI2CHID's Info.plist.
Attach a new troubleshooting archive + IOReg.

Phoenil, 2020-05-29:
@marianopela I'll take a look today at it

@ben9923 Done, 
[Send me MacBook.zip](https://github.com/VoodooI2C/VoodooI2C/files/4700952/Send.me.MacBook.zip)
[MacBook ioreg 3.zip](https://github.com/VoodooI2C/VoodooI2C/files/4700954/MacBook.ioreg.3.zip)

kprinssu, 2020-06-04:
@Phoenil, as @marianopela pointed out, you _**need to**_ set up either interrupts or polling mode. VoodooI2C does not work out of the box, please read https://voodooi2c.github.io/#Polling%20Mode/Polling%20Mode.

Your DSDT shows that you a pin of 0x0, which is invalid. See, https://voodooi2c.github.io/#Polling%20Mode/Polling%20Mode and https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning

Phoenil, 2020-06-04:
This is my unpatched DSDT 
[image]
[DSDT.aml-original.zip](https://github.com/VoodooI2C/VoodooI2C/files/4731256/DSDT.aml-original.zip)

kprinssu, 2020-06-04:
@Phoenil and myself are carrying the discussion over on Gitter. Closing this for now.

closed 2020-06-04 by kprinssu

CrisHotpatch, 2023-06-07:
This is the solution:

https://www.tonymacx86.com/threads/voodooi2c-issues-syna3602.304029/post-2177566

fmgdias, 2024-01-20:
SYNA3602 Trackpad - Solution

Show contents pkg on VoodooI2CHID.kext

Open Info.plist with PlistEditor Pro

In IOKitPersonalities: Rename VoodooI2CHIDSYNA3602Device to VoodooI2CHIDSYNA3602&Col2Device

IOClass: Rename VoodooI2CHIDSYNA3602Device to VoodooI2CHIDSYNA3602&Col2Device

IOPropertyMatch: Rename SYNA3602 To SYNA3602&Col2

If "&Col2" not work, try "&Col1"