VoodooI2C/VoodooI2C #327: High CPU Usage on GPIO pinned Lenovo 720s.

theuktr4, 2020-05-27:
**Describe the bug**
CPU Usage jumps from 1-2% to about 18% when touching the Trackpad.
It's not limited to one Process. kernel_task, hidd  and especially Window_Server all have increased CPU Usage when using my Touchpad.
**Did you read the common errors documentation?**
Yes.
**Did you read the troubleshooting documentation?**
Yes, that's where I was looking to get my GPIO Pinning working

**Have you searched the issue on Github, Gitter, or Google?**
I did. Similar Errors mostly happened in older Versions.

**System Environment**1. What is your operating system? 
Mac OS 10.5.5 Catalina
2. What version of the drivers (including drivers) are you using?
Latest Release (2.4.2)
3. What is your hardware configuration?
Lenovo 720S-14IKB (Type 80XC)
i5-7200U

**Troubleshooting Archive**
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4690899/Archive.zip)

**Additional context**
There is no entry in my System Log that shows anything related to VoodooI2C

Goshin, 2020-05-27:
Might be fixed in master.

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4691110/Archive.zip) (built on 10.15.4)
Test this.

theuktr4, 2020-05-27:
Thanks for your quick response.
Overall CPU Usage seems to be a bit lower. (Especially Kernel_task)
However, WindowServer Usage still jumps to about 17% as well as hidd with 10%

Goshin, 2020-05-27:
`WindowServer` is related to canvas rendering and managing by the OS. So it can take relatively high CPU usage if you are using a CPU monitor tool like iStat that renders the CPU chart in real-time. It is not related to VoodooI2C.

`hidd` stands for Human Interface Device Daemon. It is the process in OS that handles the touchpad input. It is required but not under the control of VoodooI2C.

The numbers look normal to me. They are `kernel_task` 5% and `hidd` 3% on my 9750H.

theuktr4, 2020-05-27:
Yeah it seems a bit strange as to why these Processes are causing an increase of CPU Usage (btw. other than first observations Window_Server does go to well over 35% in Activity Monitor)

The reason why I was connecting this issue with my I2C Touchpad is that there is this clear bump in Overall CPU Utilization (see Screenshot), which consistently appears when moving my cursor (via Touchpad). 
It's not present when using a USB Mouse and my main Problem with it is, that it harms Battery Life considerably when only doing light tasks

[image]

Goshin, 2020-05-27:
Only the CPU usage of `kernel_task` is related to the driver, and it is much better and more acceptable than the prior versions. Compared to a USB mouse, I think it is more complicated for the OS to process the multitouch touchpad input. In my case, it makes much less difference to the battery than other tasks like browser, video, etc.

Currently, to my view, I don't see a significant issue in the driver to improve the CPU load, and it is acceptable to me. In contrast, I can easily see a 50% CPU usage of `kernel_task` when under heavy network load.

Maybe you can limit the report rate to further reduce the CPU load, but it will also introduce latency and make it less responsive.

kprinssu, 2020-05-27:
@theuktr4 The high CPU usage was known bug and should have been fixed in 2.4.2. As @Goshin pointed out it is not an issue with VoodooI2C rather with Apple's own MT2 stack.

Please also upload quantitative logs that it is VoodooI2C, i.e. a video of you doing gestures or a graph of the `kernel_task` using CPU resources over time. Other than that, we cannot help you.

tiger511, 2020-05-28:
@theuktr4  Do you have touchscreen ?
If no try this: 
[Archive-ResponseLatencyMicroSecond.zip](https://github.com/VoodooI2C/VoodooI2C/files/4697375/Archive-ResponseLatencyMicroSecond.zip)

kernel_task should be around 4% while trackpad in use.Also provide output of the following command: `log show --last=boot |grep Voodoo`

marianopela, 2020-05-28:
> If no try this:
> [Archive-ResponseLatencyMicroSecond.zip]

Hey, sorry to bother but those kexts seem to be somehow on the right path to make my trackpad work in GPIO mode. Can I ask you what changes they contain?

tiger511, 2020-05-28:
> Hey, sorry to bother but those kexts seem to be somehow on the right path to make my trackpad work in GPIO mode. Can I ask you what changes they contain?

They contain a lot of changes.What does this fix for you?

marianopela, 2020-05-28:
> > Hey, sorry to bother but those kexts seem to be somehow on the right path to make my trackpad work in GPIO mode. Can I ask you what changes they contain?
> 
> They contain a lot of changes.What does this fix for you?

Here https://github.com/VoodooI2C/VoodooI2C/issues/325#issuecomment-634585226 I tried to describe what happens

tiger511, 2020-05-28:
@marianopela  So your trackpad works with this kext or you still have problem? Do you have touchscreen? are they both GPIO ?

marianopela, 2020-05-28:
> @marianopela So your trackpad works with this kext or you still have problem? Do you have touchscreen? are they both GPIO ?

Nope, it does not work completely, but it works 2 seconds after every boot and 2 seconds after every wake (counted from the moment I touch my trackpad), so it definitely seems like it could work. I don't have a touchscreen, but I'm trying to get trackpad to work in GPIO (polling works)

theuktr4, 2020-05-28:
> @theuktr4 Do you have touchscreen ?
> If no try this:
> [Archive-ResponseLatencyMicroSecond.zip](https://github.com/VoodooI2C/VoodooI2C/files/4697375/Archive-ResponseLatencyMicroSecond.zip)
>
> kernel_task [...]

Thank´s subjectively it seemed to improve my issue.

The previous answers also lead me to believe, that this issue/behaviour might not be related to VoodooI2c. After all kernel_task usage is pretty low.

Still, here is some recording as well as the log you asked 
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/4697632/log.txt)

[Rec.mov.zip](https://github.com/VoodooI2C/VoodooI2C/files/4697656/Rec.mov.zip)

 :

closed 2020-06-02 by theuktr4