VoodooI2C/VoodooI2C #328: Buggy, delayed Trackpad input under OpenCore

fuzeless, 2020-05-29:
Hi guys!

First of all, thank you for your work to the Hackintosh community. It helps many of us to be able to experience the macOS awesome gesture system.

The bug I'm experiencing here is the Trackpad being buggy and delayed. It sometimes would jump for a few cm when touched, and all gestures have input lag that would render an uneasy experience ([see my video for more detail](https://youtu.be/FoWzv31Biks)). An user on OSXLatitude experience the same problem, but it behaved normally under Clover ([Here's the topic
](https://osxlatitude.com/forums/topic/13361-dell-latitude-7490-opencore-bootloader-buggy-trackpad-support-request/))
Also, the trackpad buttons are not working. It would be swell if you take a look at that!

Here's my bug report, as per the Troubleshooting doc:
[Trackpad bug report.zip](https://github.com/VoodooI2C/VoodooI2C/files/4699418/Trackpad.bug.report.zip)

**System Environment**
OS: macOS 10.15.5 + Windows 10 2004, OpenCore 0.5.8 (EFI included in the zip above)
Kexts: VoodooI2C 2.4.2 + VoodooI2CHID bundled with the release
Hardware: Dell Latitude 7490 (i5-8350u, 8Gb RAM, Alps I2C Trackpad: _SB.PCI0.I2C1.TPD1)

Any help would be appreciated! Thanks a lot and have a great day/night!

kprinssu, 2020-05-29:
Please use the Github issue template.

fuzeless, 2020-05-29:
Updated. Please take another look!

kprinssu, 2020-05-29:
Can you try the kexts attached here, https://github.com/VoodooI2C/VoodooI2C/issues/321#issuecomment-633972056?

fuzeless, 2020-05-29:
Tried the kext you sent, sadly the problem still persists.
I also tried Clover bootloader. The bug's gone in Clover but another problem arises: the trackpad would occasionally stopped responding while using (Eg: dragging the cursor).

kprinssu, 2020-06-04:
@fuzeless on closer inspection, it looks to be an issue with how the interrupts are being queued. It's possibly an OpenCore environment issue that is causing the problem. For now, can you try an older version of VoodooI2C (2.3 should do)?Z

If 2.3 works, then there are changed from 2.4 onwards that are causing your problems.

fuzeless, 2020-06-05:
Hi. I was able to get VoodooI2C 2.4.2 to work in interrupt mode on OC by fixing my _STA and _CRS hotpatch (thanks to muttonhead411 on OSXLatitude!). But another problem arise, like Clover: Trackpad would occasionally cutoff while dragging, or using gestures (mostly 2-finger gestures), but I'll save that for another issue. Gonna close this issue for now. Thanks for your help!

closed 2020-06-05 by fuzeless

romancin, 2020-10-20:
> Hi. I was able to get VoodooI2C 2.4.2 to work in interrupt mode on OC by fixing my _STA and _CRS hotpatch (thanks to muttonhead411 on OSXLatitude!). But another problem arise, like Clover: Trackpad [...]

I have the same problem in the exact laptop. Could you please publish your EFI files? Thank you very much in advance! 🙂