VoodooI2C/VoodooI2C #329: Lenovo S145 - 15WIL - Problems with trackpad

prsolucoes, 2020-05-31:
**Describe the bug**
Thanks for the great work.

1 - When i make login the trackpad is very slow for 10 seconds and after it move normal;
2 - When i drag something and leave the trackpad the object still dragged and i need click to it leave the object dragged (sometimes it happen and sometimes no);
3 - The third problem is when i click (pressing or touching) it click sometimes OK and on other it move the cursor for random place;

**Did you read the common errors documentation?**
YES

**Did you read the troubleshooting documentation?**
YES

**Have you searched the issue on Github, Gitter, or Google?**
YES

**System Environment**
Please answer:
1. What is your operating system? macos 10.15.5
2. What version of the drivers (including drivers) are you using? (latest - zip with all: https://www.dropbox.com/s/3vpebbc63u12rjf/EFI-FINAL.zip?dl=1)
3. What is your hardware configuration?  
1,99 GHz Quad-Core Intel Core i7
20gb ram 

**Troubleshooting Archive**
attahced
[MacBook Pro.zip](https://github.com/VoodooI2C/VoodooI2C/files/4707048/MacBook.Pro.zip)

**Additional context**
n/a

kprinssu, 2020-06-04:
Can you please use an older version of VoodooI2C (<= 2.3) and report back.

prsolucoes, 2020-06-04:
Sure, i will use version 2.2 and report here.

kprinssu, 2020-06-04:
Please also try version 2.3.

prsolucoes, 2020-06-04:
ok, i will use version 2.3 so

prsolucoes, 2020-06-04:
Hi,

I have tested here and versions < 2.3 don't work in any place (boot, login and on desktop) - it stop totally. I need put external mouse to make cursor works again.

Only version 2.4 make it work again.

Thanks.

djale, 2020-06-04:
Use VoodooI2C + VoodooI2CSynaptics  and make sure to edit synaptics kext info.plist

VoodooI2CSynaptics>Show Package Contents>Contents>info.plist
Edit IOPropertyMatch>name to "SYNA2B3C"

[image]

prsolucoes, 2020-06-05:
> SYNA2B3C

Hi. It solve mt drag/drop problem. Now it is working nice. Only the TAP gesture that still failing...sometimes i need tap 4/5 times to the TAP/CLICK work.

And my trackpad is not showing more. Now it is searching for one.

[image]

djale, 2020-06-05:
You didn't patch your DSDT, later today I will send you a tutorial for everything related to Lenovo laptop build.

kprinssu, 2020-06-05:
Thanks @djale. @prsolucoes this is no longer a VoodooI2C issue. Please get battery status working for preferences to properly work.

closed 2020-06-05 by kprinssu

prsolucoes, 2020-06-05:
@djale ok, please send to me :)
thanks more one time

djale, 2020-06-05:
@prsolucoes https://github.com/djale/Lenovo-Hackintosh