VoodooI2C/VoodooI2C #332: Sync with changes in VoodooInput

usr-sse2, 2020-06-05:
Fix for acidanthera/bugtracker#945 moved finger type assignment to PS2/I2C side. The original issue is that when holding a moved item with one finger and switching desktops with other three or four fingers, when lifting three fingers the holding finger becomes index finger, and it prevents the deskop switching gesture from ending. So if a finger was detected to be a thumb, it should remain thumb until it is lifted, and fingers shouldn't change their types without reason.
VoodooInput changes are in acidanthera/VoodooInput@835731a0e1547a62b645f4b49b4d3347d3f96dbe, and you can use acidanthera/VoodooPS2@578da615795c50de1d080cc88a194bfcff53de1b for a reference implementation. New VoodooInput remains binary compatible as it reuses existing `id` field, but it can produce wrong behavior if finger types are set incorrectly.

kprinssu, 2020-06-05:
I'll have a fix in place shortly.

kprinssu, 2020-06-05:
Figured out my problem and it was fixed, PR to address this issue: https://github.com/VoodooI2C/VoodooI2C/pull/333/files

Goshin, 2020-06-05:
> The original issue is that when holding a moved item with one finger and switching desktops with other three or four fingers, when lifting three fingers the holding finger becomes index finger, and [...]

@usr-sse2 A better and simpler solution for the original issue is to set the thumb finger while dragging. https://github.com/Goshin/VoodooInput/commit/c61ce39b3db4cd9c32abc9994a36e41c67135d8b