VoodooI2C/VoodooI2C #334: the 10th generation cpu i7-10750h doesn't  work

wcedla, 2020-06-06:
my laptop is Lenovo y7000 2020 ,it's cpu is i7-10750h,but there is not match id for it, please add the 10th generation id, thank you very much! the id fot it may be pic8086,6e8 and pci8086,6e9

marianopela, 2020-06-06:
`pic8086,6e8` and `pci8086,6e9` are already included, as clearly stated in the README of this repo

wcedla, 2020-06-07:
sorry，i check the readme，but there is not pic8086,6e8 and pic8086,6e9，is it in master branch？thanks very much

wcedla, 2020-06-07:
i have manually added the id to the info.plist,but it also doesn't work
[lenovo-repport.zip](https://github.com/VoodooI2C/VoodooI2C/files/4741402/lenovo-repport.zip)here is my laptop's relate file,please help us,thank you very much

marianopela, 2020-06-07:
> sorry，i check the readme，but there is not pic8086,6e8 and pic8086,6e9，is it in master branch？thanks very much

My bad, I meant `pci8086,2e8` and `pci8086,2e9`, which are Comet Lake controllers

Remove SSDT-I2CxConf.aml, add SSDT-XOSI.aml and send another Ioreg, your full EFI with your ACPI origin

Then, read here https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning

wcedla, 2020-06-07:
> My bad, I meant `pci8086,2e8` and `pci8086,2e9`, which are Comet Lake controllers
>
> Remove SSDT-I2CxConf.aml, add SSDT-XOSI.aml and send another Ioreg, your full EFI with your ACPI origin
>
> Then, read [...]

Thank you very much,after added the file ,the touchPad can work 
[lenovo-report-2.zip](https://github.com/VoodooI2C/VoodooI2C/files/4741589/lenovo-report-2.zip)

kprinssu, 2020-06-07:
@wcedla  Next time, please use the Github Issue template. Closing as this is fixed by PR#335. Thanks @marianopela.

closed 2020-06-07 by kprinssu