VoodooI2C/VoodooI2C #338: When a touchscreen is rotated, only single-finger touches are rotated but gestures are not

usr-sse2, 2020-06-10:
I use VoodooI2CHID on a Dell Latitude 7390 2-in-1. I'm currently not using UPDD. When I rotate the touchscreen, single-finger touches are rotated so cursor movement is OK, but I have to swipe two fingers left or right to scroll vertically, swipe four fingers up or down to change desktops.

usr-sse2, 2020-06-11:
>Duplicate of https://github.com/VoodooI2C/VoodooI2C/issues/199.

@kprinssu Not a duplicate. 199 is due to multiple displays, and this is for a single display.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Goshin, 2020-06-11:
@usr-sse2 in VoodooInput, https://github.com/acidanthera/VoodooInput/blob/3668651b43246a4d275dfa0da51d83f02cde73ec/VoodooInput/VoodooInput.cpp#L116-L118

```cpp
UInt8 VoodooInput::getTransformKey() {
    return transformKey;
}
```

The var `transformKey` should be updated from its property after the screen is rotated.

kprinssu, 2020-06-12:
My bad @usr-sse2. Reopening.

idoc69, 2020-06-20:
Works correctly with release 2.3 which included blankmac's fix #172 I think.

kprinssu, 2020-06-20:
@idoc69 is there a regression with 2.4+?

idoc69, 2020-06-20:
With 2.3 two finger gestures are rotated correctly and behave as expected in portrait orientation on my spectre x360 touchscreen.
Just checked various versions of 2.4; gestures are not rotated and are at 90º to expected behaviour (eg attempting to scroll a window up and down actually scrolls the window left and right)

Goshin, 2020-06-20:
@kprinssu migration to VoodooInput in v2.4 caused the problem. The `transformKey` in VoodooInput needs to update after screen is rotated as I said in https://github.com/VoodooI2C/VoodooI2C/issues/338#issuecomment-642481322, maybe via `VoodooInput::message(...)`.

idoc69, 2020-06-21:
No I don't think so.
Looking at the code it's still aware of screen rotation (certainly for single touch), but there is a swap for handling multitouch from the VoodooI2CCSGestureEngine which handles the kIOFBTransformKey) to the VoodooI2CNativeEngine which doesn't.

Goshin, 2020-06-21:
@idoc69 I don't fully understand but VoodooI2CNativeEngine does handle `kIOFBTransformKey`.

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/4809381/Archive.zip)
Can you test this one? Notify VoodooInput once properties updated.

Related changes: 
https://github.com/Goshin/VoodooI2C/commit/c7c946ebb4d876ec82542e387ba5d6e88a8ba0c4
https://github.com/Goshin/VoodooInput/commit/9012133a8cdefcd3baa20020f94b3afa86d46d56

idoc69, 2020-06-21:
Thanks, unfortunately doesn't work.
I only said the above because a search in Xcode found that only VoodooI2CCSGestureEngine seemed to check kIOFBTransformKey.
A fix seems to be to transform the multitouch co-ordinates in the VoodooI2CTouchscreenHIDEventDriver
I added this after line 231:

                for (int i=0; i < event.transducers->getCount(); i++) {
                    VoodooI2CDigitiserTransducer* transducer = OSDynamicCast(VoodooI2CDigitiserTransducer, event.transducers->getObject(i));
                    if (!transducer)
                        continue;
                    if (transducer->is_valid) {
                        if (transducer->tip_switch) {
                            IOFixed x = transducer->coordinates.x.value();
                            IOFixed y = transducer->coordinates.y.value();
                          	checkRotation(&x, &y);
                            transducer->coordinates.x.update(x, timestamp);
                            transducer->coordinates.y.update(y, timestamp);
                    		}
                		}
                }

It seems to work on my device?

Goshin, 2020-06-21:
@idoc69 CSGestureEngine was no longer used a long time ago. MT2Simulator handles the coordinate transform, for the test kexts in https://github.com/VoodooI2C/VoodooI2C/issues/338#issuecomment-647118256, if you use OpenCore, you need to add VoodooInput in your config; if you use VoodooPS2 at the same time, you need to remove the VoodooInput kext in its plugin directory.

idoc69, 2020-06-22:
Thanks. I tried OpenCore and it works well but I can only boot from a USB stick,  it seems all touchscreen HP laptops crash when trying to boot from the internal drive. Clover has a trick to chain load via linux to get around this. Is there anything I can do in Clover to get the test kexts to work?

Goshin, 2020-06-22:
@idoc69 if you are using Clover, you need to remove all the `VoodooInput.kext` duplicates in `VoodooPS2Controller.kext/Contents/PlugIns` and other places, only keep the one buddled with `VoodooI2C.kext`.

idoc69, 2020-06-22:
Thanks Goshin. I can confirm this does fix the issue - much appreciated.

kprinssu, 2020-07-05:
@Goshin can you create a PR containing the fix please?

usr-sse2, 2020-07-19:
Thanks, now it works!

closed 2020-07-19 by usr-sse2