VoodooI2C/VoodooI2C #340: In polling mode lifting finger is not always detected

usr-sse2, 2020-06-13:
**Describe the bug**
When I was using polling mode, VoodooI2C often reported finger as still present on the touchpad when it was actually lifted. It doesn't happen in GPIO mode.

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes

**System Environment**
1. macOS 10.15.5
2. a813906da22e71cbe930264c2c2462d36e9bf265
3. Dell Latitude 7390 2-in-1, DELL0823, VoodooI2CHID

kprinssu, 2020-06-14:
A simple fix is to have a timeout of sorts (send a no-touch event after finger is lifted), however I can see this becoming ugly rather fast. This would require us to keep track of the state of fingers, which the hardware should be doing.

I don't think there's a good of way dealing with this.

kprinssu, 2020-07-09:
My best guess without looking at code is that the polling function misses the finger lift or the hardware is not properly reporting the finger lift events. 

I am not as familiar with the polling code in VoodooI2CHID. I'll take a look sometime next week.