VoodooI2C/VoodooI2C #342: Ice Lake Intel I2C

MVDB0110, 2020-06-16:
Do you have any idea, when Ice Lake will be supported in your kext? I see in the commits that a Ice Lake id is inserted in one of the plists, but there is no support in the kext for the controller.

ben9923, 2020-06-26:
Which PCI ID does your device have?
If VoodooI2C doesn't have it, can you try manually adding it & installing the kext(s)?

schwxr, 2020-06-30:
This may not be the case for you, but in my ICL build, VoodooI2C wouldn't load unless I applied the CML fix from #245 to my I2C controller. From there, I was able to proceed with my I2C device as usual.

kprinssu, 2020-08-04:
@schwxr That fix has been added to upstream since 2.4 (March 2020), see https://github.com/VoodooI2C/VoodooI2C/commit/6d37695082eab4fd9d0203d3a46c7968f9c845d. Is there an underlying issue that has not been fixed?

@MVDB0110 ICL support should've been added with https://github.com/VoodooI2C/VoodooI2C/commit/c3bf644abb510bcc96d553c625ce469e4a2d95aa. As @ben9923 pointed out, do you perhaps have a controller that is not within that list?

KhoraLee, 2020-08-10:
@kprinssu 6d37695082eab4fd95d0203d3a46c7968f9c845d This only fix CML laptops. ICL(Ice lake) Laptops also need same patch to use I2C Trackpad.

[image]
And pci8086,34e9 is missing.

After fixing two issue 
[image]
Trackpad works fine

kprinssu, 2020-08-10:
Thanks @KhoraLee. Merged the changes into master.

KhoraLee, 2020-08-10:
Now we need to figure out how to check ICL CPUs in order to apply this patch (6d37695082eab4fd95d0203d3a46c7968f9c845d) not only CML but also ICL.

KhoraLee, 2020-08-10:
@kprinssu Please check this PR. I used same method to verify CPU. 

+It is working for me

kprinssu, 2020-08-10:
@KhoraLee You want something like this for that if condition:
```
const char kIceLakeflag[2] = {'3', '4'};

if ( (tmp[0] == kCometLakeflag[0] && tmp[1] == kCometLakeflag[1] && tm[2] == kCometLakeflag[2]) || (tmp[0] == kIceLakeflag[0] && tmp[1] == kIceLakeflag[1])) {

.... 

```

KhoraLee, 2020-08-10:
@kprinssu please check the pr #363.

MVDB0110, 2020-08-10:
@KhoraLee Nice work! Sad that the fix will only work when a new version of VoodooI2C is released.

kprinssu, 2020-10-03:
2.5 contains PR#363.

closed 2020-10-03 by kprinssu