VoodooI2C/VoodooI2C #343: attempting to edit DSDT for thinkpad X1E

antgeth, 2020-07-04:
to start off, my apologies if i'm missing something very basic here.  i'm pretty new to this whole thing and it's fairly overwhelming.

i'm currently trying to edit my DSDT in order to use VoodooI2C as i have found VoodooPS2Controller to be very buggy.

currently at the GPIO pinning stage, following these instructions: https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning

*You will also need to make sure that your I2C Serial Bus Name is correctly labelled. Search for your device ACPI ID in your DSDT until you reach its device entry, you should be able to find a Name called SBFB. If you cannot find it, you may find something that looks like this instead:*

```    Method (_CRS, 0, Serialized)  // _CRS: Current Resource Settings
                {
                    Name (SBFI, ResourceTemplate ()
                    {
                        I2cSerialBusV2 (0x0015, ControllerInitiated, 0x00061A80,
                            AddressingMode7Bit, "\\_SB.PCI0.I2C1",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                        Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                        {
                            0x0000006D,
                        }
                    })
                    Return (SBFI)
                }
```
*In this case, rename SBFI to SBFB and remove the following from it:*
```
       Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
    {
        0x0000006D,
    }
```

**however**, my DSDT looks like this:
```  
    Device (TPD0)
        {
            Name (HID2, Zero)
            Name (SBFB, ResourceTemplate ()
            {
                I2cSerialBusV2 (0x0000, ControllerInitiated, 0x00061A80,
                    AddressingMode7Bit, "NULL",
                    0x00, ResourceConsumer, _Y38, Exclusive,
                    )
            })
            Name (SBFG, ResourceTemplate ()
            {
                GpioInt (Level, ActiveLow, ExclusiveAndWake, PullDefault, 0x0000,
                    "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                    )
                    {   // Pin list
                        0x0000
                    }
            })
            Name (SBFI, ResourceTemplate ()
            {
                Interrupt (ResourceConsumer, Level, ActiveLow, ExclusiveAndWake, ,, _Y39)
                {
                    0x00000000,
                }
            })

```

so perhaps you can understand my confusion.  in addition to this `TPD0` section there is also a `TPL1` section below it, with virtually identical content.  i am not certain which of these is my touchpad.  i do not have a windows installation so i cannot use device manager to find the ACPI ID.  i found these by searching for SBFI in my DSDT.  any help would be most welcome.

marianopela, 2020-07-05:
Do you have any other TPDX/TPAD devices in your ACPI? 
If not, TPD0 is definitely your trackpad. TPL1 would usually be a touchscreen, which if you don't have, don't worry about it. 

[IORegistryExplorer](https://github.com/vulgo/IORegistryExplorer/releases/tag/v2.1) would also be useful in discovering your device and future debugging, because VoodooI2C should automatically attach to your device, and be shown there.

kprinssu, 2020-07-05:
Please use Gitter for support with DSDT patching, Github is meant for issues with the code.

closed 2020-07-05 by kprinssu