VoodooI2C/VoodooI2C #347: Can multiple satellites work at the same time?

im-bravo, 2020-07-20:
My computer has HIC touch screen and ELAN Clickpad.
I've successfully driven the touch screen by I2C.

I would like to know if the design supports multi satellites?

kprinssu, 2020-07-20:
Multiple satellite kexts loading is not supported for the moment.

coolstar, 2020-07-21:
@kprinssu multiple satellite kexts works fine. I have ELAN trackpad and Atmel touch screen on my netbook and it works fine

kprinssu, 2020-07-22:
@coolstar You are right. @im-bravo you should be fine long as the VoodooI2CHID does not attach to the device.

kprinssu, 2020-08-06:
@im-bravo I am closing this issue, please re-open if you cannot load multiple satellite kexts at the same time.

closed 2020-08-06 by kprinssu

im-bravo, 2020-08-16:
@coolstar @kprinssu Thanks for your information !!!