VoodooI2C/VoodooI2C #350: Trackpad cant detect more than 2 fingers

xrokz, 2020-07-22:
**Describe the bug**
The trackpad only read 2 fingers

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes and no one helps

**System Environment**
Please answer:
1. What is your operating system? Catalina
2. What version of the drivers (including drivers) are you using? 
3. What is your hardware configuration?
```md
|      [Laptop][Lenovo Legion Y540]
|      [Trackpad][I2C-HID]
|      [OS][Windows 10, Hackintosh Catalina]     
|      [CPU][2.6 GHz 6-Core Intel Core i7-9750H]
|      [iGPU][Intel UHD Graphics 630 1536 MB]
|      [dGPU][Nvidia GTX GeForce 1650]
|      [Memory][16 GB 2667 MHz DDR4]
|      [Storage][256 SSD, 2TB HDD]
```
**Troubleshooting Archive**
Here i was touching with 4 fingers and still its only detect 2
https://cdn.discordapp.com/attachments/677900944855400481/735529462623371334/unknown.png
im using app called [FingerMgmt](https://github.com/jnordberg/FingerMgmt)

**Additional context**
i need to use 4 fingers to use the trackpad gestures

kprinssu, 2020-07-22:
As previously mentioned [here](https://github.com/VoodooI2C/VoodooI2C/issues/349) and [here](https://github.com/VoodooI2C/VoodooI2C/issues/348), please attach a troubleshooting archive. Please **read** (make sure to __read__) https://voodooi2c.github.io/#Troubleshooting/Troubleshooting. 

Without the archive, I have no idea how your system is configured and support will not be provided.

xrokz, 2020-07-22:
this is my IOReg
[xRokz’s MacBook Pro.zip](https://github.com/VoodooI2C/VoodooI2C/files/4962228/xRokz.s.MacBook.Pro.zip)

xrokz, 2020-07-22:
@kprinssu no news ? i added the archive as you asked

kprinssu, 2020-07-23:
You only uploaded the IO Registry. All I can tell from this is that, VoodooI2C is loading. You need to upload more logs and your EFI folder including: DSDT, SSDT, kexts.

Additionally, I suspect that this is not a bug in the code and Github is not the right place for configuration questions. Please use Gitter for general support.

closed 2020-07-23 by kprinssu

xrokz, 2020-07-23:
in gitter no one answer

xrokz, 2020-08-02:
so can i reopen this ?

kprinssu, 2020-08-02:
Please use Gitter for configuration support and only re-open this if it's an actual issue with the code (which I doubt it is from your description of the issue).