VoodooI2C/VoodooI2C #351: AUI165B work in GPIO model but only support two fingers gesture

ettingshausen, 2020-07-22:
**Describe the bug**
Only two fingers gesture supported and there's no touchpad setting in system setting.

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors
yes

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.
yes

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.
yes

**System Environment**
Please answer:
1. What is your operating system? 
Catalina 10.15.6 (19G73)
2. What version of the drivers (including drivers) are you using?
VoodooI2C + VoodooI2CHID (2.4.3)
3. What is your hardware configuration?  

| Name     |           Model            | Note |
| :------- | :------------------------: | :--- |
| CPU      |    Intel Core i5 8250U     |      |
| GPU      |   Intel UHD Graphics 620   |      |
| RAM      |          8G + 4G           |      |
| SSD      | Samsung 970 EVO Plus 500G  |      |
| HDD      | Seagate ST1000LM035-1RK172 |      |
| WLAN     |          DW1820a           |      |
| TouchPad |          AUI165B           |      |

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

[K43c-80-origin.zip](https://github.com/VoodooI2C/VoodooI2C/files/4961031/K43c-80-origin.zip)
[k43c-80-IOReg.zip](https://github.com/VoodooI2C/VoodooI2C/files/4961142/k43c-80-IOReg.zip)
[My patch SSDT-TPAD.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/4961105/SSDT-TPAD.aml.zip)
[sys.log.zip](https://github.com/VoodooI2C/VoodooI2C/files/4961246/sys.log.zip)

**Additional context**    

Two fingers gesture work fine and smooth, but no settings displayed.

Goshin, 2020-07-22:
Seems you didn't load VoodooInput. Try adding it into your bootloader config.

ettingshausen, 2020-07-22:
> Seems you didn't load VoodooInput. Try adding it into your bootloader config.
Sorry I can't get it, VoodooInput was contained in VoodooI2C,  extra boot args needed?
@Goshin

Goshin, 2020-07-22:
What bootloader are you using?

ettingshausen, 2020-07-22:
Clover

ettingshausen, 2020-07-22:
After tried this patch   [SSDT-RMCF-Trackpad-enable.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/4961408/SSDT-RMCF-Trackpad-enable.aml.zip)  
```
/*
 * Intel ACPI Component Architecture
 * AML/ASL+ Disassembler version 20180427 (64-bit version)(RM)
 * Copyright (c) 2000 - 2018 Intel Corporation
 * 
 * Disassembling to non-symbolic legacy ASL operators
 *
 * Disassembly of iASL8ttynS.aml, Thu Jul 23 01:21:16 2020
 *
 * Original Table Header:
 *     Signature        "SSDT"
 *     Length           0x00000078 (120)
 *     Revision         0x02
 *     Checksum         0xF9
 *     OEM ID           "hack"
 *     OEM Table ID     "RM-Track"
 *     OEM Revision     0x00000000 (0)
 *     Compiler ID      "INTL"
 *     Compiler Version 0x20180427 (538444839)
 */
DefinitionBlock ("", "SSDT", 2, "hack", "RM-Track", 0x00000000)
{
    External (_SB_.PCI0.LPCB.PS2K, DeviceObj)    // (from opcode)

    Scope (_SB.PCI0.LPCB.PS2K)
    {
        Name (RMCF, Package (0x02)
        {
            "Mouse", 
            Package (0x02)
            {
                "ActLikeTrackpad", 
                ">y"
            }
        })
    }
}
```
TouchPad Setting shown up but still only two fingers gesture .  
[image]

Goshin, 2020-07-22:
Forget about it. Clover should load VoodooInput automatically.

ettingshausen, 2020-07-22:
[k43c-80-after-SSDT-Patch.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/4961475/k43c-80-after-SSDT-Patch.ioreg.zip)

Goshin, 2020-07-22:
> After tried this patch SSDT-RMCF-Trackpad-enable.aml.zip
TouchPad Setting shown up but still only two fingers gesture .

This means it is working in PS2 mode?

ettingshausen, 2020-07-22:
>This means it is working in PS2 mode?  

It’s not possible, but it’s also a bit strange, I'm sure this is an I2C device.

ettingshausen, 2020-07-27:
Is it the problem of ALPS? This is an ALPS touchpad.
I found @alexandred 's comment on [gitter](https://gitter.im/alexandred/VoodooI2C?at=5a4eb7b829ec6ac311bd31e2)
> actually there has never been alps support in voodooi2c
i do think it should be supported by a PS2 driver
its usual of dell to have dual PS/2 / I2C devices, its what i have and VoodooPS2 and VoodooI2C both work for me  

So ALPS should use VoodooPS2 instead？

kprinssu, 2020-07-27:
@ettingshausen I have not heard of I2C based Alps touchpads (there are I2C based touchscreens by Alps). Best way to confirm is to check which driver loads in Windows or which kernel module in Linux.

ettingshausen, 2020-07-27:
@kprinssu  There's no touchscreens devices on this laptop.  I just opened it and saw the touchpad chip, it is ALPS.  

[image]
 

Is it possible to fully work？Or I have to replace it with a Synaptics one.

Goshin, 2020-07-27:
@ettingshausen try removing `VoodooPS2Mouse.kext` and `VoodooPS2Trackpad.kext` in `VoodooPS2Controller.kext/Contents/PlugIns`.

ettingshausen, 2020-07-27:
I tried @coolstar 's [OS-X-ALPS-DRIVER](https://github.com/coolstar/OS-X-ALPS-DRIVER) still can't get it work.

ettingshausen, 2020-07-27:
> @ettingshausen try removing `VoodooPS2Mouse.kext` and `VoodooPS2Trackpad.kext` in `VoodooPS2Controller.kext/Contents/PlugIns`.

I tried, but nothing has changed.

Goshin, 2020-07-27:
mind providing the new IOReg file?

ettingshausen, 2020-07-27:
[k43c-80.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/4983827/k43c-80.ioreg.zip)

Goshin, 2020-07-27:
[image]

I can see `IOHIDEventDriver` attached, instead of `VoodooI2CPrecisionTouchpadHIDEventDriver`.

ettingshausen, 2020-07-27:
@Goshin Do you mean it recognized as a touchscreen device?

blankmac, 2020-07-27:
If you look at the device usage pairs, it does not support the windows precision touchpad protocol but instead uses a vendor defined protocol for precision touch and is not supported by VoodooI2C for anything but simple mouse input.

Goshin, 2020-07-27:
As @blankmac mentioned, the device doesn't support `0xD, 0x5` usage.

[image]

ettingshausen, 2020-07-29:
Finally, I bought a ELAN online and replaced the ALPS one. 
It works like a charm. 

[image]

Thank you guys! Love you 3000!

closed 2020-07-29 by ettingshausen