VoodooI2C/VoodooI2C #353: Dont load in big sur

corint1, 2020-08-03:
**Describe the bug**
only 2.2 version load in big sur...last version 2.4.3 dont load ...use last voodoops2 kext from acidanthera
**Did you read the common errors documentation?**
yes
**Did you read the troubleshooting documentation?**
yes
**Have you searched the issue on Github, Gitter, or Google?**
no
**System Environment**
Please answer:
1. Big Sur beta 3
2. any 
3.  lenovo 530s

kprinssu, 2020-08-03:
The drivers have been confirmed to work with macOS 11. The command you are using doesn't seem correct to me; kextload is complaining it cannot find the VoodooGPIO dependency. 

Can you try with `sudo kextload <satellite kext> -d VoodooI2C.kext`? Can you also upload the kext you are using (as requested within the Github issue template)?

corint1, 2020-08-03:
MacBook-Pro:~ bogdanmicu$ kextstat | grep -v com.apple
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   45    6 0                  0x2a000    0x2a000    as.vit9696.Lilu (1.4.6) ADB78536-EC4A-3B25-B5FC-B986C5B11FA6 <8 6 5 3 2 1>
   46    0 0                  0xa2000    0xa2000    as.vit9696.WhateverGreen (1.4.1) A603BEFA-3E48-348A-B5DD-40D8797BE774 <45 14 8 6 5 3 2 1>
   47    0 0                  0x160000   0x160000   as.vit9696.AppleALC (1.5.1) B77C9589-B6E1-3D50-BEF6-8624C99B0DDF <45 14 8 6 5 3 2 1>
   48    1 0                  0x2e000    0x2e000    as.vit9696.VirtualSMC (1.1.5) 135EEB52-6B41-352F-A555-A9EBFA779A26 <45 13 8 6 5 3 2 1>
   50    0 0                  0x1d000    0x1d000    ru.usrsse2.SMCBatteryManager (1) 71576E26-E441-38B2-B7F7-C14C1C028E89 <49 48 45 13 8 6 5 3 2 1>
   51    0 0                  0xc000     0xc000     xyz.racermaster.NoTouchID (1.0.3) 65553FA7-4620-3C7E-8D0D-B7E0C74A5A83 <45 8 6 5 3 2 1>
   56    0 0                  0x18000    0x18000    as.lvs1974.AirportBrcmFixup (2.0.8) D646CED0-012D-3F67-B319-BF9FB3153626 <55 52 45 43 14 8 6 5 3 2 1>
   57    0 0                  0xb000     0xb000     com.alexandred.VoodooI2CServices (1) DBFA5EBE-2783-30ED-8E39-5829C42D6832 <5 3>
   58    1 0                  0x2a4000   0x2a4000   as.acidanthera.BrcmFirmwareStore (2.5.3) 5E3D354B-F342-3FE2-B34C-0BA861FF7E63 <6 5 3>
   65    0 0                  0x9000     0x9000     org.rehabman.driver.VoodooTSCSync (1.5.0) 440D31DE-A40E-3E2B-9F50-B0DCF1FCFF05 <8 5 3>
   72    3 0                  0x14000    0x14000    as.acidanthera.voodoo.driver.PS2Controller (2.1.6) B2B40368-4D21-3702-97A9-9E1F01112B3C <13 8 6 5 3 1>
   73    0 0                  0x21000    0x21000    org.coolstar.VoodooGPIO (1.1) 652E2512-9BF1-35AD-85ED-880C9C5ABE9F <13 6 5 3>
   90    0 0                  0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.1.6) CE5D8919-D86C-376A-A038-78431E09D5F4 <72 68 8 6 5 3 1>
   91    0 0                  0x15000    0x15000    as.acidanthera.voodoo.driver.PS2Trackpad (2.1.6) 3A34F269-1D51-3BA6-AA64-79FBAB0021C6 <72 68 6 5 3>
   92    0 0                  0xb000     0xb000     as.acidanthera.voodoo.driver.PS2Mouse (2.1.6) 0E13614E-EA5A-3B79-AE40-25D6C185269E <72 68 6 5 3>
  104    0 0                  0xd000     0xd000     as.acidanthera.BrcmPatchRAM3 (2.5.3) 8250806A-3B84-3593-93AB-48E1703639A2 <58 22 6 5 3 1>
  140    0 0xffffff7f9f4b3000 0x2000     0x2000     com.intel.driver.EnergyDriver (3.7.0) 35E739F9-BF6C-3024-A67C-750711B3FB64 <8 6 5 3>
  166    0 0xffffff7f9f5e5000 0x40000    0x40000    com.paragon-software.filesystems.ntfs (106.5.15) 4575C073-7080-3B2F-AF62-A365668ACD34 <8 6 5 1>
MacBook-Pro:~ bogdanmicu$

corint1, 2020-08-03:
I dont know what happend but over 1000 reset, reset nvram , reinstall big sur , reinstall all kext , abd many many many other things and 4 days loss in final is working ...not all gesture , not all functions but working ...it is ok now ? 

see pictures

[image]
[image]

closed 2020-08-03 by corint1