VoodooI2C/VoodooI2C #354: Can't load both VoodooI2CELAN and VoodooI2CHID on Catalina 10.15.6

adamantinum, 2020-08-05:
Hi,
On Catalina 10.15.6 I can't load both satellites HID and ELAN, needed, as far as I know, to enable gestures on my ELAN1000 touchpad. If I inject both kext, just ELAN is loaded and touchpad doesn't works. If HID is the only satellite, it loads fairly and touchpad works without gestures, but there is not showed in System Preferences.
I have also tried to boot Big Sur installer and both kext are loaded correctly, but AppleIntelLpssI2C is not blocked.
I have read documentation, troubleshooting and asked on Gitter but without results, the issue happens not only with the last VoodooI2C release, but also with the 2019 last one, that used to work on Catalina 10.15.3.
[archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5027880/archive.zip)

kprinssu, 2020-08-05:
Please use Gitter for help with configuration.

closed 2020-08-05 by kprinssu

adamantinum, 2020-08-05:
It isn't a help request, since the configuration is correct and it worked in the past, but a report of a common – as far as I have read on forums – bug on ELAN100 touchpad, especially in ASUS laptop.

kprinssu, 2020-08-05:
ELAN takes precedence over HID. You should only need one or the other. To me, this is not a code issue. Please ask for help on Gitter.