VoodooI2C/VoodooI2C #355: Version cannot be updated

puzhengwu, 2020-08-06:
Why is 2.1.4 still displayed when the version is updated to 2.4.3? Is it because the file name of the old version is not overwritten, other drivers can normally upgrade to the latest version

kprinssu, 2020-08-06:
I am assuming you are referring to the satellite kexts? If so, yes this a known problem as their versions automatically are not updated when VoodooI2C is released.

puzhengwu, 2020-08-07:
[image]
hackintools

kprinssu, 2020-08-07:
That is not an issue with VoodooI2C. That shows that you are using 2.1.4, Hackintool is warning there is a newer version out.

Closing as this is an user error.

closed 2020-08-07 by kprinssu