VoodooI2C/VoodooI2C #357: Random Kernel Panics due to VoodooI2C On Big Sur

cheesycod, 2020-08-08:
**Describe the bug**
A clear and concise description of what the bug is.
On macOS 11, my pc randomly reboots due to a kernel panic caused by VoodooI2C. I am unable to get a log as it happens randomly.
**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors
yes
**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.
yes (I cannot show a log of the panic right now, since it happens randomly)
**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.
yes
**System Environment**
Please answer:
1. What is your operating system? macOS 11 developer beta 4
2. What version of the drivers (including drivers) are you using? 2.4.4 (I made a small patch to the plist for VoodooI2CHID to prevent it from hooking onto my trackpad)
3. What is your hardware configuration? Dell Inspiron 7568, Intel Core i5-6200U. I am only using VoodooI2C for just my touchscreen and not my PS2 through a small plist patch due to broken I2C0

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

**Additional context**
Add any other context about the problem here.

cheesycod, 2020-08-08:
I do not use a custom DSDT as my PC apparently does it properly and this issue did not happen on Catalina

cheesycod, 2020-08-08:
https://pastebin.com/BJ4vkeiq

kprinssu, 2020-08-08:
Sorry, custom forks of VoodooI2C is not supported. Please also attach a troubleshooting archive and the kernel panic log.

cheesycod, 2020-08-08:
[a.zip](https://github.com/VoodooI2C/VoodooI2C/files/5045451/a.zip)

cheesycod, 2020-08-08:
Here is the plist change

cheesycod, 2020-08-08:
[kp.zip](https://github.com/VoodooI2C/VoodooI2C/files/5045455/kp.zip)

cheesycod, 2020-08-08:
And here are all the kernel panics I have ever gotten (the most current one is in a.zip)

kprinssu, 2020-08-08:
It looks like you built VoodooI2CHID from scratch rather than just editing the Info.plist. May ask what changes you have made?

Goshin, 2020-08-08:
Seems like a null pointer dereference bug in `VoodooI2CAccelerometerSensor::setPowerState`.
Try removing `Sensor Hub Event Driver` of `IOKitPersonalities` in VoodooI2CHID's `info.plist` or applying the fix here:

```diff
--- a/VoodooI2CHID/Sensors/VoodooI2CAccelerometerSensor.cpp
+++ b/VoodooI2CHID/Sensors/VoodooI2CAccelerometerSensor.cpp
@@ -98,7 +98,9 @@ IOReturn VoodooI2CAccelerometerSensor::setPowerState(unsigned long whichState, I
         }
     } else {
         if (!awake) {
-            setElementValue(change_sensitivity, 3);
+            if (change_sensitivity) {
+                setElementValue(change_sensitivity, 3);
+            }

             awake = true;
         }
```

kprinssu, 2020-08-12:
Thanks @Goshin. Can you please create a PR so we can add it in to HID master?

Goshin, 2020-08-12:
Sure but we'd better wait for @cheesycod's confirmation?

vjkrvkarma, 2020-08-19:
> Seems like a null pointer dereference bug in `VoodooI2CAccelerometerSensor::setPowerState`.
> Try removing `Sensor Hub Event Driver` of `IOKitPersonalities` in VoodooI2CHID's `info.plist` or applying [...]

I am also facing this problem.
macOS Catalina is booting fine.
Facing this problem only with macOS Big Sur.

Tried removing **Sensor Hub Event Driver**, still facing the same issue.

Any other suggestions to fix this?

Using the latest release versions (VoodooI2C and VoodooI2CHID).
Device: Synaptics SYNA7DB5
System Files that can help and video: https://we.tl/t-PI6hMWlRIE

Goshin, 2020-09-11:
@vjkrvkarma 

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5208618/Archive.zip)
test this. Paste the log if you encounter kernel panic.

xtrs84zk, 2020-10-10:
I was experiencing the same on Big Sur, and managed to boot by downgrading VoodooPS2 to 2.0.4 (before the introduction of VoodooInput) 
Happy to provide logs or open another issue if required.

usr-sse2, 2020-11-14:
> Seems like a null pointer dereference bug in `VoodooI2CAccelerometerSensor::setPowerState`.
> Try removing `Sensor Hub Event Driver` of `IOKitPersonalities` in VoodooI2CHID's `info.plist` or applying [...]

No, panic with this patch again.

usr-sse2, 2020-11-14:
> Seems like a null pointer dereference bug in `VoodooI2CAccelerometerSensor::setPowerState`.
> Try removing `Sensor Hub Event Driver` of `IOKitPersonalities` in VoodooI2CHID's `info.plist` or applying [...]

Seems to fix the panic. Before this patch almost every second reboot (including Recovery) gave a panic in setPowerState, now no panics, except unrelated AppleIntelFramebufferCapri which happened one time.

commit d97cc7c8 (2020-12-20): Fixed #357 panic caused by HID's accelerometer
closed 2020-12-20 by kprinssu