VoodooI2C/VoodooI2C #358: Alps + I2C touchpad = trackpad gestures not showing up + touchpad partly working

LucasDondo, 2020-08-10:
Hi developers!
First, thank you for the great work and making the hackintoshing procedure easier and less painful!

**Some specs:**

- macOS Catalina 10.15.6
- OpenCore 0.6.0
- All drivers up to date at the time of posting this

So, I have a touchpad model Alps AUI1667 I2C. There is a [kext to use Alps touchpads](https://github.com/1Revenger1/VoodooPS2-Alps), but it's just available for PS2.
I can confirm that my config.plist is correctly configured, the kexts' order is fine and I have no extra I2C kexts loaded.

This is more like asking for a kext to solve this more than an issue. I'll wait for your replies!

ettingshausen, 2020-08-10:
You aren't likely to get it full working for an ALPS device.
See #351

LucasDondo, 2020-08-10:
> You aren't likely to get it full working for an ALPS device.
> See #351

Mmm... That issue is a sad story...

kprinssu, 2020-08-10:
Sorry, but I2C based Alps devices that use a proprietary protocol are not supported.

closed 2020-08-10 by kprinssu

SP4RKiOP, 2022-06-20:
Any fix ?