VoodooI2C/VoodooI2C #362: Click+drag bug/problem with ELAN12

okram24, 2020-08-10:
I'm using VoodooI2C with VoodooI2CHID in OC kext folder, pinned with GPI0 0x1B. This is the problem I've encountered. A restart usually fixes the problem but sometimes it stays even after multiple reboots. Only after a period of time of computer inactivity does it go back to normal. I also have a HID touchscreen, maybe that's interfering.

Sorry for not reading the issue format... I have searched this issue on the internet and I have found nothing like my problem.
[VoodooI2C problem report.zip](https://github.com/VoodooI2C/VoodooI2C/files/5052063/VoodooI2C.problem.report.zip)

okram24, 2020-08-11:
After future testing, I found my device works with gestures by only using the following SSDT (and also SSDT-OSCHECK). I have removed the _CRS patch and the pinning. This is weird... I've also tried many pins and ALL pins that I've tried worked with the previous setup, including 0x0000.

Only issue is the dragging problem the video above shows.

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5056356/Archive.zip)

kprinssu, 2020-08-11:
This seems to be a configuration issue rather than a code issue. Please use Gitter for support.

closed 2020-08-11 by kprinssu

okram24, 2020-08-11:
I can guarantee it's not a configuration issue, the kexts work and all gestures, I've pinpointed the main problem, once I pinch in with 3 fingers and the thumb as to open Launchpad, or pinch out, I instantly get this problem. Is there a way to disable this gesture inside the kexts?