VoodooI2C/VoodooI2C #364: VoodooI2CSynaptics requires 3 fingers for secondary click

TheRandMan, 2020-08-11:
**Describe the bug**
With VoodooI2CSynaptics, secondary click (right click) requires 3 fingers instead of 2. Same case with tap to click on or off.

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes

**System Environment**
Please answer:
1. What is your operating system? 
 - MacOS Catalina 10.15.6
2. What version of the drivers (including drivers) are you using?
 - 2.4.4 VoodooI2C and VoodooI2CSynaptics
3. What is your hardware configuration?
 - Dell Chromebook 13 7310 (SYNA0000)

I believe it's related to this change in VoodooI2C https://github.com/VoodooI2C/VoodooI2C/pull/333

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5054623/Archive.zip)

kprinssu, 2020-08-11:
An easy to way to troubleshoot if that this is a regression (or potential issue we need to fix with Synaptics satellite) is to use an older version of the kext that does not contain the above commit. Can you please report back after doing so?

TheRandMan, 2020-08-11:
Can confirm the issue doesn't exist on version 2.4.3 of VoodooI2C.kext
Tested with VoodooI2CSynaptics.kext from 2.4.3 and 2.4.4

kprinssu, 2020-08-11:
The only changes within 2.4.4 for Synaptics are the following:

- https://github.com/VoodooI2C/VoodooI2CSynaptics/commit/804db5c7da71365807555f94e77258f83e57d616
- https://github.com/VoodooI2C/VoodooI2CSynaptics/commit/401d7860b33ff954062f50674f28649839caabad

I suggest you individually revert them and test it.

TheRandMan, 2020-08-11:
Sorry, I probably didn't word that very well. Both versions of the Synaptics kext work well with the v2.4.3 of VoodooI2C.kext.

The issue is only present when using v2.4.4 of VoodooI2C.kext specifically (regardless of Synaptics version).

I suspect the Synaptics kext might have some incompatibility with the new fingertypes changes to VoodooInput/VoodooI2C.

Goshin, 2020-08-11:
VoodooI2CSynaptics reports `contact_count` as the number of transducers rather than the current finger touches, so VooodooI2C will always set a wrong finger type for the first touch.

https://github.com/VoodooI2C/VoodooI2CSynaptics/blob/6e2eed3cdd0a30a3f79d062e1d6a7430de0e059b/VoodooI2CSynaptics/VoodooI2CSynapticsDevice.cpp#L127
```c++
    if (mt_interface) {
        VoodooI2CMultitouchEvent event;
        event.transducers = transducers;
        event.contact_count = transducers->getCount();
        
        mt_interface->handleInterruptReport(event, timestamp);
```

The issue can be fixed here in VoodooI2CSynaptics by simply counting the valid touches and correcting the `contact_count`, but it requires the valid transducers are not arranged discretely or it can cause other problems.
```diff
@@ -124,7 +124,15 @@ void VoodooI2CSynapticsDevice::TrackpadRawInput(uint8_t report[40]){
     if (mt_interface) {
         VoodooI2CMultitouchEvent event;
         event.transducers = transducers;
-        event.contact_count = transducers->getCount();
+        int count = 0;
+        for (int i = 0; i < transducers->getCount(); i++) {
+            if (VoodooI2CDigitiserTransducer* transducer = OSDynamicCast(VoodooI2CDigitiserTransducer, transducers->getObject(i))) {
+                if (transducer->is_valid) {
+                    count++;
+                }
+            }
+        }
+        event.contact_count = count;

         mt_interface->handleInterruptReport(event, timestamp);
     }
```

Or we can fix it in the VoodooI2C side by only counting the valid touch when setting the finger type.

@TheRandMan 
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5058720/Archive.zip) You can test this and see if it works for you.

TheRandMan, 2020-08-12:
@Goshin That did the trick! 2 finger clicks work. All gestures still work. Not seeing any negative side-effects. The work you guys do is amazing.

TheRandMan, 2020-08-12:
One odd thing I have noticed (and this is a bit difficult to explain) -  if I do a 2 finger click then try to move the cursor by releasing one finger, it only works if I release the second finger to touch the touchpad. Not a critical issue but a bit odd.

Goshin, 2020-08-12:
Yes that is the side effect I meant

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5061517/Archive.zip) you can try this, I fix it in VoodooI2C side,

```diff
@@ -28,6 +28,8 @@ MultitouchReturn VoodooI2CNativeEngine::handleInterruptReport(VoodooI2CMultitouc
     if (transducer->type == kDigitiserTransducerStylus)
         stylus_check = 1;

+    int valid_touch_count = 0;
+
     for (int i = 0; i < event.contact_count; i++) {
         VoodooI2CDigitiserTransducer* transducer = (VoodooI2CDigitiserTransducer*) event.transducers->getObject(i+stylus_check);
         VoodooInputTransducer* inputTransducer = &message.transducers[i];
@@ -42,6 +44,9 @@ MultitouchReturn VoodooI2CNativeEngine::handleInterruptReport(VoodooI2CMultitouc
         inputTransducer->type = (transducer->type == DigitiserTransducerType::kDigitiserTransducerFinger) ? VoodooInputTransducerType::FINGER : VoodooInputTransducerType::STYLUS;

         inputTransducer->isValid = transducer->is_valid;
+        if (inputTransducer->isValid) {
+            valid_touch_count++;
+        }
         inputTransducer->isTransducerActive = transducer->tip_switch.value();
         inputTransducer->isPhysicalButtonDown = transducer->physical_button.value();

@@ -71,7 +76,7 @@ MultitouchReturn VoodooI2CNativeEngine::handleInterruptReport(VoodooI2CMultitouc
     }

     // set the thumb to improve 4F pinch and spread gesture and cross-screen dragging
-    if (event.contact_count >= 4 || transducer->physical_button.value()) {
+    if (valid_touch_count >= 4 || transducer->physical_button.value()) {
         // simple thumb detection: to find the lowest finger touch in the vertical direction.
         UInt32 y_max = 0;
         int thumb_index = 0;
```

TheRandMan, 2020-08-12:
That has completely fixed the reported 2 finger click issue but seems to have brought back another issue I remember reading about in the past. When performing a 3 or 4 finger swipe gesture (between full screen apps for example), all fingers must be lifted in order for the gesture to be completed.

Personally, I don't mind the behavior, but I know you worked to fix it in the past so I thought I should mention it.

Goshin, 2020-08-12:
I am not sure, but isn't that expected?

TheRandMan, 2020-08-12:
Sorry, I thought I read previously that that wasn't "proper" behavior for MacOS. However, you are correct. I just tested on a MacBook and the behavior is the same. All is good and everything is working perfectly as far as I see. 

Is there somewhere I can donate to the project?

alexandred, 2020-08-12:
@TheRandMan We don’t accept donations as the project itself has no expenses and there are now too many contributors for it to be in any way fairly distributed between them.

kprinssu, 2020-08-12:
@Goshin can you spin up a PR(s) with the fixes?

the-eric-kwok, 2020-08-12:
Works perfectly for me! Great job! Looking forward to the next release already!

closed 2020-08-12 by kprinssu

the-eric-kwok, 2020-08-14:
Touchpad don't respond my touch after sleep and wake.
[log.zip](https://github.com/VoodooI2C/VoodooI2C/files/5072328/log.zip)

kprinssu, 2020-08-14:
@the-eric-kwok please create a new Github issue, please try reverting https://github.com/VoodooI2C/VoodooI2CSynaptics/commit/804db5c7da71365807555f94e77258f83e57d616

PR #367 (2020-08-12): Count valid touch events when setting finger type
- Count the valid touch events when setting the thumb in VoodooI2CNativeEngine. 

Fix #364