VoodooI2C/VoodooI2C #368: VoodooI2CPCIController does not support 8th Gen Intel Core Processor LPSS: I2C Controller # 2

Scrubber777, 2020-08-13:
Hello!
Add to support 8th Gen Intel Core Processor LPSS: I2C Controller # 2 (8086 9DEA 17AA 3805 Intel Corporation)
Serial bus controlle CI0@0/I2C2@15,2 pci8086,9dea    PciRoot(0x0)/Pci(0x15,0x2)

    "DevicePath" : "PciRoot(0x0)\/Pci(0x15,0x2)",
    "DeviceName" : "???",
    "ShadowDevice" : 40426,
    "IORegPath" : "IOService:\/AppleACPIPlatformExpert\/PCI0@0\/AppleACPIPCI\/I2C2@15,2",
    "SlotName" : "Internal@0,21,2",
    "ASPM" : "Disabled",
    "IORegIOName" : "pci8086,9dea",
    "ClassCode" : 819200,
    "Address" : 1376258,
    "PCIDebug" : "00:15.2",
    "VendorName" : "Intel Corporation",
    "ClassName" : "Serial bus controller",
    "SubDeviceID" : 14341,
    "VendorID" : 32902,
    "IORegName" : "\/PCI0@0\/I2C2@15,2",
    "Model" : "???",
    "SubClassName" : "???",
    "SubVendorID" : 6058,
    "ShadowVendor" : 32902,
    "DeviceID" : 40426,
    "BundleID" : "com.apple.driver.AppleACPIPlatform"

kprinssu, 2020-08-13:
Please add your controller to the the Info.plist like so:

https://github.com/VoodooI2C/VoodooI2C/commit/bbb77747f289880961399a89ce4794b4f9c8a514

You want to add:
`<string>pci8086,9dea</string>`

VoodooI2C should support your controller, but it's not loading. The above change will make VoodooI2C to attempt a load your controller.

zhen-zen, 2020-09-11:
This id should have been included in #370.

PR #370 (2020-10-08): Separate VoodooI2CPCILakeController
Quick implementation for #363.
Need testing for CML/ICL (should work without issue).
Anyone has a better name for the inherited controller?

Also close #368.
closed 2020-10-08 by kprinssu